frank-w/BPI-Router-Linux #5: Add Mali GPU driver

xbgmsharp, 2018-01-05:
https://developer.arm.com/products/software/mali-drivers/utgard-kernel

jfdelnero, 2018-01-11:
i plan to try this way : 
https://github.com/yuq/mesa-lima/

frank-w, 2018-01-13:
ok, currently i try to get hdmi itself working...it seems there are some kernel-options missing (comparing with options from 4.4.70)

any information welcome :)

xbgmsharp, 2018-03-18:
Using, DX910-SW-99002-r8p1-00rel0.tgz. 
Same issue as this thread, https://community.arm.com/graphics/f/discussions/9762/mali-400-utgard-build-problem

 ```
$ export ARCH=arm;export CROSS_COMPILE=arm-linux-gnueabihf-
$ crosscompile=1
$ KDIR=~/BPI-R2-4.14/ USING_UMP=0 BUILD=release make
make ARCH=arm -C /home/xbgmsharp/BPI-R2-4.14/ M=/home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali modules
make[1]: Entering directory '/home/xbgmsharp/BPI-R2-4.14'
  CC [M]  /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_atomics.o
  CC [M]  /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_irq.o
  CC [M]  /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_wq.o
  CC [M]  /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_locks.o
  CC [M]  /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_wait_queue.o
  CC [M]  /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_low_level_mem.o
  CC [M]  /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_math.o
  CC [M]  /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_memory.o
  CC [M]  /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_misc.o
  CC [M]  /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_mali.o
  CC [M]  /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_notification.o
/home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_notification.c: In function ‘_mali_osk_notification_create’:
/home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_notification.c:60:30: error: ‘__GFP_REPEAT’ undeclared (first use in this function)
    GFP_KERNEL | __GFP_HIGH | __GFP_REPEAT);
                              ^~~~~~~~~~~~
/home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_notification.c:60:30: note: each undeclared identifier is reported only once for each function it appears in
scripts/Makefile.build:328: recipe for target '/home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_notification.o' failed
make[2]: *** [/home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_notification.o] Error 1
Makefile:1519: recipe for target '_module_/home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali' failed
make[1]: *** [_module_/home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali] Error 2
make[1]: Leaving directory '/home/xbgmsharp/BPI-R2-4.14'
Makefile:207: recipe for target 'all' failed
make: *** [all] Error 2
```

xbgmsharp, 2018-03-18:
https://github.com/96boards/linux/pull/23

frank-w, 2018-03-18:
replace __GFP_REPEAT by __GFP_RETRY_MAYFAIL or comment it out

xbgmsharp, 2018-03-19:
THanks one step further. in parallele I try to embedded the mali driver inside kernel 4.14 as it is done 4.4 and in the pull request with no luck.
```
$ KDIR=~/BPI-R2-4.14/ USING_UMP=0 BUILD=release make
make ARCH=arm -C /home/xbgmsharp/BPI-R2-4.14/ M=/home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali modules
make[1]: Entering directory '/home/xbgmsharp/BPI-R2-4.14'
  CC [M]  /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_notification.o
  CC [M]  /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_time.o
  CC [M]  /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_timers.o
  CC [M]  /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_osk_bitmap.o
  CC [M]  /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_memory.o
In file included from /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_sync.h:30:0,
                 from /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/common/mali_timeline.h:19,
                 from /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/common/mali_pp_job.h:22,
                 from /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_memory_swap_alloc.h:18,
                 from /home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_memory.c:31:
/home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_internal_sync.h:115:2: error: unknown type name ‘wait_queue_t’
  wait_queue_t work;
  ^~~~~~~~~~~~
scripts/Makefile.build:328: recipe for target '/home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_memory.o' failed
make[2]: *** [/home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali/linux/mali_memory.o] Error 1
Makefile:1519: recipe for target '_module_/home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali' failed
make[1]: *** [_module_/home/xbgmsharp/DX910-SW-99002-r8p1-00rel0/driver/src/devicedrv/mali] Error 2
make[1]: Leaving directory '/home/xbgmsharp/BPI-R2-4.14'
Makefile:207: recipe for target 'all' failed
make: *** [all] Error 2

```

xbgmsharp, 2018-03-20:
Using kernel 4.9 from your repo i was able to compile the mali driver as a module.
```
# insmod /lib/modules/4.9.44-bpi-r2-v1+/kernel/extras/mali.ko 
# lsmod 
Module                  Size  Used by
mali                  271545  0
mtkhnat                14788  0
mtk_thermal             5385  0
thermal_sys            52939  1 mtk_thermal
mt6577_auxadc           5340  0
spi_mt65xx              8636  0
pwm_mediatek            3316  0
bridge                129855  0
nfsd                  110691  11
nfs_acl                 3127  1 nfsd
lockd                  70891  1 nfsd
grace                   2885  1 lockd
sunrpc                286964  17 nfsd,nfs_acl,lockd
ip_tables              12340  0
x_tables               19090  1 ip_tables
ipv6                  387579  58 bridge
root@bpi-r2-emmc:~# dmesg | tail
[  459.404818] mali: loading out-of-tree module taints kernel.
[  459.416803] Mali: 
[  459.418647] Mali device driver loaded
```

xbgmsharp, 2018-04-01:
For Kernel 4.9 add Mali support via commit beb8c1156605fbb83099d97b66d6464462b7de34

frank-w, 2020-06-14:
mali is part of mainline linux 5.3+

closed 2020-06-14 by frank-w

Jir8taiwan, 2020-09-30:
I get help about:
frank-w commented on 19 Mar 2018
replace __GFP_REPEAT by __GFP_RETRY_MAYFAIL or comment it out

however, I still have another alarm:
```
/usr/local/src/DX910-SW-99002-r10p0-00rel0/driver/src/devicedrv/ump/linux/ump_kernel_memory_backend_os.c:138:91: error: ?_GFP_COLD??undeclared (first use in this function)
  138 |    new_page = alloc_page(GFP_HIGHUSER | __GFP_ZERO | __GFP_RETRY_MAYFAIL | __GFP_NOWARN | __GFP_COLD);
      |      
                                                                                    ^~~~~~~~~~
``` 

Does anyone can help me as a new learner?
Thanks a lot.

My system is Banana Pi M1, Armbian 20.08.03 with Kernel linux-source-5.8.11-sunxi

frank-w, 2020-09-30:
5.4+ has mali driver included...you do not need to add external source