frank-w/BPI-Router-Linux #6: Add Bluetouh support

xbgmsharp, 2018-01-05:
There is no bluetouh interface on 4.4.11.
Install:
```
apt-get install bluez bluez-utils
```
Enable interface:
```
hciconfig hci0 up
```

frank-w, 2018-01-05:
i had only tested wifi-AP after merging the combo-device-driver...there are also Problems with BT on 4.4:

http://forum.banana-pi.org/t/bpi-r2-new-image-ubuntu-16-04-v1-2-1-bt-and-wifi-ap-mode-are-working-fine-2017-11-27/4291/16

currently there is much work on other things so i will accept patches but do not work on BT now

frank-w, 2018-01-07:
debug merged to main

bt works like in 4.4
occasionally hangs and bluetoothd does not work

frank-w, 2018-01-07:
BT is actually not build in...

`obj-$(CONFIG_MTK_COMBO_BT_HCI) += bt/`

is missing in drivers/misc/mediatek/connectivity/Makefile

xbgmsharp, 2018-01-11:
Kernel config to activate BT:

```
$ grep BT .config | grep -v '#'
CONFIG_BT=y
CONFIG_BT_BREDR=y
CONFIG_BT_HS=y
CONFIG_BT_LE=y
CONFIG_BT_DEBUGFS=y
CONFIG_MTK_BTIF=y
CONFIG_MTK_COMBO_BT=y
CONFIG_MTK_COMBO_BT_HCI=y
```

frank-w, 2018-01-11:
added that to branch "debug_bt"

alxchk, 2019-01-10:
I managed to make bluetooth work.
This is a bit tricky.

MTK connectivity requires userspace shit to initialize device (both WIFI and BT). In case module compiled into the kernel, routine which completed device setup executed before chip can actually be initialized.

So, to fix things up you need to build stp_chrdev_bt.ko as a module, and insert it after stp_uart_launcher is running. In this case things just works.

frank-w, 2019-01-10:
nice to hear that, can you figure out why this happen and how i can build the only BT as module? afair i cannot build the combo-device as module.

alxchk, 2019-01-10:
Yes.. Shortly saying you need to get here - https://github.com/frank-w/BPI-R2-4.14/blob/4.14-main/net/bluetooth/hci_core.c#L2104
Which is called from hci_power_on, which is called from https://github.com/frank-w/BPI-R2-4.14/blob/4.14-main/net/bluetooth/hci_core.c#L3160 (hci_register_dev).

You need set `CONFIG_MTK_COMBO_BT=m`, and add to simple hacks, as here - https://github.com/alxchk/BPI-R2-4.14/commit/5ea5c3cacc7fa0300a4fe9149e0df60797dab52f (bluetooth_drv_init.c, stp_chrdev_bt.c)

xbgmsharp, 2019-01-11:
Thanks for the update.
Is`CONFIG_MTK_COMBO_BT_HCI` needed? Does both need to be compile as module?
```
$ grep CONFIG_MTK_COMBO_BT arch/arm/configs/mt7623n_evb_fwu_defconfig
CONFIG_MTK_COMBO_BT=y
CONFIG_MTK_COMBO_BT_HCI=y
```

According to https://github.com/alxchk/BPI-R2-4.14/commit/5ea5c3cacc7fa0300a4fe9149e0df60797dab52f it should like:
```
CONFIG_MTK_COMBO_BT=m
CONFIG_MTK_COMBO_BT_HCI=y
```

frank-w, 2019-01-11:
afaik on next boot BT-Module gets loaded before you can call stp_uart_launcher...

so BT have to initialized via script (unload module, run stp_uart_launcher, load module)

am i right? pushed to 4.19-bt, but had not yet time to test :(

alxchk, 2019-01-11:
This module will not be autoloaded. So modprobe it manually after stp launcher.
I have bunch of systemd unit files for that, can share.

frank-w, 2019-01-11:
have to test it before, currently busy with some mainline-discussions/tests (dsa, fbdev, linux 5.0)

i think i will integrate it in wifi.sh, which calles stp_uart_launcher for wifi...

as far as i see you disable code...imho if that works...code should be dropped completely

frank-w, 2019-01-11:
```
root@bpi-r2:~# bluetoothctl                                                     
[NEW] Controller 00:00:46:85:90:01 bpi-r2 [default]                             
[bluetooth]# power on                                                           
Changing power on succeeded                                                     
[bluetooth]# scan on                                                            
Discovery started                                                               
[CHG] Controller 00:00:46:85:90:01 Discovering: yes                             
[NEW] Device 00:14:03:05:08:AE 00-14-03-05-08-AE                                
[NEW] Device 9C:8C:6E:4A:F8:23 [TV] Samsung 8 Series (65)                       
[NEW] Device B4:9D:0B:5F:2A:B9 Aquaris X5 Plus                                  
[CHG] Device 00:14:03:05:08:AE LegacyPairing: no                                
[CHG] Device 00:14:03:05:08:AE Name: BPI-R2                                     
[CHG] Device 00:14:03:05:08:AE Alias: BPI-R2                                    
[bluetooth]# scan off                                                           
[CHG] Device B4:9D:0B:5F:2A:B9 RSSI is nil                                      
[CHG] Device 9C:8C:6E:4A:F8:23 RSSI is nil                                      
[CHG] Device 00:14:03:05:08:AE RSSI is nil                                      
[CHG] Controller 00:00:46:85:90:01 Discovering: no                              
Discovery stopped                                                               
[bluetooth]# pair B4:9D:0B:5F:2A:B9                                             
Attempting to pair with B4:9D:0B:5F:2A:B9                                       
[CHG] Device B4:9D:0B:5F:2A:B9 Connected: yes                                   
[CHG] Device B4:9D:0B:5F:2A:B9 Modalias: bluetooth:v000Fp1200d1436              
....   
[CHG] Device B4:9D:0B:5F:2A:B9 ServicesResolved: yes                            
[CHG] Device B4:9D:0B:5F:2A:B9 Paired: yes                                      
Pairing successful
```
nice ;)

https://github.com/frank-w/BPI-R2-4.14/commits/4.19-bt

frank-w, 2019-02-25:
@alxchk

There seem still issues with btle,can you guide me/us to debug this? Where should we add debug-printk?

xptsp, 2022-01-15:
I don't know if this needs to be posted, however, it seems Bluetooth was added at some point.

Around 14 days ago (from this message), Bluetooth was fixed for kernels 5.4, 5.6 thru 5.10 and 5.15.  I have personally have not tested other versions.  Please have a look at the thread I'm referencing: [https://forum.banana-pi.org/t/bpi-r2-internal-wifi-bt-mt6625l-kernel/4307/355](https://forum.banana-pi.org/t/bpi-r2-internal-wifi-bt-mt6625l-kernel/4307/355) .

Maybe this issue should be closed?

frank-w, 2022-01-15:
BTLE is imho not possible,but basicly BT is working again.
Thx for pointing to it

closed 2022-01-15 by frank-w