frank-w/BPI-Router-Linux #8: Stuck at "waiting for root device ..."

WeiqiChenCN, 2018-01-07:
Hi , Thank you for porting kernel for R2.

I compiled the kernel form your repository, and stuck at "Waiting for root device..".
I compare `.config` bewteen version 4.4 and 4.14 and it seem no option about enabling mmc.

Is it no emmc / sd mmc support yet?

frank-w, 2018-01-07:
Hi,

Both should working (did it in my system).

your root-device should be /dev/mmcblk0p2 if booting from sd and /dev/mmcblk1p2 if booting from emmc like numbered on 4.4. Please look at cmdline whe kernel starts booting (after loaded by uboot). Maybe you have to change your uEnv.txt. maybe i have to change default cmdline in kernel

WeiqiChenCN, 2018-01-07:
I re-compiled the kernel, still not work

Here is the full log:
```

[USBD] USB PRB0 LineState: 0

[USBD] USB cable/ No Cable inserted!

[PLFM] Keep stay in USB Mode
Platform initialization is ok
wait for frequency meter finish, CLK26CALI = 0x81
mt_pll_post_init: mt_get_cpu_freq = 1040000Khz
wait for frequency meter finish, CLK26CALI = 0x90
mt_pll_post_init: mt_get_bus_freq = 273000Khz
wait for frequency meter finish, CLK26CALI = 0x81
mt_pll_post_init: mt_get_mem_freq = 133250Khz
[PWRAP] pwrap_init_preloader
[PWRAP] pwrap_init
[PWRAP] _pwrap_init_sistrobe [Read Test] fail,index=0,rdata=2D52
[PWRAP] _pwrap_init_sistrobe [Read Test] fail,index=1,rdata=2D52
[PWRAP] _pwrap_init_sistrobe [Read Test] fail,index=2,rdata=2D52
[PWRAP] _pwrap_init_sistrobe [Read Test] fail,index=3,rdata=5AA4
[PWRAP] _pwrap_init_sistrobe [Read Test] pass,index=4 rdata=5AA5
[PWRAP] _pwrap_init_sistrobe [Read Test] pass,index=5 rdata=5AA5
[PWRAP] _pwrap_init_sistrobe [Read Test] pass,index=6 rdata=5AA5
[PWRAP] _pwrap_init_sistrobe [Read Test] pass,index=7 rdata=5AA5
[PWRAP] _pwrap_init_sistrobe [Read Test] pass,index=8 rdata=5AA5
[PWRAP] _pwrap_init_sistrobe [Read Test] pass,index=9 rdata=5AA5
[PWRAP] _pwrap_init_sistrobe [Read Test] pass,index=10 rdata=5AA5
[PWRAP] _pwrap_init_sistrobe [Read Test] fail,index=11,rdata=B50B
[PWRAP] _pwrap_init_sistrobe [Read Test] fail,index=12,rdata=B54B
[PWRAP] _pwrap_init_sistrobe [Read Test] fail,index=13,rdata=B54B
[PWRAP] _pwrap_init_sistrobe [Read Test] fail,index=14,rdata=B54B
[PWRAP] _pwrap_init_sistrobe [Read Test] fail,index=15,rdata=B54B
[PWRAP] _pwrap_init_sistrobe [Read Test] fail,index=16,rdata=B54B
[PWRAP] _pwrap_init_sistrobe [Read Test] fail,index=17,rdata=B54B
[PWRAP] _pwrap_init_sistrobe [Read Test] fail,index=18,rdata=B54B
[PWRAP] _pwrap_init_sistrobe [Read Test] fail,index=19,rdata=6297
[PWRAP] _pwrap_init_sistrobe [Read Test] fail,index=20,rdata=6A97
[PWRAP] _pwrap_init_sistrobe [Read Test] fail,index=21,rdata=6A97
[PWRAP] _pwrap_init_sistrobe [Read Test] fail,index=22,rdata=6A97
[PWRAP] _pwrap_init_sistrobe [Read Test] fail,index=23,rdata=6A97
[PWRAP] _pwrap_init_reg_clock
[PMIC_WRAP]wrap_init pass,the return value=0.
[pmic6323_init] Preloader Start..................
[pmic6323_init] PMIC CHIP Code = 0x2023
INT_MISC_CON: 1  TOP_RST_MISC: 1
pl pmic powerkey Release
[pmic6323_init] powerKey = 0
[pmic6323_init] is USB in = 0xB004
[pmic6323_init] Reg[0x11A]=0x1B
pmic setup LED
[pmic6323_init] Done...................
mt7623 disable long press reset ->>>>>
mt7623 disable long press reset <<<<<-
mt7623 VPA supplied by 1.0V to MT7530 ->
mt7623 VPA supplied by 1.0V to MT7530 <-
mt7623 enables RG_VGP1_EN for LCM ->
mt7623 enables RG_VGP1_EN for LCM <-
MT7623 E2 setting =>
MT7623 E2 setting <=
[PLFM] Init I2C: OK(0)
[PLFM] Init PWRAP: OK(0)
[PLFM] Init PMIC: OK(0)
[PLFM] chip[CA00]

[BLDR] [Support SD/eMMC] Build Time: 20170905-120917
==== Dump RGU Reg ========
RGU MODE:     4D
[... 1122 lines cut ...]
[    4.742295] ThumbEE CPU extension supported.
[    4.746556] Registering SWP/SWPB emulation handler
[    4.752740] Btrfs loaded, crc32c=crc32c-generic
[    4.758046] Key type encrypted registered
[    4.770601] mt6577-uart 11002000.serial: could not find pctldev for node /pinctrl@10005000/uart@0, deferring probe
[    4.781118] mt6577-uart 11003000.serial: could not find pctldev for node /pinctrl@10005000/uart@1, deferring probe
[    4.791568] mt6577-uart 11005000.serial: could not find pctldev for node /pinctrl@10005000/uart@3, deferring probe
[    4.802206] mtk-spi 1100a000.spi: could not find pctldev for node /pinctrl@10005000/spi@0, deferring probe
[    4.812391] xhci-mtk 1a1c0000.usb: 1a1c0000.usb supply vbus not found, using dummy regulator
[    4.820861] xhci-mtk 1a1c0000.usb: fail to get vusb33
[    4.826411] xhci-mtk 1a240000.usb: 1a240000.usb supply vbus not found, using dummy regulator
[    4.834876] xhci-mtk 1a240000.usb: fail to get vusb33
[    4.840178] i2c-mt65xx 11007000.i2c: could not find pctldev for node /pinctrl@10005000/i2c@0, deferring probe
[    4.850191] i2c-mt65xx 11008000.i2c: could not find pctldev for node /pinctrl@10005000/i2c@1, deferring probe
[    4.860193] mtk-msdc 11240000.mmc: could not find pctldev for node /pinctrl@10005000/mmc1default, deferring probe
[    4.870528] mtk-msdc 11230000.mmc: could not find pctldev for node /pinctrl@10005000/mmc0default, deferring probe
[    4.881554] ALSA device list:
[    4.884506]   No soundcards found.
[    4.888377] md: Waiting for all devices to be available before autodetect
[    4.895120] md: If you don't use raid, use raid=noautodetect
[    4.901389] md: Autodetecting RAID arrays.
[    4.905488] md: autorun ...
[    4.908261] md: ... autorun DONE.
[    4.911737] Waiting for root device PARTUUID=5700d056-02...
```

WeiqiChenCN, 2018-01-07:
Here is my uboot environment
```
BPI-EMMC> printenv
arch=arm
baudrate=115200
board=bpi-r2
board_name=mt7623_evb
boot0=setenv mmc 0; run bootcmd
boot1=setenv mmc 1; run bootcmd
boot2=run bootcmd
boot3=setenv mmc 0; run bootfallback
boot4=setenv mmc 1; run bootfallback
bootargs=board=bpi-r2 earlyprintk console=tty1 fbcon=map:0 console=ttyS0,115200 vmalloc=496M debug=7 t
bootcmd=mmc init $mmc; run loadbootenv; run uenvcmd
bootdelay=3
bootenv=uEnv.txt
bootfallback=mmc init $mmc; run loadbootenv; run ufallbackcmd
bootfile=iverson_uImage
bootmenu_0=1. Boot from internal emmc.=run boot0
bootmenu_1=2. Boot from external sd-card.=run boot1
bootmenu_2=3. Boot from default medium.=run boot2
bootmenu_3=4. Fallback boot from internam emmc.=run boot3
bootmenu_4=5. Fallback boot from external sd-card.=run boot4
bpi=bananapi
bpiver=1
chip=MT7623N
cpu=armv7
debug=7
device=mmc
ethact=mtk_eth
ethaddr=00:0C:E7:11:22:33
ipaddr=10.10.10.254
kernel=uImage
loadaddr=0x84000000
loadbootenv=fatload ${device} ${partition} ${scriptaddr} ${board}/${service}/${bootenv} ||fatload ${d;
mmc=0
partition=0:1
scriptaddr=0x83000000
serverip=10.10.10.3
service=linux
soc=mt7623
stderr=serial
stdin=serial
stdout=serial
vendor=mediatek
```

WeiqiChenCN, 2018-01-07:
Here is the info in the fat32 partition:
```
$ find
.
./uEnv.txt
./bpi-r2
./bpi-r2/4.14.12-bpi-r2
./bpi-r2/4.14.12-bpi-r2/uImage
./bpi-r2/linux
./bpi-r2/linux/uImage
./bpi-r2/linux/uEnv.txt.backup
./bpi-r2/4.4.70-bpi-r2-2
./bpi-r2/4.4.70-bpi-r2-2/uImage
$ cat uEnv.txt 
bpi=bananapi
board=bpi-r2
chip=mt7623n
#
kernel=uImage
initrd=uInitrd
dtb=mt7623n-bpi-r2.dtb
#
target=4.14.12-bpi-r2
fallback=linux
root=PARTUUID="5700d056-02" rootfstype=ext4 rootwait
#
kaddr=0x84000000
rdaddr=0x86000000
dtaddr=0x88000000
#
console=earlyprintk console=tty1 console=ttyS0,115200 fbcon=map:0
bootopts=vmalloc=496M debug=7 initcall_debug=0
abootargs=setenv bootargs board=${board} console=${console} root=${root} service=${service} ${bootopt}
ahello=echo Banana Pi ${board} chip: $chip Service: $service
## load kernel
aload_kernel_cmd=fatload $device $partition $kaddr ${board}/${service}/${kernel}
aload_kernel=if run aload_kernel_cmd ; then echo Kernel loaded. ; else exit; fi
## load initrd
aload_initrd_cmd=fatload $device $partition $rdaddr ${board}/${service}/${initrd}
aload_initrd=if run aload_initrd_cmd; then echo Initrd loaded. ; else set rdaddr - ; fi
## load device tree
aload_device_tree_cmd=fatload $device $partition $rdaddr ${board}/${service}/${dtb}
aload_device_tree=if run aload_device_tree_cmd; then echo Device Tree loaded. ; else set dtaddr - ; fi
##bootm
abootm=bootm $kaddr $rdaddr $dtaddr
#
uenvcmd=setenv service ${target}; run ahello abootargs aload_kernel aload_initrd aload_device_tree abm
ufallbackcmd=setenv service ${fallback}; run ahello abootargs aload_kernel aload_initrd aload_device_m
########################################################
########################################################
########################################################
########################################################

```

WeiqiChenCN, 2018-01-07:
I seem that some warnning message in boot-up log..

WeiqiChenCN, 2018-01-07:
And here is how i compiled the kernel `work.sh`:
```
#!/bin/bash
export ARCH=arm
export KPKG_ARCH=armhf
export CROSS_COMPILE=arm-linux-gnueabihf-
export LOADADDR=0x80008000
export LOCALVERSION=""
##you may make mt7623n_evb_defconfig
#fakeroot make-kpkg --initrd --uimage --verbose binary-arch
#fakeroot make-kpkg --uimage --verbose binary-arch
make bindeb-pkg
```

`mkimg.sh`
```
#!/bin/bash
export ARCH=arm
export KPKG_ARCH=armhf
export CROSS_COMPILE=arm-linux-gnueabihf-
export LOADADDR=0x80008000
export LOCALVERSION=""
make uImage dtbs
cat arch/arm/boot/zImage arch/arm/boot/dts/mt7623n-bananapi-bpi-r2.dtb > arch/arm/boot/zImage-dtb
mkimage -A arm -O linux -T kernel -C none -a 80008000 -e 80008000 -n "Linux Kernel $(git describe --tags)" -d arch/arm/boot/zImage-dtb ./uImage
```

frank-w, 2018-01-07:
Please try with my build.sh and define a root in uenv pointing to /dev not partuuid

frank-w, 2018-01-07:
which config do you use?

./build.sh importconfig
./build.sh

WeiqiChenCN, 2018-01-07:
I copied .config from 4.4.70 before.
Now I'm trying the default config: `make mt7623n_evb_fwu_defconfig`...

WeiqiChenCN, 2018-01-07:
It works now!!

```
make mt7623n_evb_fwu_defconfig
```

fix the problem.

Thank for your help!!!

closed 2018-01-07 by WeiqiChenCN