frank-w/BPI-Router-Linux #10: HDMI not working

frank-w, 2018-01-14:
added hdmi-nodes from 4.4.70 in hdmi-branch, but kernel seems not to be loaded (debug-uart stops after "Starting kernel ..."-message from uboot)

jfdelnero, 2018-01-23:
In mt7623.dtsi :

Enabling all these sections in the same time : 

smi_common,
mipi_tx0,
iommu,
mali,
display_components,
color,
dsi0,
mutex,
vdecsys, 
larb0, 
larb1, 
larb2, 
imgsys

Doesn't crash the BPI R2 but after enabling **one** of these sections : 

disp_rdma0,
disp_rdma1, 
wdma, 
ovl

the kernel doesn't boot anymore.

jfdelnero, 2018-01-23:
In mt7623n-bananapi-bpi-r2.dts

After enabling all disabled section :
vdd_fixed_vgpu_reg,
&mali,
mipi_dsi_pin,
and
&mipi_tx0

with all the working sections in mt7623.dtsi listed in my previous post, the kernel still working.

The problem seems to be around disp_rdma0, disp_rdma1, wdma and ovl.
the presence of the iommus references seems to be a common point here.

frank-w, 2018-01-23:
good to know, but i assume there is still missing driver-code outside the dts(i)...somewhere in drivers/...

jfdelnero, 2018-01-23:
new test : 

Reactivate all drivers/sections but comment the iommus = <&iommu > references into  disp_rdma0, disp_rdma1, wdma and ovl make the kernel working.
Now it's time try to understand why ;)

jfdelnero, 2018-01-23:
[image]

rdma0, rdma1, and ovl seems to be now present.

frank-w, 2018-01-23:
did you changed anything except enabling the nodes? it's strange that it does now boot...

jfdelnero, 2018-01-23:
Yes exactly. But of course this doesn't make the hdmi working. iommu is needed here.

frank-w, 2018-01-23:
i assume that immu needs a driver in drivers-directory of kernel...maybe you can found one in the 4.4.70-repo

jfdelnero, 2018-01-23:
well linux-mt/drivers/iommu/mtk_iommu_v1.c seems to be a good candidate.

frank-w, 2018-01-23:
https://github.com/frank-w/BPI-R2-4.14/blob/main/drivers/iommu/mtk_iommu.c exists also in 4.14...mhm. you can compare this with the 4.4-file...maybe something is changed in h-file too

jfdelnero, 2018-01-23:
yes just see this too. and there is a mtk_iommu.c too, not compiled here.

frank-w, 2018-01-23:
maybe some of these 4.4-options are missing in my config:

```
CONFIG_IOMMU_API=y
CONFIG_IOMMU_SUPPORT=y
CONFIG_IOMMU_IOVA=y
CONFIG_OF_IOMMU=y
CONFIG_IOMMU_DMA=y
# CONFIG_MTK_IOMMU is not set
CONFIG_MTK_IOMMU_V1=y
```

jfdelnero, 2018-01-23:
mtk_iommu.c doesn't seems to be compiled too in the 4.4.70.
i will check the options.

jfdelnero, 2018-01-23:
Here is what i have with the 4.14

CONFIG_ARM_DMA_USE_IOMMU=y
CONFIG_ARM_DMA_IOMMU_ALIGNMENT=8
CONFIG_IOMMU_HELPER=y
CONFIG_IOMMU_API=y
CONFIG_IOMMU_SUPPORT=y
# Generic IOMMU Pagetable Support
# CONFIG_IOMMU_IO_PGTABLE_LPAE is not set
# CONFIG_IOMMU_IO_PGTABLE_ARMV7S is not set
CONFIG_OF_IOMMU=y
# CONFIG_MTK_IOMMU is not set
CONFIG_MTK_IOMMU_V1=y

CONFIG_IOMMU_IOVA & CONFIG_IOMMU_DMA seems to be missing, or maybe they have been renamed...

jfdelnero, 2018-01-23:
In fact CONFIG_IOMMU_IOVA & CONFIG_IOMMU_DMA are selected with mtk_iommu.c only. This one doesn't seems to be enabled with the BPI R2 4.4.70. Is it in your 4.4.70 config ?

frank-w, 2018-01-23:
```
 .config - Linux/arm 4.14.14 Kernel Configuration

  │ Symbol: IOMMU_IOVA [=n]                                                 │  
  │ Type  : tristate                                                        │  
  │   Defined at drivers/iommu/Kconfig:63                                   │  
  │   Depends on: IOMMU_SUPPORT [=y]                                        │  
  │   Selected by: SCIF [=n] && 64BIT && PCI [=y] && X86 && SCIF_BUS [=n] & │  

 Symbol: IOMMU_DMA [=n]                                                  │  
  │ Type  : boolean                                                         │  
  │   Defined at drivers/iommu/Kconfig:71                                   │  
  │   Depends on: IOMMU_SUPPORT [=y]                                        │  
  │   Selects: IOMMU_API [=y] && IOMMU_IOVA [=n] && NEED_SG_DMA_LENGTH [=y] │  
  │   Selected by: MTK_IOMMU [=n] && IOMMU_SUPPORT [=y] && (ARM [=y] || ARM │  

```

also available in 4.14

frank-w, 2018-01-23:
you can try to enable also MTK_IOMMU, maybe this was enabled in 4.4 and later disabled so the DMA-Option (and IOVA) is selected, but no more the MTK_IOMMU

jfdelnero, 2018-01-23:
[image]
Maybe I am missing something but the IOMMU_DMA selection depend on the MTK_IOMMU activation. And this one shouldn't activated on this card since MTK_IOMMU_V1 replace it.

frank-w, 2018-01-23:
maybe this 2 can be set in config manually...IOMMU_DMA and IOMMU_IOVA are definitely set in 4.4 (loaded menuconfig of 4.4-repo) and only MTK_IOMMU_V1 (not without V1)

i don't really understand the driver-structure, only try to port it from 4.4 to 4.14

jfdelnero, 2018-01-23:
There is something wrong in the iommu 4.14 KConfig. These flags are missing for the MTK_IOMMU_V1 selection versus the v4.4: 

	select COMMON_CLK_MT2701_MMSYS
	select COMMON_CLK_MT2701_IMGSYS
	select COMMON_CLK_MT2701_VDECSYS

frank-w, 2018-01-23:
and all are missing in my current defconfig...can you try to set them and running kernel with them?

jfdelnero, 2018-01-23:
Done but still crash. Something else is missing. I am now checking any possible code change.

jfdelnero, 2018-01-23:
There are tons of changes on these MT drivers... We definitively need something to debug the kernel and get the crash/freeze point to go further...

jfdelnero, 2018-01-23:
After fighting with the early boot printk, here is what happens :

[crash_log.txt](https://github.com/frank-w/BPI-R2-4.14/files/1657755/crash_log.txt)

We are not blind anymore ;)

To enable the earlyprintk : 
-> make menuconfig -> kernel hacking -> early printk

options used in uDev.txt to have the  debugs on the serial port : 

root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
console=ttyS0,115200 earlycon=uart8250,mmio32,0x11004000 earlyprintk=serial,ttyS0,115200 debug=7 loglevel=7 initcall_debug=1 ignore_loglevel=1
bootopts=vmalloc=496M

frank-w, 2018-01-24:
support from Vendor on 4.16: http://forum.banana-pi.org/t/what-s-the-best-practice-to-build-own-kernel/3937/111
i can add you as collaborator so you can add your changes to hdmi-branch

jfdelnero, 2018-01-24:
Thanks. According to the logs + some debug printk i have added, it seems that the device structure passed to the functions is not valid. The attached device drivers is probably not loaded. I will enable the same debugs output on the 4.4.70 to compare the drivers load process.

jfdelnero, 2018-01-24:
Oh wait ! 
https://patchwork.kernel.org/patch/10179801/
this sound to me related ! ( Date Jan. 23, 2018, 8:39 a.m. !)

jfdelnero, 2018-01-24:
Well, after applying the above patch, there is no crash anymore ;).

frank-w, 2018-01-24:
maybe then author of this patch, Yong Wu, can help us getting HDMI working

jfdelnero, 2018-01-24:
Others changes to apply in the dts :
change the larbidx properties name to larb-id into larb declarations.

frank-w, 2018-01-24:
can you make pull-request for your current state (patch, options for defconfig [early printk], larb-changes) to hdmi-branch?

jfdelnero, 2018-01-25:
i have pushed the most important changes. For the debugs output, since this is only some options to enable in the "kernel-hacking" i didn't change the default config file.

jfdelnero, 2018-01-25:
Btw here it is a detailed boot log of the fixed kernel :
[kernel_log.txt](https://github.com/frank-w/BPI-R2-4.14/files/1664203/kernel_log.txt)
the mali, drm and others video stuff seems to start properly.

frank-w, 2018-01-25:
does hdmi working yet?

the log itself sounds that there is something missing (or is this only the "test"-overlay)?
```
[   35.495913] OF: overlay: Failed to apply single node @/testcase-data-2/substation@100/motor-1
[   35.495919] OF: overlay: apply failed '/testcase-data-2/substation@100'
[   35.495929] ### dt-test ### of_overlay_create() (ret=-22), 2
[   35.496238] OF: overlay: Failed to apply prop @/__symbols__/hvac_1
[   35.496243] OF: overlay: apply failed '/__symbols__'
[   35.496253] ### dt-test ### of_overlay_create() (ret=-22), 3
[   35.496259] ### dt-test ### end of unittest - 190 passed, 0 failed
[   35.496276] initcall of_unittest+0x0/0x2b74 returned 0 after 46510 usecs
[   35.496288] calling  tcp_congestion_default+0x0/0x24 @ 1
[   35.496300] initcall tcp_congestion_default+0x0/0x24 returned 0 after 3 usecs
[   35.496308] calling  ip_auto_config+0x0/0x1164 @ 1
[   35.496336] initcall ip_auto_config+0x0/0x1164 returned 0 after 18 usecs
[   35.496345] calling  clear_boot_tracer+0x0/0x3c @ 1
[   35.496356] initcall clear_boot_tracer+0x0/0x3c returned 0 after 0 usecs
[   35.496366] calling  clk_disable_unused+0x0/0x120 @ 1
[   35.497172] initcall clk_disable_unused+0x0/0x120 returned 0 after 775 usecs
[   35.497182] calling  regulator_init_complete+0x0/0x70 @ 1
[   35.497271] vmc: disabling
[   35.497294] vgp1: disabling
[   35.497339] vcamaf: disabling
[   35.497383] initcall regulator_init_complete+0x0/0x70 returned 0 after 185 usecs
```

jfdelnero, 2018-01-25:
no, the hdmi still doesn't work.

yes the above error come from the device tree test. i have enabled it to check any possible error. Seems fine.

frank-w, 2018-01-25:
to bad that in kernel 4.4 it is not clear, what is changed for hdmi because its enabled in the first commit :(

jfdelnero, 2018-01-25:
4.4 boot logs. 
[kernel_log_4_4.txt](https://github.com/frank-w/BPI-R2-4.14/files/1664808/kernel_log_4_4.txt)

frank-w, 2018-01-26:
there are more errors in 4.4 but hdmi is working, right?

jfdelnero, 2018-01-27:
Yes in 4.4 hdmi is working.

Yes the iommu error in the 4.14 is interesting.

frank-w, 2018-01-27:
have you seen that are additional patches on patchwork for m4u?

https://patchwork.kernel.org/project/linux-mediatek/list/

frank-w, 2018-03-03:
just for reference https://github.com/yuq/mesa-lima/issues/20

@jfdelnero any new progress?

frank-w, 2018-06-11:
merged hdmi to 4.16 and 4.14

closed 2018-06-11 by frank-w