frank-w/BPI-Router-Linux #11: Default config kernel 4.14

xbgmsharp, 2018-01-20:
I have compare the default config from kernel 4.4. Here is a diff, you might want to include those on the default config fro kernel 4.14.

Add CONFIG_MTK_PLATFORM="mt7623"
Add CONFIG_CRYPTO_DEV_MEDIATEK=y
NOt really about that one, CONFIG_MEDIATEK_MT6577_AUXADC?

```
$ egrep 'MEDIATEK|MTK' .config | grep -v '#'
CONFIG_ARCH_MEDIATEK=y
CONFIG_PCIE_MEDIATEK=y
CONFIG_ARM_MEDIATEK_CPUFREQ=y
CONFIG_NET_DSA_TAG_MTK=y
CONFIG_MTK_PLATFORM="mt7623"
CONFIG_MTK_BTIF=y
CONFIG_MTK_COMBO=y
CONFIG_MTK_COMBO_CHIP_CONSYS_7623=y
CONFIG_MTK_COMBO_CHIP="CONSYS_7623"
CONFIG_MTK_COMBO_PLAT_PATH=""
CONFIG_MTK_COMBO_COMM=y
CONFIG_MTK_COMBO_COMM_UART=y
CONFIG_MTK_COMBO_COMM_SDIO=y
CONFIG_MTK_COMBO_COMM_NPWR=y
CONFIG_MTK_COMBO_COMM_APO=y
CONFIG_MTK_COMBO_BT=y
CONFIG_MTK_COMBO_BT_HCI=y
CONFIG_MTK_COMBO_WIFI=y
CONFIG_MTK_WAPI_SUPPORT=y
CONFIG_MTK_PASSPOINT_R1_SUPPORT=y
CONFIG_MTK_PASSPOINT_R2_SUPPORT=y
CONFIG_MTK_WIFI_MCC_SUPPORT=y
CONFIG_MTK_DHCPV6C_WIFI=y
CONFIG_AHCI_MTK=y
CONFIG_NET_VENDOR_MEDIATEK=y
CONFIG_NET_MEDIATEK_SOC=y
CONFIG_WLAN_VENDOR_MEDIATEK=y
CONFIG_HW_RANDOM_MTK=y
CONFIG_PINCTRL_MTK=y
CONFIG_MTK_THERMAL=m
CONFIG_MEDIATEK_WATCHDOG=y
CONFIG_IR_MTK=y
CONFIG_DRM_MEDIATEK=y
CONFIG_DRM_MEDIATEK_HDMI=y
CONFIG_SND_SOC_MEDIATEK=y
CONFIG_USB_XHCI_MTK=y
CONFIG_MMC_MTK=y
CONFIG_COMMON_CLK_MEDIATEK=y
CONFIG_MTK_TIMER=y
CONFIG_MTK_IOMMU_V1=y
CONFIG_MTK_INFRACFG=y
CONFIG_MTK_PMIC_WRAP=y
CONFIG_MTK_SCPSYS=y
CONFIG_MTK_SMI=y
CONFIG_MEDIATEK_MT6577_AUXADC=m
CONFIG_PWM_MTK_DISP=m
CONFIG_PWM_MEDIATEK=m
CONFIG_PHY_MTK_TPHY=y
CONFIG_CRYPTO_DEV_MEDIATEK=y
```

Your default:
```
$ egrep 'MEDIATEK|MTK' arch/arm/configs/mt7623n_evb_fwu_defconfig | grep -v '#'
CONFIG_ARCH_MEDIATEK=y
CONFIG_ARM_MEDIATEK_CPUFREQ=y
CONFIG_AHCI_MTK=m
CONFIG_NET_VENDOR_MEDIATEK=y
CONFIG_NET_MEDIATEK_SOC=y
CONFIG_MEDIATEK_WATCHDOG=y
CONFIG_IR_MTK=y
CONFIG_MMC_MTK=y
CONFIG_MTK_IOMMU_V1=y
CONFIG_MTK_PMIC_WRAP=y
CONFIG_MTK_COMBO=y
CONFIG_MTK_COMBO_CHIP_CONSYS_7623=y
CONFIG_MTK_COMBO_COMM=y
CONFIG_MTK_COMBO_WIFI=y
CONFIG_MTK_COMBO_BT=y
CONFIG_MTK_COMBO_BT_HCI=y
CONFIG_PCIE_MEDIATEK=y
CONFIG_PHY_MTK_TPHY=y
CONFIG_PWM_MEDIATEK=m
CONFIG_MTK_THERMAL=m
CONFIG_DRM_MEDIATEK=m
CONFIG_DRM_MEDIATEK_HDMI=m
CONFIG_USB_XHCI_MTK=y
```

As a diff:
```
$ diff myconfig defconfig 
1c1
< CONFIG_AHCI_MTK=y
---
> CONFIG_AHCI_MTK=m
4,8c4,5
< CONFIG_COMMON_CLK_MEDIATEK=y
< CONFIG_CRYPTO_DEV_MEDIATEK=y
< CONFIG_DRM_MEDIATEK_HDMI=y
< CONFIG_DRM_MEDIATEK=y
< CONFIG_HW_RANDOM_MTK=y
---
> CONFIG_DRM_MEDIATEK_HDMI=m
> CONFIG_DRM_MEDIATEK=m
10d6
< CONFIG_MEDIATEK_MT6577_AUXADC=m
13d8
< CONFIG_MTK_BTIF=y
16d10
< CONFIG_MTK_COMBO_CHIP="CONSYS_7623"
18,21d11
< CONFIG_MTK_COMBO_COMM_APO=y
< CONFIG_MTK_COMBO_COMM_NPWR=y
< CONFIG_MTK_COMBO_COMM_SDIO=y
< CONFIG_MTK_COMBO_COMM_UART=y
23d12
< CONFIG_MTK_COMBO_PLAT_PATH=""
26,27d14
< CONFIG_MTK_DHCPV6C_WIFI=y
< CONFIG_MTK_INFRACFG=y
29,31d15
< CONFIG_MTK_PASSPOINT_R1_SUPPORT=y
< CONFIG_MTK_PASSPOINT_R2_SUPPORT=y
< CONFIG_MTK_PLATFORM="mt7623"
33,34d16
< CONFIG_MTK_SCPSYS=y
< CONFIG_MTK_SMI=y
36,39d17
< CONFIG_MTK_TIMER=y
< CONFIG_MTK_WAPI_SUPPORT=y
< CONFIG_MTK_WIFI_MCC_SUPPORT=y
< CONFIG_NET_DSA_TAG_MTK=y
44d21
< CONFIG_PINCTRL_MTK=y
46,47d22
< CONFIG_PWM_MTK_DISP=m
< CONFIG_SND_SOC_MEDIATEK=y
49d23
< CONFIG_WLAN_VENDOR_MEDIATEK=y
```

frank-w, 2018-01-20:
First one is for combo-driver that should be set by kconfig when selecting the mt7623 combo chip (not done yet). I don't want to include it in defconfig to keep that as small as possible.

Second one is for cryptodev and should be selected only if cryptodev is used.

Last one is more interesting because auxadc is for thermal afaik

frank-w, 2018-01-20:
fixed in commit f3d187b

closed 2018-01-20 by frank-w

bburky, 2018-02-06:
It looks good now. I hope you found this useful and made the 4.14.17 update easier.