frank-w/BPI-Router-Linux #18: cryptodev

xbgmsharp, 2018-03-16:
Why is cryptodev store outside of the default modules kernel path?

```
modules/4.14.26-main-02712-g9509a8f0d3c0/kernel/sound/
modules/4.14.26-main-02712-g9509a8f0d3c0/kernel/sound/core/
modules/4.14.26-main-02712-g9509a8f0d3c0/kernel/sound/core/snd-seq-device.ko
modules/4.14.26-main-02712-g9509a8f0d3c0/kernel/sound/core/seq/
modules/4.14.26-main-02712-g9509a8f0d3c0/kernel/sound/core/seq/snd-seq.ko
modules/4.14.26-main-02712-g9509a8f0d3c0/kernel/sound/drivers/
modules/4.14.26-main-02712-g9509a8f0d3c0/kernel/sound/drivers/snd-dummy.ko
modules/4.14.26/
modules/4.14.26/kernel/
modules/4.14.26/kernel/extras/
modules/4.14.26/kernel/extras/cryptodev.ko
```

It lead to confusion.
```
# ll /lib/modules/
total 20
drwxr-xr-x 3 1000 1000 4096 Mar  8 23:12 4.14.26
drwxr-xr-x 3 root root 4096 Mar  7 04:10 4.14.26-main-02704-g192e093549dc
drwxr-xr-x 3 1000 1000 4096 Mar  8 23:12 4.14.26-main-02712-g9509a8f0d3c0
```

MyLive, 2018-03-16:
please use the newest Version an do the following
./build.sh importconfig
./build.sh buid
./build.sh cryptodev

From this Output "bash -x ./build.sh pack" i need the last 40 lines please :-)

frank-w, 2018-03-16:
@xbgmsharp why is your kernel named "4.14.26-main-02712-g9509a8f0d3c0" (why is commit-id appended and another number?), have you compiled with build.sh?

@MyLive: your path here seems to be wrong
https://github.com/frank-w/BPI-R2-4.14/blob/main/build.sh#L119

xbgmsharp's kernel is named "4.14.26-main-02712-g9509a8f0d3c0", cryptodev is copied to "4.14.26"

MyLive, 2018-03-16:
@frank .. i know .. it wars a bug .. i have fixed that ..
I need more Input .. please follow the instruction from my post https://github.com/frank-w/BPI-R2-4.14/issues/18#issuecomment-373617056

 use the newest Version an do the following
./build.sh importconfig
./build.sh buid
./build.sh cryptodev

From this Output "bash -x ./build.sh pack" i need the last 40 lines please :-)

xbgmsharp, 2018-03-16:
I do create the kernel using the build command, ``./build.sh``
I did not run ``./build.sh cryptodev`` but it did compile it.

```
  CC      sound/core/seq/snd-seq.mod.o
  LD [M]  sound/core/seq/snd-seq.ko
  CC      sound/core/snd-seq-device.mod.o
  LD [M]  sound/core/snd-seq-device.ko
  CC      sound/drivers/snd-dummy.mod.o
  LD [M]  sound/drivers/snd-dummy.ko
Image Name:   Linux Kernel 4.14.26-main
Created:      Thu Mar  8 23:12:06 2018
Image Type:   ARM Linux Kernel Image (uncompressed)
Data Size:    6170737 Bytes = 6026.11 kB = 5.88 MB
Load Address: 80008000
Entry Point:  80008000
Build CryptoDev
make -C /lib/modules/4.9.0-4-amd64/build M=/home/xbgmsharp/BPI-R2-4.14/cryptodev/cryptodev-linux ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- clean
make[1]: Entering directory '/usr/src/linux-headers-4.9.0-4-amd64'
make[1]: Leaving directory '/usr/src/linux-headers-4.9.0-4-amd64'
rm -f  *~
CFLAGS= KERNEL_DIR=/lib/modules/4.9.0-4-amd64/build make -C tests clean
make[1]: Entering directory '/home/xbgmsharp/BPI-R2-4.14/cryptodev/cryptodev-linux/tests'
rm -f *.o *~ cipher cipher-aead hmac speed async_cipher async_hmac async_speed sha_speed hashcrypt_speed fullspeed cipher-gcm cipher-aead-srtp cipher_comp hash_comp hmac_comp
make[1]: Leaving directory '/home/xbgmsharp/BPI-R2-4.14/cryptodev/cryptodev-linux/tests'
make -C /home/xbgmsharp/BPI-R2-4.14 M=/home/xbgmsharp/BPI-R2-4.14/cryptodev/cryptodev-linux ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- modules
make[1]: Entering directory '/home/xbgmsharp/BPI-R2-4.14'
  CC [M]  /home/xbgmsharp/BPI-R2-4.14/cryptodev/cryptodev-linux/ioctl.o
  CC [M]  /home/xbgmsharp/BPI-R2-4.14/cryptodev/cryptodev-linux/main.o
  CC [M]  /home/xbgmsharp/BPI-R2-4.14/cryptodev/cryptodev-linux/cryptlib.o
  CC [M]  /home/xbgmsharp/BPI-R2-4.14/cryptodev/cryptodev-linux/authenc.o
  CC [M]  /home/xbgmsharp/BPI-R2-4.14/cryptodev/cryptodev-linux/zc.o
  CC [M]  /home/xbgmsharp/BPI-R2-4.14/cryptodev/cryptodev-linux/util.o
  LD [M]  /home/xbgmsharp/BPI-R2-4.14/cryptodev/cryptodev-linux/cryptodev.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/xbgmsharp/BPI-R2-4.14/cryptodev/cryptodev-linux/cryptodev.mod.o
  LD [M]  /home/xbgmsharp/BPI-R2-4.14/cryptodev/cryptodev-linux/cryptodev.ko
make[1]: Leaving directory '/home/xbgmsharp/BPI-R2-4.14'
build successful
===========================================
1) pack
2) install to SD-Card
choice [12]:1
Pack Kernel to Archive
cleanup...
copy...
INSTALL_MOD_PATH: ../SD/BPI-ROOT/
  INSTALL arch/arm/crypto/aes-arm-bs.ko
  INSTALL arch/arm/crypto/aes-arm-ce.ko
  INSTALL arch/arm/crypto/aes-arm.ko
...
  INSTALL sound/core/seq/snd-seq.ko
  INSTALL sound/core/snd-seq-device.ko
  INSTALL sound/drivers/snd-dummy.ko
  DEPMOD  4.14.26-main-02712-g9509a8f0d3c0
Copy CryptoDev
pack...
-rw-r--r-- 1 xbgmsharp xbgmsharp 59M Mar  8 23:12 /home/xbgmsharp/SD/bpi-r2_4.14.26_main.tar.gz
```

xbgmsharp, 2018-03-16:
After updating the git repository.
```
  INSTALL sound/core/seq/snd-seq.ko
  INSTALL sound/core/snd-seq-device.ko
  INSTALL sound/drivers/snd-dummy.ko
  DEPMOD  4.14.27-main-02858-g2560a36b949d
Copy CryptoDev
depmod: ERROR: could not open directory /home/xbgmsharp/BPI-R2-4.14/../SD/BPI-ROOT//lib/modules/4.14.26-bpi-r2-main: No such file or directory
depmod: FATAL: could not search modules: No such file or directory
pack...
-rw-r--r-- 1 xbgmsharp xbgmsharp 59M Mar  9 05:56 /home/xbgmsharp/SD/bpi-r2_4.14.27_main.tar.gz
```

MyLive, 2018-03-16:
ok .. is there a SD directory in /home/xbgmsharp/ ?

which output did you get for command "ll /home/xbgmsharp/BPI-R2-4.14/../SD/"

xbgmsharp, 2018-03-16:
running with -x
```
+ CRYPTODEV=cryptodev/cryptodev-linux/cryptodev.ko
+ mkdir -p ../SD/BPI-ROOT//etc/modules-load.d
++ rev
++ cut -d/ -f1
++ rev
++ find ../SD/BPI-ROOT/lib/modules/4.14.27-main-02858-g2560a36b949d-dirty -maxdepth 0 -type d
+ LOCALVERSION=4.14.27-main-02858-g2560a36b949d-dirty
+ EXTRA_MODUL_PATH=../SD/BPI-ROOT/lib/modules/4.14.27-main-02858-g2560a36b949d-dirty/kernel/extras
+ '[' -e cryptodev/cryptodev-linux/cryptodev.ko ']'
+ echo Copy CryptoDev
Copy CryptoDev
+ mkdir -p ../SD/BPI-ROOT/lib/modules/4.14.27-main-02858-g2560a36b949d-dirty/kernel/extras
+ cp cryptodev/cryptodev-linux/cryptodev.ko ../SD/BPI-ROOT/lib/modules/4.14.27-main-02858-g2560a36b949d-dirty/kernel/extras
+ echo cryptodev
+ /sbin/depmod -b ../SD/BPI-ROOT/ 4.14.26-bpi-r2-main
depmod: ERROR: could not open directory /home/xbgmsharp/BPI-R2-4.14/../SD/BPI-ROOT//lib/modules/4.14.26-bpi-r2-main: No such file or directory
depmod: FATAL: could not search modules: No such file or directory
+ cp utils/wmt/config/WMT_SOC.cfg ../SD/BPI-ROOT/system/etc/firmware/
+ cp utils/wmt/src/wmt_loader utils/wmt/src/wmt_loopback utils/wmt/src/stp_uart_launcher ../SD/BPI-ROOT/usr/bin/
+ cp -r utils/wmt/firmware/ROMv2_lm_patch_1_0_hdr.bin utils/wmt/firmware/ROMv2_lm_patch_1_1_hdr.bin utils/wmt/firmware/WIFI_RAM_CODE_7623 utils/wmt/firmware/nvram ../SD/BPI-ROOT/etc/firmware/
++ grep CONFIG_MT76= .config
+ [[ -n '' ]]
+ echo pack...
pack...
++ pwd
+ olddir=/home/xbgmsharp/BPI-R2-4.14
+ cd ../SD
+ fname=bpi-r2_4.14.27_main.tar.gz
+ tar -cz --owner=root --group=root -f bpi-r2_4.14.27_main.tar.gz BPI-BOOT BPI-ROOT
+ md5sum bpi-r2_4.14.27_main.tar.gz
++ pwd
+ ls -lh /home/xbgmsharp/SD/bpi-r2_4.14.27_main.tar.gz
-rw-r--r-- 1 xbgmsharp xbgmsharp 59M Mar  9 06:18 /home/xbgmsharp/SD/bpi-r2_4.14.27_main.tar.gz
+ cd /home/xbgmsharp/BPI-R2-4.14
```

frank-w, 2018-03-16:
updated and all-pack

```
  DEPMOD  4.14.27-bpi-r2-main
Copy CryptoDev
depmod: ERROR: could not open directory /media/data_ext/bpi-r2-kernel/github/../SD/BPI-ROOT//lib/modules/4.14.26-bpi-r2-main: No such file or directory
depmod: FATAL: could not search modules: No such file or directory

```

MyLive, 2018-03-16:
Thanks, i have found the Bug.
For testing i´ve used a static var and i had forgotten to delete it.

Please update and retest :-)

frank-w, 2018-03-16:
still uses my local kernel instead of the kernel-dir

problem is present if using the cryptodev/build.sh

frank-w, 2018-03-16:
```
diff --git a/cryptodev/cryptodev-linux/Makefile b/cryptodev/cryptodev-linux/Makefile
index 5a080e0..890344a 100644
--- a/cryptodev/cryptodev-linux/Makefile
+++ b/cryptodev/cryptodev-linux/Makefile
@@ -5,7 +5,7 @@
 #
 CRYPTODEV_CFLAGS ?= #-DENABLE_ASYNC
 KBUILD_CFLAGS += -I$(src) $(CRYPTODEV_CFLAGS)
-KERNEL_DIR ?= /lib/modules/$(shell uname -r)/build
+#KERNEL_DIR ?= /lib/modules/$(shell uname -r)/build
 VERSION = 1.9
 
 prefix ?= /usr/local
```

so it works for me...

imho ?= should set the value if it is not set by environment-var, but here it does not work...maybe my make (GNU Make 3.81) is too old?

frank-w, 2018-03-16:
manually copied kernel+modules including cryptodev in kernel/extras to my sd-card booted up and loaded cryptodev....

root@bpi-r2:~# modprobe cryptodev
[  182.615751] cryptodev: loading out-of-tree module taints kernel.
[  182.623317] cryptodev: driver 1.9 loaded.

maybe cryptodev should be copied to another location

xbgmsharp, 2018-03-16:
Thanks work as expected.
cryptodev is not build by default.
```
$ ./build.sh cryptodev
$ ./build.sh pack
```

xbgmsharp, 2018-03-17:
All good, thanks work as expected.
```
# ll /lib/modules/4.14.27-main-02862-g7a28948f2bde/kernel/extras/
total 1048
-rw-r--r-- 1 1000 1000 1072912 Mar  9 10:20 cryptodev.ko
# modprobe cryptodev
# lsmod 
Module                  Size  Used by
cryptodev              53248  0
...
# ll /dev/crypto 
crw-rw-rw- 1 root root 10, 57 Mar 17 09:06 /dev/crypto
```

closed 2018-03-17 by xbgmsharp