frank-w/BPI-Router-Linux #21: Uboot

BGFWDev, 2018-05-09:
IS there a newer 2018.03 Uboot available for MT7623?

chwe17, 2018-05-26:
less an issue more an enhancement... I forked your u-boot and adjusted it to my needs... The full bootmenu is removed (IMO useless) and only for SD-Card (yet). It tries to boot with a bootscript sitting in mmc 1:1 /boot/boot.scr and has some fallbacks in case there is no bootscript (or execution fails). Since I don't use 'appended device tree' and uImages anymore. You might adjust some variables and bootcommands for your usage (or, you change to flattened devicetree too.. ;) it works  reliable and IMO more flexible in case you start to play with devicetree too).

https://github.com/chwe17/u-boot-mt

frank-w, 2018-05-26:
Hi,
i dont want to change uImage itself (remove appended dtb)...is your bootmenu dynamic (configurable via uenv.txt)?

what do i have to change to use existing uImages?

chwe17, 2018-05-26:
there is no bootmenu anymore... :P I don't like it, I don't use it. This part here is more or less relevant for you: https://github.com/chwe17/u-boot-mt/blob/f5916a4c6d0ad8acf0e9b8fe3f649425272f5e6a/include/configs/mt7623-bpi-r2.h#L361-L413
First it tries to boot with a boot script sitting on mmc 1:1 /boot/boot.scr and in case this doesn't work it falls back to some defaults. Which you have to adjust due to Sinovoips buildscript expect different bootparameters than (e.g. I don't use FAT and rootfs is in root=/dev/mmcblk0p). But this is only for the fallbacks in case something goes wrong with the bootscript where you can define whatever you want for your boot. e.g. 
```
setenv root "/dev/mmcblk0p1"
setenv rootfs "ext4"
setenv verbosity "1"
setenv bootargs initcall_debug console=ttyS0,115200n1 root=${root} rw rootfstype=${rootfs} rootwait audit=0 loglevel=${verbosity}
ext4load mmc ${mmcnum}:${mmcpart} ${fdtaddr} ${mmcfdtfile}
ext4load mmc ${mmcnum}:${mmcpart} ${rdaddr} ${mmcinitrdfile}
ext4load mmc ${mmcnum}:${mmcpart} ${kernel_addr_r} ${mmckernfile}
echo "Booting ${mmckernfile} ${mmcinitrdfile} ${mmcfdtfile} from: mmc ${mmcnum}:${mmcpart} using bootargs=${bootargs}"
bootz ${kernel_addr_r} ${rdaddr} ${fdtaddr}
# Recompile with:
# mkimage -C none -A arm -T script -d /boot/boot.cmd /boot/boot.scr
```
That's the current bootscript I use. So you can easy adjust it to a different kernel without recompile the whole u-boot. You would need to adjust the fallbacks to a known working kernel (and it's location) and for testings you can adjust the bootscript (can be done on your R2 as long as u-boot-tools are installed or you mount the SD card on your normal computer and 'recompile' boot.cmd with the parameters in the end of the bootscript. uImages with 'appended devicetree' should still work but zImages need flattened devicetree as long as #define CONFIG_OF_LIBFDT is set (https://github.com/chwe17/u-boot-mt/blob/f5916a4c6d0ad8acf0e9b8fe3f649425272f5e6a/include/configs/mt7623-bpi-r2.h#L296)

The bootscript is normally used things like network boot but it's IMO very useful for everything else. A small comparison:
https://stackoverflow.com/questions/28891221/uenv-txt-vs-boot-scr

chwe17, 2018-05-27:
chances are rather low that someone port to 'mainline u-boot' soon (I didn't see any attempts on mailinglist). I'm sure that you can load a uenv with a bootscript.. :P Just look through various bootscripts in armbians project: https://github.com/armbian/build/tree/master/config/bootscripts
Cause, actually both is used. I would look into those where a bit older bootloaders are used. E.g. odroid c2, uses at the moment a 2015 version (switch to upstream u-boot is in testing). 

> and/or activate hdmi in uboot (currently only purple screen)

No, I only 'fixed' u-boot to a point where it works 'reliable' with bootscripts ('recompilation' of a bootscript means only adding a valid header - done in seconds). This (https://github.com/wtolkien/meta-mediatek/tree/master/recipes-bsp/u-boot) might also give you some inspiration, cause Wolfgang uses uEnv.txt files too. Since there's no HDMI driver, or at least, none I can see in MT7623 related files, this will probably be something which isn't solve that fast (there are a few boards with supported HDMI with u-boot 2014.4 but for sure over my skills to implement such a driver, and also of minor interest yet, cause the R2 will run headless anyway for my purposes).

For debug the Kernel (some enhancements which are not/wrong set by your defconfig):
- earlyprintk (not set by default in your kernelconfigs with defconfig)
- CONFIG_DEBUG_UART_PHYS=0x1100**4**000 CONFIG_DEBUG_UART_VIRT=0xf100**4**000 whereas your defconfig sets it to 0x1100**2**000 (UART0)
- CONFIG_REGULATOR_FIXED_VOLTAGE=y is not set, and therefore USB spies you some errors in dmesg (xhci-mtk 1a1c0000.usb: fail to get vbus)

In case you're interested you can follow my progress [here](https://forum.armbian.com/topic/7296-bananapi-r2-csc/) (network performance is horrible at the moment, probably due to wrong settings).

frank-w, 2018-10-05:
Mainline-uboot prepared for bpi-r2 is here: https://github.com/frank-w/u-boot

Currently without saveenv and ethernet-support

chwe17, 2018-10-05:
great work! I'll probably use some of your commits in patchform as well. Cause BPi Images have some different needs compared to Armbians needs I'll keep it on top of mainline. But your uEnv.txt may give me some inspiration. :)

frank-w, 2018-12-21:
just to make an update...

bpi-r2 is now supported by upstream uboot and ethernet-patches are out too

i merged both to my uboot-repo:

https://github.com/frank-w/u-boot/

(currently https://github.com/frank-w/u-boot/tree/bpi-r2_eth_cve has all actual patches)

closed 2018-12-21 by frank-w