frank-w/BPI-Router-Linux #38: Issues with gcc-arm-linux-gnueabihf package

leifkotsch, 2018-07-29:
Hello Frank,

I've got some trouble installing the "gcc-arm-linux-gnueabihf" packet via apt.
I also tried manualy downloading the packet and installing it with dpkg but it says I've got the wrong architecture.
If it helps: uname -a output: "Linux server-00-bpi 4.4.70-BPI-R2-Kernel #1 SMP Tue Nov 28 10:59:01 CST 2017 armv7l armv7l armv7l GNU/Linux"

Regards

frank-w, 2018-07-29:
You try to compile on r2 directly? Then you do not need this package...try to remove it from script

leifkotsch, 2018-07-29:
Ok. Well, I compiled the kernel on my BPI2 and received a "bananapi-r2-image-4.14-main_4.14.59-1_armhf.deb" package. I tried to install it via dpkg --install, it succeeded, but I'm still on the old kernel after a reboot.

frank-w, 2018-07-29:
Please check with "mount"  if /boot is your mounted boot-partition or simply a folder. Official images do not mount /boot and uboot did not found your kernel because it lay on BPI-ROOT/boot instead of BPI-BOOT

I'm also not sure if postinst works (add your new kernel to uenv.txt)

leifkotsch, 2018-07-29:
There is nothing mounted to /boot.
While compiling kernel the script said that there's a problem with /boot, after I did choose to install the kernel directly.
I'm not sure what to do now.

frank-w, 2018-07-29:
- rename /boot to any other
- create an empty /boot
- mount /dev/mmcblk0p1 /boot
try again install directly

leifkotsch, 2018-07-29:
I just receive:
root@server-00-bpi:~# sudo dpkg --install /home/pi/bpi-r2-4.14/debian/bananapi-r2-image-4.14-main_4.14.59-1_armhf.deb
(Reading database ... 252129 files and directories currently installed.)
Preparing to unpack .../bananapi-r2-image-4.14-main_4.14.59-1_armhf.deb ...
Unpacking bananapi-r2-image-4.14-main (4.14.59-1) over (4.14.59-1) ...
dpkg: error processing archive /home/pi/bpi-r2-4.14/debian/bananapi-r2-image-4.14-main_4.14.59-1_armhf.deb (--install):
 error setting ownership of './boot/bananapi/bpi-r2/linux/uImage_4.14.59-main': Operation not permitted
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
/var/lib/dpkg/tmp.ci/postrm: 7: /var/lib/dpkg/tmp.ci/postrm: Syntax error: end of file unexpected (expecting ")")
dpkg: error while cleaning up:
 subprocess new post-removal script returned error exit status 2
Errors were encountered while processing:
 /home/pi/bpi-r2-4.14/debian/bananapi-r2-image-4.14-main_4.14.59-1_armhf.deb

frank-w, 2018-07-29:
as i said, debian-package is experiemental...i try a bit to get the maintainer-scripts running for complete install...

previous version has only unpacked the files...in my local repo i fixed the syntax-error, but postinst-script seems not to be called

leifkotsch, 2018-07-29:
Is there another way to get the 4.14 Kernel working? I did try the precompiled 4.14 kernel, but my bpi wont boot after copying the files from the archive.

frank-w, 2018-07-29:
you can use build.sh to install directly on sd-card
if using the precompiled images you have to unpack the files to the BPI-OOT-folder to BPI-BOOT-Partition and from inside BPI-ROOT to BPI-ROOT

leifkotsch, 2018-07-29:
The installation process succeeded, but now i got the same problem like with the precompiled kernel. The board isn't booting. The red lights lights for about 15 seconds and then turns off and nothing happens. Is there any way to fix that?

frank-w, 2018-07-29:
if you have a usb2serial-cable you can connect to the board through debug-uart and look with minicom/putty what happens

https://www.fw-web.de/dokuwiki/doku.php?id=en/bpi-r2/debug-uart

frank-w, 2018-07-29:
Btw if you switch from 4.4 to 4.14 you have to change network-config because of dsa-driver in 4.14.

leifkotsch, 2018-07-29:
In what way? In the /etc/network/infterfaces file?

frank-w, 2018-07-29:
Right. In my wiki you'll find an example https://www.fw-web.de/dokuwiki/doku.php?id=en:bpi-r2:network:start#permanent

but this should not poweroff the r2...it is simply not reachable because ip-config in 4.4 is done on ethx and in 4.14 on wan/lanx

frank-w, 2018-07-30:
btw. Green and blue near gpio go off this is right,red should stay on...4.4 and mainline-kernel have a bug where thiis leds are stay on at boot which i have fixed

leifkotsch, 2018-07-30:
Looking at the UART output, the BPI seems to boot correctly but fails to start Raise network interfaces.
It might be the /etc/network/interfaces file.

frank-w, 2018-07-30:
Then just use the content from my wiki. simply replace ip/netmasq/gateway to your settings and don't forget config for eth0/eth1 these have not to be changed

closed 2018-07-30 by leifkotsch