frank-w/BPI-Router-Linux #39: Kernel 4.14 deb-script problems

leifkotsch, 2018-07-30:
I compiled the latest 4.14.59 Kernel and selected 3 to get a .deb package, but while I tried to install it on my BananaPi I received this exception:
`root@server-00-bpi:~# dpkg --install bananapi-r2-image-4.14-main_4.14.59-1_armhf.deb`
`(Reading database ... 251892 files and directories currently installed.)`
`Preparing to unpack bananapi-r2-image-4.14-main_4.14.59-1_armhf.deb ...`
`Unpacking bananapi-r2-image-4.14-main (4.14.59-1) ...`
`dpkg: error processing archive bananapi-r2-image-4.14-main_4.14.59-1_armhf.deb (--install):`
 `error setting ownership of './boot/bananapi/bpi-r2/linux/uImage_4.14.59-main': Operation not permitted`
`dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)`
`installation aborted`

frank-w, 2018-07-31:
i got this error only if this file (uimage_x) exists (previout install)

leifkotsch, 2018-07-31:
I now used direct install. That works fine. While direct install the script prompted the name of the uImage: The default was set to uImage_4.14.59-main, but I had to use uImage. Perhaps here thats the same problem: That the "./boot/bananapi/bpi-r2/linux/uImage_4.14.59-main" file is instead named "./boot/bananapi/bpi-r2/linux/uImage"

leifkotsch, 2018-08-01:
There is 45.9% free space on /boot.
My base image is one of the first given images from the bpi-forum. Ubuntu 16.04.

frank-w, 2018-08-01:
these files are annoying on official images, right. you'll find also modules on rootfs for these kernels and log-entries not matching r2. i have removed unneeded files from /boot in my images (debian 9 and ubuntu 18.04) and rootfs is cleaned before unpack bootstrapped fs.

uboot loads bananapi/bpi-r2/linux/uEnv.txt after selecting the default menuentry (boot from SD) and execute load kernel file in same directory defined by variable "kernel". all other dirs/files on /boot are unnecessary. my deb copies the uImage-file to that dir and make an entry in uenv.txt (override this kernel-var).
i do not set any permissions/ownership manually, but dpkg seems to set them by default, which failes with your dpkg-version on fat (there is not owner/permission possible).

i will try to write same image as you and test it there.

frank-w, 2018-08-02:
i have downloaded this http://forum.banana-pi.org/t/bpi-r2-new-image-ubuntu-16-04-v1-2-1-bt-and-wifi-ap-mode-are-working-fine-2017-11-27/4291 and installed to my sd-card, but /boot does not contain uImages for other boards, so i assume, that is not the image you use...can you point me to the right one?

leifkotsch, 2018-08-02:
I think it should be this one: forum.banana-pi.org/t/bpi-r2-new-image-ubuntu-16-04-mate-desktop-docker-preview2-bpi-r2-sd-emmc-img-2017-08-01/3615

frank-w, 2018-08-03:
i have reproduced the error with 2017-11-28-ubuntu-16.04-mate-desktop-bpi-r2-sd-emmc.img too, but it seems to be a dpkg-problem, because it works with ubuntu 18.4 and i have not found an option to ignore the kernelfile for ownership/permission setting (will be done automaticly). i have also found no way to mount the /boot as root (it always get uid=1000,gid=1000, so all files are owned by user pi)

```
rootfs=$(mount | grep 'on / '| awk '{print $1}')
mount ${rootfs:0:-1}1 /boot
/dev/mmcblk0p1 on /boot type vfat (rw,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro)
mount ${rootfs:0:-1}1 /boot -o remount,rw,uid=0,gid=0
/dev/mmcblk0p1 on /boot type vfat (rw,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro)
```

leifkotsch, 2018-11-04:
I usually install the new kernel directly on /boot.
That works for me very well.