frank-w/BPI-Router-Linux #46: When building I get a cryptodev error...

whalley, 2018-09-16:
I'm a noob to this so forgive a simple question.

```
Build CryptoDev
make -C  M=/home/mark/banana-pi/BPI-R2-4.14/cryptodev/cryptodev-linux ARCH=arm CROSS_COMPILE=ccache arm-linux-gnueabihf- clean
make[1]: *** M=/home/mark/banana-pi/BPI-R2-4.14/cryptodev/cryptodev-linux: No such file or directory.  Stop.
Makefile:39: recipe for target 'clean' failed
make: *** [clean] Error 2
make -C /home/mark/banana-pi/BPI-R2-4.14 M=/home/mark/banana-pi/BPI-R2-4.14/cryptodev/cryptodev-linux ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- modules
```

Any pointers?

Mark

whalley, 2018-09-16:
Thanks for quick response, appreciated. My board should arrive in next week or so, I'm just getting prepared :-)

I am actually building 4.14-main (based on this looking like to most compatible build feature-wise) - looks like it the clean throwing the error as you noted.

frank-w, 2018-09-16:
If building 4.14-main,the error should be different.

make[1]: *** M=/home/mark/banana-pi/BPI-R2-4.14/cryptodev/cryptodev-linux: No such file or directory. Stop.

The directory itself should exist

https://github.com/frank-w/BPI-R2-4.14/tree/4.14-main/cryptodev/cryptodev-linux

The error should say that files cannot be removed because they does not exist if not previous compiled. But as i said if you does not need cryptodev you can ignore that

whalley, 2018-09-16:
Don't think I need the crypto... but it was niggling me....

In the cryptodev/build.sh

should:
```
make clean
make KERNEL_DIR=$kerneldir CROSS_COMPILE=arm-linux-gnueabihf- ARCH=arm
```

not be:
```
make KERNEL_DIR=$kerneldir clean
make KERNEL_DIR=$kerneldir CROSS_COMPILE=arm-linux-gnueabihf- ARCH=arm
```

This allows me the build it cleanly in the cryptodev directory.

But building the full kernel now gives:

```
Build CryptoDev
make -C /home/mark/banana-pi/BPI-R2-4.14 M=/home/mark/banana-pi/BPI-R2-4.14/cryptodev/cryptodev-linux ARCH=arm CROSS_COMPILE=ccache arm-linux-gnueabihf- clean
make[1]: Entering directory '/home/mark/banana-pi/BPI-R2-4.14'
./scripts/gcc-version.sh: line 26: ccachegcc: command not found
./scripts/gcc-version.sh: line 27: ccachegcc: command not found
make[1]: ccachegcc: Command not found
make[1]: ccachegcc: Command not found
make[1]: *** No rule to make target 'arm-linux-gnueabihf-'.  Stop.
make[1]: Leaving directory '/home/mark/banana-pi/BPI-R2-4.14'
Makefile:39: recipe for target 'clean' failed
make: *** [clean] Error 2
make -C /home/mark/banana-pi/BPI-R2-4.14 M=/home/mark/banana-pi/BPI-R2-4.14/cryptodev/cryptodev-linux ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- modules
make[1]: Entering directory '/home/mark/banana-pi/BPI-R2-4.14'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory
```

I'm not sure why the 
`ARCH=arm CROSS_COMPILE=ccache arm-linux-gnueabihf-`

is being added to the clean in this case!

frank-w, 2018-09-16:
my script does not add it

https://github.com/frank-w/BPI-R2-4.14/blob/902d112b8a7cf645e40d2aeeadedc2403dd84aee/cryptodev/build.sh#L7

it seems it is added by Makefile

https://github.com/frank-w/BPI-R2-4.14/blob/902d112b8a7cf645e40d2aeeadedc2403dd84aee/cryptodev/cryptodev-linux/Makefile#L39

whalley, 2018-09-17:
Yes, the Makefile should be:

`$(MAKE) -C $(KERNEL_DIR) M=$(CURDIR) clean`

That, and the change to the build.sh to add the KERNEL_DIR works. Thanks for the pointer.

frank-w, 2018-09-17:
`$(MAKE) -C $(KERNEL_DIR) M=$(CURDIR) clean`
does also not work...clean seems to treat M=... as dir not as var and tries to access it...

whalley, 2018-09-17:
Have you also make the change to cryptodev/build.sh I mentioned above ?

In the cryptodev/build.sh

```
make clean
make KERNEL_DIR=$kerneldir CROSS_COMPILE=arm-linux-gnueabihf- ARCH=arm
```

should be:

```
make KERNEL_DIR=$kerneldir clean
make KERNEL_DIR=$kerneldir CROSS_COMPILE=arm-linux-gnueabihf- ARCH=arm
```

frank-w, 2018-09-17:
right, i only hae to change cryptodev/build.sh and it works if i call it alone....in main build.sh i got the mentioned error where ccache and crosscompiler and concatenated

whalley, 2018-09-17:
Yes you need to change the Makefile to: 

`$(MAKE) -C $(KERNEL_DIR) M=$(CURDIR) clean`

to fix the errors

frank-w, 2018-09-17:
see the last commit...works here...

whalley, 2018-09-17:
Yep, all good. Thanks.

closed 2018-09-17 by whalley