frank-w/BPI-Router-Linux #52: Wrong DTS reference

alxchk, 2018-11-19:
https://github.com/frank-w/BPI-R2-4.14/blob/ee7516478dafaa8ccf7f6e763d88a8e03758af37/drivers/crypto/mediatek/mtk-platform.c#L581

https://github.com/frank-w/BPI-R2-4.14/blob/4.14-main/arch/arm/boot/dts/mt7623.dtsi#L1222

frank-w, 2018-11-20:
thank you, can you verify it works now?

frank-w, 2018-11-20:
which config relies to it? havn't used crypto yet...

`CONFIG_CRYPTO_DEV_MEDIATEK=y`

?

alxchk, 2018-11-20:
Yes. I suggest to build this as module to be able to check the difference (it's very unexpected, but anyway..)

frank-w, 2019-01-28:
```
[08:58] frank@bpi-r2-e:~$ grep mtk /proc/crypto
driver       : mtk-hmac-sha512
driver       : mtk-hmac-sha384
driver       : mtk-sha512
driver       : mtk-sha384
driver       : mtk-hmac-sha256
driver       : mtk-hmac-sha224
driver       : mtk-hmac-sha1
driver       : mtk-sha256
driver       : mtk-sha224
driver       : mtk-sha1
driver       : gcm-aes-mtk
driver       : ctr-aes-mtk
driver       : ecb-aes-mtk
driver       : cbc-aes-mtk
[08:58] frank@bpi-r2-e:~$
```
with actual 4.14...this is how it should look,right? How to test performance (with/without)?

alxchk, 2019-01-28:
You can check scripts in my fork of your fork :)

Take actual cryptodev, and.or compile af_alg to the kernel. Then recompile openssl with cryptodev. They do have support for cryptodev for BSD, so you need manually add CFLAGS for that. In recent debian (testing) af_alg is present in the default package.

Then you can use openssl speed for aes/sha. This unit has nice performance, but as with any hw module you should pass a lot there to have nice speed. In other case performance will be worse because of DMI interrupts, context switches etc.

alxchk, 2019-01-28:
Also there are some stability issues with it. Maybe race conditions in DMA handler, or god knows where. To find this, you need load your system with something with interrupts. Run several benchmarks, then fill ram with /dev/urandom to make it swap, use iperf to load network and something with SATA. Likely in 10-20 mins something will die.

frank-w, 2019-10-20:
Have you found out more?

closed 2023-03-04 by frank-w