frank-w/BPI-Router-Linux #62: Network and filesystem unstable

Josue-T, 2019-04-15:
Since the version ~4.14.81-82 I have some some unstability.

About the network I get at the boot some errors like that:
```
Apr 15 21:01:38 the_hostname kernel: [   45.246392] Hardware name: Mediatek Cortex-A7 (Device Tree)
Apr 15 21:01:38 the_hostname kernel: [   45.246436] [<c01132d0>] (unwind_backtrace) from [<c010d5e0>] (show_stack+0x20/0x24)
Apr 15 21:01:38 the_hostname kernel: [   45.246453] [<c010d5e0>] (show_stack) from [<c0aa8544>] (dump_stack+0x90/0xa4)
Apr 15 21:01:38 the_hostname kernel: [   45.246465] [<c0aa8544>] (dump_stack) from [<c023c46c>] (warn_alloc+0xc8/0x17c)
Apr 15 21:01:38 the_hostname kernel: [   45.246476] [<c023c46c>] (warn_alloc) from [<c023d7bc>] (__alloc_pages_nodemask+0x11e8/0x122c)
Apr 15 21:01:38 the_hostname kernel: [   45.246485] [<c023d7bc>] (__alloc_pages_nodemask) from [<c023d9ec>] (page_frag_alloc+0x14c/0x160)
Apr 15 21:01:38 the_hostname kernel: [   45.246497] [<c023d9ec>] (page_frag_alloc) from [<c089240c>] (netdev_alloc_frag+0x40/0x48)
Apr 15 21:01:38 the_hostname kernel: [   45.246511] [<c089240c>] (netdev_alloc_frag) from [<c0717ab0>] (mtk_rx_alloc+0xac/0x34c)
Apr 15 21:01:38 the_hostname kernel: [   45.246522] [<c0717ab0>] (mtk_rx_alloc) from [<c07180f4>] (mtk_open+0x3a4/0x5f8)
Apr 15 21:01:38 the_hostname kernel: [   45.246532] [<c07180f4>] (mtk_open) from [<c08aefe4>] (__dev_open+0xb8/0x11c)
Apr 15 21:01:38 the_hostname kernel: [   45.246543] [<c08aefe4>] (__dev_open) from [<c08af3a8>] (__dev_change_flags+0x184/0x1ac)
Apr 15 21:01:38 the_hostname kernel: [   45.246553] [<c08af3a8>] (__dev_change_flags) from [<c08af3f8>] (dev_change_flags+0x28/0x58)
Apr 15 21:01:38 the_hostname kernel: [   45.246564] [<c08af3f8>] (dev_change_flags) from [<c08c2658>] (do_setlink+0x32c/0x968)
Apr 15 21:01:38 the_hostname kernel: [   45.246575] [<c08c2658>] (do_setlink) from [<c08c4898>] (rtnl_newlink+0x4d0/0x7fc)
Apr 15 21:01:38 the_hostname kernel: [   45.246585] [<c08c4898>] (rtnl_newlink) from [<c08c1ae8>] (rtnetlink_rcv_msg+0x218/0x280)
Apr 15 21:01:38 the_hostname kernel: [   45.246595] [<c08c1ae8>] (rtnetlink_rcv_msg) from [<c08eaef8>] (netlink_rcv_skb+0xd0/0xf0)
Apr 15 21:01:38 the_hostname kernel: [   45.246606] [<c08eaef8>] (netlink_rcv_skb) from [<c08c18cc>] (rtnetlink_rcv+0x20/0x24)
Apr 15 21:01:38 the_hostname kernel: [   45.246615] [<c08c18cc>] (rtnetlink_rcv) from [<c08ea648>] (netlink_unicast+0x190/0x258)
Apr 15 21:01:38 the_hostname kernel: [   45.246624] [<c08ea648>] (netlink_unicast) from [<c08eaa68>] (netlink_sendmsg+0x280/0x350)
Apr 15 21:01:38 the_hostname kernel: [   45.246634] [<c08eaa68>] (netlink_sendmsg) from [<c088a024>] (sock_sendmsg+0x24/0x34)
Apr 15 21:01:38 the_hostname kernel: [   45.246642] [<c088a024>] (sock_sendmsg) from [<c088a7e4>] (___sys_sendmsg+0x1ec/0x214)
Apr 15 21:01:38 the_hostname kernel: [   45.246651] [<c088a7e4>] (___sys_sendmsg) from [<c088b674>] (__sys_sendmsg+0x50/0x7c)
Apr 15 21:01:38 the_hostname kernel: [   45.246660] [<c088b674>] (__sys_sendmsg) from [<c088b6b8>] (SyS_sendmsg+0x18/0x1c)
Apr 15 21:01:38 the_hostname kernel: [   45.246671] [<c088b6b8>] (SyS_sendmsg) from [<c0108cc0>] (ret_fast_syscall+0x0/0x54)
```

And about the (ext4) filesystem (with the sata port) for some file when I try to read it I get this:
```
ls: impossible d'ouvrir le répertoire '/usr/local/lib/python3.5/dist-packages/pip': Ne peut allouer de la mémoire
d????????? ? ?    ?        ?            ? pip
```
And in the kernel log I get this:
```
Apr 15 21:08:03 the_hostname kernel: [  399.051628] grow_buffers: requested out-of-range block 16879718369374422450 for device sda1
```
With the all new kernel version (4.14 and 4.19) I have theses error. The only version wich run cleanly is <= 4.14.67. Since the version 4.14.74 I detected some issue about the network. And between the version 4.14.81 and the version 4.14.82 I get the filesystem problem.

I think that it's due to a service which generate this unstability because with a clean rootfs I don't get theses problem.

I give you my boot sequence (of a unstable kernel version if it could help you):
```
[    5.181665] mtk-msdc 11230000.mmc: GPIO lookup for consumer wp
[    5.187468] mtk-msdc 11230000.mmc: using device tree for GPIO lookup
[    5.193812] of_get_named_gpiod_flags: can't parse 'wp-gpios' property of node '/mmc@11230000[0]'
[    5.202561] of_get_named_gpiod_flags: can't parse 'wp-gpio' property of node '/mmc@11230000[0]'
[    5.211217] mtk-msdc 11230000.mmc: using lookup tables for GPIO lookup
[    5.217700] mtk-msdc 11230000.mmc: lookup for GPIO wp failed
[    5.267609] mmc0: host does not support reading read-only switch, assuming write-enable
[    5.277724] mmc0: new high speed SDHC card at address 0001
[    5.284266] of_get_named_gpiod_flags: parsed 'gpios' property of node '/leds/green[0]' - status (0)
[    5.293314] no flags found for gpios
[    5.296975] of_get_named_gpiod_flags: parsed 'gpios' property of node '/leds/blue[0]' - status (0)
[    5.305909] no flags found for gpios
[    5.309554] of_get_named_gpiod_flags: parsed 'gpios' property of node '/leds/red[0]' - status (0)
[    5.318396] no flags found for gpios
[    5.322373] mmcblk0: mmc0:0001 00000 29.3 GiB 
[    5.328881]  mmcblk0: p1 p2 p3
[    5.332415] hidraw: raw HID events driver (C) Jiri Kosina
[    5.337918] usbcore: registered new interface driver usbhid
[    5.343479] usbhid: USB HID core driver
[    5.348406] mtk-iommu-v1 10205000.mmsys_iommu: bound 14010000.larb (ops mtk_smi_larb_component_ops)
[    5.357426] mtk-iommu-v1 10205000.mmsys_iommu: bound 16010000.larb (ops mtk_smi_larb_component_ops)
[    5.366431] mtk-iommu-v1 10205000.mmsys_iommu: bound 15001000.larb (ops mtk_smi_larb_component_ops)
[    5.376452] NET: Registered protocol family 17
[    5.381428] NET: Registered protocol family 41
[    5.385959] 8021q: 802.1Q VLAN Support v1.8
[... 368 lines cut ...]
[   44.998861] br0: port 2(lan1) entered disabled state
[   44.999168] device lan1 entered promiscuous mode
[   45.012090] br0: port 3(lan2) entered blocking state
[   45.012100] br0: port 3(lan2) entered disabled state
[   45.012401] device lan2 entered promiscuous mode
[   45.026435] br0: port 4(lan3) entered blocking state
[   45.026444] br0: port 4(lan3) entered disabled state
[   45.026743] device lan3 entered promiscuous mode
[   45.054073] IPv6: ADDRCONF(NETDEV_UP): br0: link is not ready
[  OK  ] Started System Logging Service.
[  OK  ] Started LSB: Start/stop sysstat's sadc.
[  OK  ] Started LSB: Add IP and MAC in /etc/issue.
[  OK  ] Started LSB: Load kernel modules needed to enable cpufreq scaling.
[  OK  ] Started LSB: Start the GNUstep distributed object mapper.
[  OK  ] Started LSB: Provide limited super user privileges to specific users.
[  OK  ] Started LSB: Demon mouse.
[  OK  ] Started Avahi mDNS/DNS-SD Stack.
         Starting LSB: set CPUFreq kernel parameters...
[  OK  ] Started Login Service.
[  OK  ] Started LSB: service and resource monitoring daemon.
[  OK  ] Started LSB: set CPUFreq kernel parameters.
[FAILED] Failed to start Raise network interfaces.
See 'systemctl status networking.service' for details.
[  OK  ] Reached target Network.
```

frank-w, 2019-04-15:
i did not see this network-instability before (have also 4.14 running and look daily in my dmesg)...
crash seem to happen in mtk_open

strange is this callstack:

```
Apr 15 21:01:38 the_hostname kernel: [   45.246522] [<c0717ab0>] (mtk_rx_alloc) from [<c07180f4>] (mtk_open+0x3a4/0x5f8)
Apr 15 21:01:38 the_hostname kernel: [   45.246532] [<c07180f4>] (mtk_open) from [<c08aefe4>] (__dev_open+0xb8/0x11c)
```

because in mtk_open there is no call to mtk_rx_alloc

https://github.com/frank-w/BPI-R2-4.14/blob/4.14-main/drivers/net/ethernet/mediatek/mtk_eth_soc.c#L1865

the only call to mtk_rx_alloc is in mtk_dma_init which is called by mtk_start_dma, so it seems that is an initialisation issue and should happen only once at bootup.

https://github.com/frank-w/BPI-R2-4.14/blob/4.14-main/drivers/net/ethernet/mediatek/mtk_eth_soc.c#L1717

there was a network-problem imho between 4.14.92-98 but this is fixed. which is the last kernel-version you've tried?

have you any special network-setting? bridge, vlan, ... any way to reproduce the problem?

how about 4.19, here i have not added QDMA/BQL-Patches...only the second gmac alone (dsa-driver). you can revert it easier to check if it is the problem.

have you tried 5.0/5.1?
you could try mainline 4.14 and swap mmc + put uart2 as first uart in mt7623.dtsi and use my config to check if problem comes up by any of my patches.

for filesystem i guess you have a hardware-error or your filesystem is broken...

Josue-T, 2019-04-17:
Hello,

Thank you for your answer,

After some test I detected that the network issue happen when avahi-daemon start. So when I disable it the network issue is solved. But I still think that it's not really normal that avahi-daemon break the kernel stability... Maybe can you try to reproduce this bug if you install avahi-daemon ?

> there was a network-problem imho between 4.14.92-98 but this is fixed. which is the last kernel-version you've tried?

4.14.107 and 4.19.31.

> have you any special network-setting? bridge (i see it...have you tried without it?), vlan, ... any way to reproduce the problem?

Yes use a bridge between all lan interfaces. I don't use a vlan.

> how about 4.19, here i have not added QDMA/BQL-Patches...only the second gmac alone (dsa-driver). you can revert it easier to check if it is the problem.

Well about the 4.19 I only tested the version 4.19.29 and 4.19.31

> do you use hwnat/hwqos?

I use the nat with iptables. As I understand with the 4.14 It use the hwnat.

> have you tried 5.0/5.1?

No

> you could try mainline 4.14 and swap mmc + put uart2 as first uart in mt7623.dtsi and use my config to check if problem comes up by any of my patches.

> for filesystem i guess you have a hardware-error or your filesystem is broken...

Not really sure. Since my last test the issue look like independent. But it's also strange because the issue happen only when I boot on my main rootfs with a kernel >= 4.14.82. Note If I boot from on an other rootfs and after I mount the same partition with a kernel >= 4.14.82 I don't have this issue with the same files.

frank-w, 2019-04-17:
please try 5.x and 4.19 before 2nd gmac is merged

try to figure out when the network-problem comes up (between which kernel-dot-releases) so we can look in git-commits which one may break it

currently i have much work and less time :(

Josue-T, 2019-04-17:
Do you have any build of the 5.x ?

Josue-T, 2019-04-17:
So, I tried this version here https://github.com/frank-w/BPI-R2-4.14/releases/download/CI-BUILD-20190412_101804-f78e2765e/bananapi-r2-image-4.19-main_4.19.33-1_armhf.deb and with this version it's ok.

> try to figure out when the network-problem comes up (between which kernel-dot-releases) so we can look in git-commits which one may break it

I tried, but it's really difficult because sometime I don't get the bug. So since my last test I would say that the version 4.14.67 (build : 20180829062337) is OK. The build 4.14.70 (build 20180917113512) is broken.

frank-w, 2019-04-17:
if 4.19 have shown the problem and now it's gone with 4.19.33 it should also be fixed in 4.14.110 (normally same patches are applied per dot-release across main versions).

i have autobuild only activated for lts-kernels (4.14,4.19), if your issue is now fixed we don't need to test 5.0 now...if not i can upload one on gdrive

Josue-T, 2019-04-18:
Well,

As I understand the issue happen when some service try to do a syscall about network and the network is not already initialized.

By example while avahi-daemon try to start before the network "service" we have this bug. I also discovered this issue with monitorix which also start before the network. I we start theses services (manually) when the network is initialized it's ok.

So about the 4.19, I don't know because with my last test a saw that some part of the network initialization (mt7530 and bridge initialisation) is before all other (critical) service. In this case the bug don't happen.

Josue-T, 2019-05-16:
Hello,

I did some new tests and I saw that the network issue happens with all kernel version >=4.14.63. About the 4.19 I don't get this error at boot, but after the boot after a while I got also this error in the kernel log (on the version 4.19.41).

The only version which I never got this bug is the version 4.14.55 (that I got on your google drive).

frank-w, 2019-05-16:
can you try to figure out which is the first kernel-version in 4.14 which have this error? it should be betweeb 4.14.55 and 4.14.63 if i know the first we can look over the changes between this and the prior version to get the cause

have you tried with 4.19 without my second gmac-patches (and maybe without wifi because it modifies watchdog)?

frank-w, 2019-05-16:
you have to revert changes and compile yourself

second gmac:

0f5ff06adcb2 net: dsa: mt7530 add linking to mdio
943ed2bae585 net: dsa: tell GDMA when we are turning on the special tag
954b359077f7 net: dsa: dsa multi cpu (mt7530.c)
a80c992c9372 net: dsa: add support for GMAC2 wired to ext
8325a7cbf964 net: dsa: adding handling of second CPU-Port
90adf3828368 net: dsa: add helper functions
329ff45aafea net: dsa: adding fields for holding information about upstream-port

wifi:

b3bf5911a5d9 [wifi] activated wifi-options
203a5a7727a8 [gcc] gcc8-fixes by Dominik Koch + nic_rx-patch from https://bugs.linaro.org/show_bug.cgi?id=3823
83ffbaceffed [wifi] adding wifi-related changes outside driver-directory
ce8c582c6121 [wifi] adding driver-folder

Josue-T, 2019-05-16:
> have you found a way to reproduce it?

No, I didn't found a way. The problem is that it come randomly. Until yesterday I never got this issue with the 4.14.63 and yesterday I got the issue with all version >=4.14.63...

Probably it liked to more than one element. But don't know which element need to match to reproduce for now.

With the 4.19 it's complicated to test because the issue happens not at the boot and come only a long time after the boot.

Maybe I might try to find which commit break the 4.14

frank-w, 2019-05-16:
on 4.14  i have included 2nd gmac in 4.14.52, imho you need to revert hnat and hwqos too

b3ba09aea285 (origin/4.14-hnat, 4.14-hnat) [HNAT] applied full Patch from hnat-branch
966f5efa66c9 (origin/4.14-hwqos, 4.14-hwqos) added hw-qos
a293bdb3d53a [gmac] added second gmac

still searching the right commits for wifi

frank-w, 2019-05-17:
For 4.14 could you try to revert this patch:

https://github.com/openwrt/openwrt/blob/master/target/linux/mediatek/patches-4.14/0027-net-next-mediatek-fix-DQL-support.patch

frank-w, 2019-10-20:
Any new information here? For network i reverted a patch which breaks network sometimes (dql)

closed 2023-03-04 by frank-w