frank-w/BPI-Router-Linux #64: Compiling with Docker

marwonline, 2019-07-08:
Today I wanted to compile the Kernel on a fresh Archlinux and realized that installing all those cross-compiler packages may take a time. So I hacked a small Dockerfile for me, which I'll share with you:

Dockerfile:
```
FROM ubuntu:xenial

RUN useradd -ms /bin/bash builder

RUN apt-get update               \
 && apt-get -y -q upgrade        \
 && apt-get -y -q install        \
    bc                           \
    binutils-arm-linux-gnueabihf \
    bison                        \
    build-essential              \
    ccache                       \
    flex                         \
    gcc-arm-linux-gnueabihf      \
    gccgo-4.7-arm-linux-gnueabi  \
    gcc-aarch64-linux-gnu        \
    git                          \
    libc6-armhf-cross            \
    libncurses-dev               \
    libssl-dev                   \
    make                         \
    u-boot-tools                 \
    wget                         \
    xz-utils                     \
 && apt-get clean

USER builder
WORKDIR /srv/
```

Create the local image with: `docker build . --tag bpi-compile:1`

Run the compile environment (adjust the path before the colon): 
`docker run -it -v $(pwd)/BPI-R2-4.14:/srv bpi-compile:1 /bin/bash`

Is it worth to create a pull request for this?

frank-w, 2019-07-08:
nice idea to create a docker container...isn't there a bionic base?

it seems you create the docker-file 1 level above the git repo (and mounting this to /srv), right?

need to setup a docker environment to test it...

marwonline, 2019-07-09:
Yes it one level outside of the git repo, but you can adjust the paths easily which are behind the `-v` flag.

Upgrading to bionic should be rather easy and also worth to try :+1:

frank-w, 2019-07-09:
imho the docker-file should be inside utils-directory (folder "docker" or similar) with a script to initialize it (your additional commands).

if you make a patch (git format-patch) i can apply it directly...should be easier as a pull request

marwonline, 2019-07-10:
I changed the stuff you mentioned and create the patch:

- bionic
- moved to `/docker` folder
- removed unused packages from apt

Download the [patch](https://github.com/frank-w/BPI-R2-4.14/files/3378722/adding-docker-env.patch.txt)

marwonline, 2019-07-10:
I just noticed that the docker daemon will create the `SD` folder if not present with root rights. I changed this in this patch: 
[adding-docker-env.patch.txt](https://github.com/frank-w/BPI-R2-4.14/files/3378745/adding-docker-env.patch.txt)

frank-w, 2019-07-10:
can you move docker folder to utils/ (utils/docker/...)? and squash the 3 Patches?

you can include the fixes/upgrade in single Patch instead of creating old value and than change it

frank-w, 2019-08-04:
merged your patch, but please make a patch to move docker folder into utils/

and some usage-instructions will be nice ;) as me and some other are not experienced with docker

marwonline, 2019-08-05:
As wished, I ...
- moved the Docker stuff into `./utils` and
- documented the usage in the `README.md`

[moved_Docker_tools_to___utils_and_added_some_documentation_for_it.patch.txt](https://github.com/frank-w/BPI-R2-4.14/files/3468581/moved_Docker_tools_to___utils_and_added_some_documentation_for_it.patch.txt)

frank-w, 2019-08-10:
seems patchfile is broken

```
[16:08:29]$ patch -p1 --dry-run < ~/Downloads/moved_Docker_tools_to___utils_and_added_some_documentation_for_it.patch
checking file docker/build_container.sh
patch: **** malformed patch at line 5: --- 2,1 ----
```

can you please make a patch with git-format-patch?

frank-w, 2019-08-10:
i have added the changes manually to 4.14-main, can you please try it?

marwonline, 2019-08-12:
I tried the version from `4.14-main` and everything works fine. :+1:

frank-w, 2019-08-12:
Ok,added it also to 4.19 and set you as author

closed 2019-08-12 by frank-w