frank-w/BPI-Router-Linux #68: Is It possibile install Wireguard?

Threefish4096, 2020-02-19:
As the title suggests, is it possible to run Wireguard VPN on BananaPi R2? 
I tried to install it from the unstable debian repositories, it goes, but once you start the quick start it gives me the following error:

ip link add dev wg0 type wireguard RNETLINK answers: Operation not supported.

I tried to follow the following site[Site](https://askubuntu.com/questions/973297/rnetlink-answers-operation-not-supported-fresh-ubuntu-fresh-wireguard)  and to install the missing linux header of the kernel version 4.19, but I cannot proceed because the linux header is missing in the /lib/module folder.

Can you help me or tell how i van obtein the linux header andò the wireguard module?

Thanks a lot!

frank-w, 2020-02-19:
I guess you have debian/ubuntu?
You can create linux-headers package by compiling my repo with make deb-pkg / bindeb-pkg...
I have a patch in 5.4-deb that creates binary-package (containing uImage+modules) too. You need to create headers for running kernelversion.

Imho in 5.6 wireguard was added,but it's not final and no lts

Threefish4096, 2020-02-19:
Thanks a lot!

Threefish4096, 2020-02-20:
I have debian with kernel 4.19.101.
In another PC with Linux I downloaded with git clone (git clone https://github.com/frank-w/BPI-R2-4.14.git 5.4-main) but from this point I have not quite understood how to continue to run the script you mentioned and how create the linux header.
Excuse the disorder

frank-w, 2020-02-20:
Apply top 3 commits of https://github.com/frank-w/BPI-R2-4.14/commits/5.4-deb in your tree and run build.sh pack_deps after compiling kernel. Install kernel and install linux-headers*.deb

Threefish4096, 2020-02-20:
I tray, but i obtein this...

[image]

frank-w, 2020-02-20:
Why not trying 5.4 first? Imho it should be possible cloning repo to /usr/src/$(uname -r)

Threefish4096, 2020-02-20:
I can't solve ..

Can you compile it (the lkernel with the linux-header.deb) and load it on the git or some clod? Many thanks in advance!

I tried first with 5.4 by downloading it from the following command, git clone https://github.com/frank-w/BPI-R2-4.14.git 5.4-deb, but the result is always the same, if I do 'make kernelversion' it exits with result 4.14.169
[image]

I don't know in which folder i need to work, all give the above result..

[image] In SD there is the 4.14.169

frank-w, 2020-02-21:
```
git clone --depth 1 --single-branch --branch 5.4-deb https://github.com/frank-w/BPI-R2-4.14
cd BPI-R2-4.14
git checkout 5.4-deb
make kernelversion
./build.sh importconfig
./build.sh
./build.sh pack_debs
```

Threefish4096, 2020-02-23:
Thank you a lot for the asistence!

closed 2020-02-23 by Threefish4096