frank-w/BPI-Router-Linux #69: Problem 5.4.21-bpi-r2-main

Threefish4096, 2020-02-24:
Here I am again bothering you, but I need to install the header.
After moving to the penultimate kernel build installed with sudo dpkg -i x.deb i have rebooted and the bananapi r2 changed ip, no problem.

[image]

I logged in and proceeded to install in the same way linux-libc-dev_5.4.21-bpi-r2-main-1_armhf.deb, no problem.

[image]

I had previously installed it, but I wanted to provide a screen that it works.

I reboot again and after the wget for obtein the image and the header i try to install it in the same way, but i obtein this errors..
[image]

[image]

How can I solve it? Many thanks in advance

frank-w, 2020-02-24:
you have already installed this kernel-version...you need to uninstall first to install again

Threefish4096, 2020-02-25:
Perfect, I managed to fix the 5.4.21 kernel.
I put the header, the image and the libc in the bananapi.
Now the problem is here ..

[image]

During the wireguard installation the necessary modules are compiled with dkms for older kernels, they complete without any problem, but..

[image]

this is where the problems start, because for my current kernel it doesn't compile in the least and the problem I don't understand where it can be.
Consulting the error log file gives me this, but I don't understand if it is a kernel problem or a Wireguard problem.

[image]

Also, if I try to see if it has loaded wireguard or to see if something is starting, I get nothing and the following error.

[image]
And so if I try to start it, it gives me errors like

[image]
with root mode

Thanks a lot in advance for the assistance!

frank-w, 2020-02-25:
Problem with fixdep is maybe because it's not a armhf binary...remove it and maybe start compilation of kernel so that this binary is build directly on r2

have no idea why wireguard takes 4.19.x source instead of 5.4...

closed 2020-04-23 by Threefish4096