frank-w/BPI-Router-Linux #70: To adapt to armbian

muravjov, 2020-03-02:
I ported [Armbian](https://github.com/muravjov/armbian-build) for BPI R64 so that kernels built in your repository are compatible to install to it.

Now the kernel used is from latest r64 build : https://github.com/muravjov/armbian-build/blob/banana-pi-r64/lib/distributions.sh#L202

I tested it, it works well, see http://forum.banana-pi.org/t/bpi-r64-armbian/10741.

But it turns out that armbian uses kernel features zram + swap (to avoid writing constantly logs to SD card).

It would be nice if you append to r64 config https://github.com/frank-w/BPI-R2-4.14/blob/5.4-main/arch/arm64/configs/mt7622_bpi-r64_defconfig those changes:

```
$ diff -Naur '/home/ilya/opt/bpi-r64/tmp/.config.5.4.17-wg-embedded' .config | grep -v '#'
--- /home/ilya/opt/bpi-r64/tmp/.config.5.4.17-wg-embedded       2020-02-29 20:55:30.000000000 +0300
+++ .config     2020-03-01 18:15:26.031627017 +0300
@@ -27,7 +27,7 @@
 CONFIG_BUILD_SALT=""
 CONFIG_DEFAULT_HOSTNAME="(mt7622)"
+CONFIG_SWAP=y
 CONFIG_SYSVIPC=y
 CONFIG_SYSVIPC_SYSCTL=y
 CONFIG_POSIX_MQUEUE=y
@@ -128,6 +128,8 @@
 CONFIG_CGROUPS=y
 CONFIG_PAGE_COUNTER=y
 CONFIG_MEMCG=y
+CONFIG_MEMCG_SWAP=y
+CONFIG_MEMCG_SWAP_ENABLED=y
 CONFIG_MEMCG_KMEM=y
 CONFIG_BLK_CGROUP=y
 CONFIG_CGROUP_WRITEBACK=y
@@ -412,6 +414,7 @@
 CONFIG_SUSPEND=y
 CONFIG_SUSPEND_FREEZER=y
 CONFIG_PM_SLEEP=y
 CONFIG_PM_SLEEP_SMP=y
@@ -667,10 +670,15 @@
+CONFIG_FRONTSWAP=y
+CONFIG_ZSWAP=y
+CONFIG_ZPOOL=y
+CONFIG_ZSMALLOC=y
 CONFIG_GENERIC_EARLY_IOREMAP=y
@@ -1318,6 +1326,7 @@
 
@@ -1390,7 +1399,23 @@
 CONFIG_OF_RESERVED_MEM=y
+CONFIG_BLK_DEV=y
+CONFIG_ZRAM=m
+CONFIG_ZRAM_WRITEBACK=y
 
@@ -1509,6 +1534,7 @@
 CONFIG_BLK_DEV_SD=y
@@ -4051,6 +4077,7 @@
 
 CONFIG_NVMEM=y
 CONFIG_NVMEM_SYSFS=y
@@ -4426,10 +4453,10 @@
-CONFIG_CRYPTO_DEFLATE=m
+CONFIG_CRYPTO_DEFLATE=y
+CONFIG_CRYPTO_LZO=y
+CONFIG_CRYPTO_LZ4=y
 
@@ -4510,8 +4537,12 @@
 CONFIG_AUDIT_ARCH_COMPAT_GENERIC=y
 CONFIG_AUDIT_COMPAT_GENERIC=y
-CONFIG_ZLIB_INFLATE=m
-CONFIG_ZLIB_DEFLATE=m
+CONFIG_ZLIB_INFLATE=y
+CONFIG_ZLIB_DEFLATE=y
+CONFIG_LZO_COMPRESS=y
+CONFIG_LZO_DECOMPRESS=y
+CONFIG_LZ4_COMPRESS=y
+CONFIG_LZ4_DECOMPRESS=y
 CONFIG_GENERIC_ALLOCATOR=y
 CONFIG_ASSOCIATIVE_ARRAY=y
```

frank-w, 2020-03-03:
added options to https://github.com/frank-w/BPI-R2-4.14/commits/5.4-main

muravjov, 2020-03-03:
Two more options are required for zram to function without errors:

```
$ diff -Naur .config.old .config
--- .config.old 2020-03-01 18:15:26.031627017 +0300
+++ .config     2020-03-03 17:23:45.004036296 +0300
@@ -674,8 +674,8 @@
 # CONFIG_CMA is not set
 CONFIG_ZSWAP=y
 CONFIG_ZPOOL=y
-# CONFIG_ZBUD is not set
-# CONFIG_Z3FOLD is not set
+CONFIG_ZBUD=y
+CONFIG_Z3FOLD=y
 CONFIG_ZSMALLOC=y
 # CONFIG_PGTABLE_MAPPING is not set
 # CONFIG_ZSMALLOC_STAT is not set
```

Without them I got weird kernel log
```
zswap: pool creation failed
```
I believe it is enough for zram

frank-w, 2020-03-04:
have added these 2, too

muravjov, 2020-03-05:
Thanks, it works now

closed 2020-03-05 by muravjov