frank-w/BPI-Router-Linux #71: May be include wireguard in 5.4-main?

muravjov, 2020-03-02:
Your work about 5.4-main is just fine.

I would like to use your release kernels at https://github.com/frank-w/BPI-R2-4.14/releases , but they lacks wireguard.

Yep, I know possibility to install headers and dkms (and it even works), but it turns out that to patch sources is better for me for the following reasons:
* building wireguard.ko is much longer than in my thinkpad (10x longer) => installing new kernel from deb involves rebuild => unexpected long
* headers deb now contains 2 utilities, fixdep and modpost, compiled for x86, not arm64 :( so you need manually recompile them and only then you can use them

So, may be cherry-pick wireguard from here? https://github.com/muravjov/BPI-R2-4.14/commit/c2e547f9744e5245d663af40ab7b7bede37993bc

I applied wireguard like armbian does it 
https://github.com/armbian/build/blob/master/lib/compilation-prepare.sh#L114

frank-w, 2020-03-03:
Can you give me advice on how to include wireguard sources (from git) and crosscompile it?

muravjov, 2020-03-05:
I propose to follow armbian-way to do it, and so I extract code into add_wireguard.sh ,
see https://github.com/muravjov/armbian-build/blob/banana-pi-r64/add_wireguard.sh

How to use:
```
git clone https://github.com/muravjov/armbian-build

$ armbian-build/add_wireguard.sh /home/ilya/opt/bpi-r64/linux
Trying to patch kernel version=5.4.0 ...
[ o.k. ] Adding [ WireGuard VPN for Linux 3.10 - 5.5 branch:master  ]
[ o.k. ] Checking git sources [ wireguard master ]
[ .... ] Creating local copy 
[ .... ] Fetching updates 
remote: Enumerating objects: 172, done.
remote: Counting objects: 100% (172/172), done.
remote: Compressing objects: 100% (136/136), done.
remote: Total 172 (delta 10), reused 93 (delta 8)
Receiving objects: 100% (172/172), 405.46 KiB | 539.00 KiB/s, done.
Resolving deltas: 100% (10/10), done.
From https://git.zx2c4.com/wireguard-linux-compat
 * branch            master     -> FETCH_HEAD
 * [new branch]      master     -> origin/master
[ .... ] Checking out 
[ o.k. ] Patching WireGuard [ Applying workaround for headers compilation ]
```
You can use either last commit to get latest Wireguard, master branch, or previous commit 13c23c76fce53641058fc42fec780b8ca8329e58 which applies tag:0.0.20191219 (of Wireguard)

muravjov, 2020-03-05:
Thanks for applied  wireguard
I close this one, if you need please reopen

closed 2020-03-05 by muravjov

frank-w, 2020-03-05:
Is last tag also compatible with 5.4?

muravjov, 2020-03-05:
I didn't tested that. 
But I've tested your last released kernel  with wireguard=tag:0.0.20191219 and it works

muravjov, 2020-03-05:
Nevertheless, it is compatible, yes. Because it is from https://git.zx2c4.com/wireguard-linux-compat
and from it one may see https://git.zx2c4.com/wireguard-linux-compat/tree/README.md :

```
WireGuard was merged into the Linux kernel for 5.6. This repository contains a backport of WireGuard for kernels 3.10 to 5.5, as an out of tree module.

```

frank-w, 2020-03-05:
I appplied your patch for the december 19th version...this makes an in tree-module because code lies in net/

Can you test latest tag?

muravjov, 2020-03-06:
Done in https://github.com/frank-w/BPI-R2-4.14/pull/73