frank-w/BPI-Router-Linux #72: openvswitch issue

ifakmme, 2020-03-05:
Adding the ports lan0-lan3 to an openvswitch bridge does not seem to work. Arp requests are forwarded correctly, but the replies do not make it to the userspace, even if the corresponding port is in promiscuous mode.

The same problem (in another scenario) is described here:
http://forum.banana-pi.org/t/ethernet-in-promiscuous-mode/6773

I tried the 4.14.169-main and 4.19.101-main kernel.
Is there something I can do to use an openvswitch bridge on the BPI-R2?

Thanks,
Marco

frank-w, 2020-03-05:
I guess dsa-driver does not really support promiscious mode...

But as i do not know openvswitch i think i cannot not help here much.

Do you see arp-packets reaching r2? Maybe they are not send to it by switch before.

ifakmme, 2020-03-06:
I investigated the problem a bit further.
In fact, promiscuous mode does not seem to work at all. According to dmesg, eth0 and lanX is set to promiscuous mode, but no unicast packet with a different mac address hits the userspace (tried to observe via tcpdump).

Interestingly, using a kernel bridge seems to work:
`ip link add name br0 type bridge`
`ip link set br0 up`
`ip link set dev lan2 master br0`
`ip link set dev lan3 master br0`

But in this setup, the packets passing through the bridge cannot be observed via tcpdump on the lan2 and lan3 interfaces. In a setup with two distinct NICs and a kernel bridge, the behavior is different.

What kind of magic does the switch driver to?
Would it be possible to add support for promiscuous mode to the dsa-driver, or do you see any other way of making this setup work?

Thanks,
Marco

frank-w, 2020-03-06:
Afaik traffic on all dsa-ports get a special tagging and routed (hardcoded) to gmac0 (eth0) if no second gmac patch is applied (like 5.4-main,was done in 4.14 and 4.19).

I don't know how promiscious mode can be implemented in current code :(

Tankernn, 2020-10-10:
I think I have a related problem. I'm using https://github.com/frank-w/openwrt and `5.4-main`. I cannot receive packets on any of the virtual interfaces (`...@eth0`). I tried with the `5.4-gmac` branch, and the `wan` interface started working. Sending packets works fine on both branches. (I can see the ARP requests and replies in `tcpdump` on another machine)

Is there some additional configuration I'm missing, or is the switch driver broken in some way on these branches?

frank-w, 2020-10-10:
5.4-gmac uses another gmac (#2) for wan.there are 2 patches related.one introduces the second gmac and the other allows renaming it because it does not use dsa and is named eth1 by default

frank-w, 2023-03-04:
No progress - closing

closed 2023-03-04 by frank-w