frank-w/BPI-Router-Linux #74: Bridging between lanx and tap devices not working

mKenfenheuer, 2020-06-06:
Hi,

i am currently trying to setup a simple vpn client which connects to a remote vpn network and bridges this to lan0-3 on the bpi-r2. with the official kernel the bridging works but i can only reach 3MB/s of 10MB/s download, upload is working fine. 

Setup with the official kernel 4.4 is as follows:
VPN => tap_01 => br0 => eth1 (all lan ports, no port separation) => client

After switching to 5.4-main i cannot get any connection between br0 and tap_01.
Setup is as follows: 
VPN => tap_01 => br0 => lan0, lan1, lan2, lan3 => client

Furthermore i have seen that if i connect two devices to lan0 and lan1 (or any other two lan ports) they can communicate. Therefore i assume that bridging between tap_01 and br0 is not working properly. 

Any idea?

mKenfenheuer, 2020-06-06:
More investigation:

When assigning an ip address to the bridge, network traffic is being bridged, but with worse performance.

Up and download is now at 1MB/s, after some seconds the lanX connection disconnects and connects again.

frank-w, 2020-06-08:
how is the real topology?

do you create a lanbr (lan0-lan3) and then bridging this bridge with your tap-device?

```
                     br0
        lanbr                     tap
lan0 lan1 lan2 lan3
```

or do you add your tap-device to the lanbr?

mKenfenheuer, 2020-06-11:
I am creating a single lanbr with lan0-3 and the tap device.
Shall i try the other way you suggested?

closed 2023-03-04 by frank-w