frank-w/BPI-Router-Linux #77: Banana Pi R2 Armbian

martinl, 2020-08-15:
Hi

I'm currently looking into options for adding official Banana Pi R2 support to Armbian (https://github.com/armbian/build/issues/2147) using your 5.4-main as base. Can you please explain which patches are needed for R2 from your 5.4-main branch vs mainline 5.4

Also, would you be interested in testing and supporting R2 on Armbian to some extent

Martin

frank-w, 2020-08-15:
`git logone v5.4.58..5.4-main | grep -v "Merge tag"`

there are some commits that should not be catched (seems like from 5.4.21...i removed them and replaced by ...) because they are not from me (don't know why), but it filters many out from the merges

```
$ git logone v5.4.58..5.4-main | grep -v "Merge tag"
95d702080506 defconfig: r2: add squashfs
e5ef215df9b5 defconfig: r2: add usb_printer
e9be7441fee3 Add module reset before eMMC init
90fb2149034a net: dsa: mt7530: fix tagged frames pass-through in VLAN-unaware mode
d4dc1e7e87d8 Revert "net: dsa: enable vlan without bridge on dsa user port"
53cae12d7d82 Revert "arm: dts: workaround non-working trgmii"
318b661c9578 net: ethernet: mediatek: move mt7623 settings out off the mt7530
0cb5221062b6 net: dsa: mt7530: move mt7623 settings out off the mt7530
ae24ad6a9054 build.sh: fix check for enabled kernel (need uenv.txt be modified)
2946caa5862d mt76: add firmware for mt7612e
d5325fb58f9d wireguard: fix implicit declaration skb_reset_tc
a3a3d4a17df3 build.sh: show only build kernel, compare with old and ask for edit uenv
9414adbdfcd5 mt6625l: fix makefile for module compilation (BT)
0c645101a841 defconfig: r2: enable internal BT module
119ba42b2f62 build.sh: do not prompt for commit-message on updatesrc + show max
76ff28a85894 readme: openvpn+lxc reported as working in 4.19
c62152c80834 utils: add script for building chroot and compiling hostapd
d822d17e804e added wireguard from master
3f295ef506b6 Revert "added wireguard, tag:0.0.20191219"
1beb6809a118 defconfig: r64: adding ZBUD/Z3FOLD for ZSWAP (armbian)
dabd02495f84 added wireguard, tag:0.0.20191219
6f8e40393d73 defconfig: r64: add options for armbian
0b9b9e3c95b8 travis: include pack_debs in travis config
2dc7c5ddc90d build-deb: pack uImage for r2 by default
168256ce9fd8 build.sh: pack_debs
710815d067b2 kbuild: packaging images in deb
...
2e4b86dfcf32 defconfig: add mt76x2 and policy routing
d87e1a23e511 lima: power on/off via register (function)
be435b41a6df defconfig: add lima/mali related options
9a25a149ad7c arm: dts: drop larb3
6468d187f126 arm: dts: mt7623: add Mali-450 device nodes
502577561fb7 defconfig: add hwmon / thermal
9847677bda49 thermal: mediatek: execute hwmon-code only if hwmon_thermal is set
24fbb7bc18a4 defconfig: add gpio-leds and hwmon/thermal
aa97a4429717 thermal: mediatek: add sensors-support
336c58dfb453 dts64: add blue LED for bananapi r64
89e77799248a build.sh: fix dts-command for r64
dfc7ec67832a defconfig: r64: add module unload and rfkill
734e1f4cee47 defconfig: r2: disable NET_SCH_NETEM
ecf847615cdd cryptodev: fix makefile for crosscompile
6d2eeaf91f12 utils: cryptodev: Fix module loading with Linux v5.0-rc5
c3b89e4625bb utils: add cryptodev 1.10
c5778ceacbbd utils: add build-script for openssl
3286019d196e utils: add wifi.sh
418b8fed6475 utils: add wmt-utils from 4.19
4386f5dab24c utils: add mmc-utils
b740cbeb8db4 utils: add docker from 4.19
901e9cab4b7e arm: dts: workaround non-working trgmii
020d0238fc87 thermal: fix thermal on bananapi-r2
8a98d219b9de usb: musb: Add support for MediaTek musb controller
d2466ebbb542 usb: musb: Add musb_clearb/w() interface
a727040290a1 usb: musb: Add noirq type of dma create interface
d63df18a2c7e usb: musb: Add get/set toggle hooks
e194477fb763 arm: dts: mt2701: Add usb2 device nodes
14e3096c0288 dt-bindings: usb: musb: Add support for MediaTek musb controller
581413d5d096 defconfig: r64: add net, ip, netfilter, cgroups
a4f87b2e9c4f readme: added known issues and some small fixes
5c2c4553ac89 build.sh: fix deb-function for r64
f80d575d2c62 travis: add r64 to build
a44483012c2f README: add README.md
c32b80ad0e05 travis-ci: add travis-ci config
fa3ca92bd497 build.sh: disable switch in name (pack,upload)
a7b193b05735 build.sh: remove r64 version code
[... 30 lines cut ...]
922ecdd3bba1 mt6625l: Drop usage of wakelocks Android API, use wake sources instead.
f35ad18d31f9 mt6625l: mark implicit fallthrough as wanted to fix build
49a0771c25f3 mt6625l: adding wifi-related changes outside driver-directory
846b1dc5520e mt6625l: include wifi-folder
06d376a01ea0 mt6625l: add driver-folder from 5.3
324a081bd7a6 serial: 8250-mtk: Use platform_get_irq_optional() for optional irq
34dd2a86c9e3 drm: fix implicit declaration of function 'DRM_INFO'
f4ca44932ccf drm/mediatek: dpi/dsi: change the getting possible_crtc way
452b508468a4 drm: Add get_possible_crtc API for dpi, dsi
ba9c6527b05d drm/mediatek: fix boot up for 720 and 480 but 1080
ae7a8d61a108 arm: dts: mt7623: add display subsystem related device nodes
6e3f7375acdc drm/mediatek: config component output by device node port
c3db4c1163cc arm: dts: mt6323: add keys, power-controller, rtc and codec
ba012207e860 MAINTAINERS: add Mediatek shutdown drivers
559614ab0ae2 power: reset: add driver for mt6323 poweroff
1695fbd300f4 rtc: mt6397: add compatible for mt6323
546eb08339db rtc: mt6397: improvements of rtc driver
de8d437020b4 rtc: mt6397: move some common definitions into rtc.h
fc34f8ccb24b dt-bindings: rtc: mediatek: add missing mt6397 rtc
43c6973ff98e [dtsi] swap mmc and put debug-uart (2) first
a90c2da3c6eb add own files (build.sh,build.conf,defconfig) + update gitignore
```
as i do the most work linux, uboot, images and other support (in forum), i cannot support armbian/openwrt too

basicly mt6625l (drivers/misc/mediatek + some patches outside), hdmi, poweroff was not integrated and some tools. i guess trgmii-fix was added in the latest 5.4 dot-release after putting it to mainline

martinl, 2020-08-16:
Thanks, I made one big diff 5.4-main vs 5.4 mainline for now. Removed wireguard since Armbian already has patch for that

martinl, 2020-08-16:
Full diff is about 284k lines https://github.com/martinl/banana-pi-r2/tree/master/armbian/patch/kernel/mt7623-dev

frank-w, 2020-08-16:
it is the internal wifi-driver which is very huge...i cleaned it a bit, but it's to much to be upstreamed (same for mainline and openwrt)...maybe you should split wifi-changes from hdmi,thermal and the rest...something like squashing related commits together (e.g. tagged with mt6625l for internal wifi)

martinl, 2020-08-16:
yes, I took a quick look and I do not see the point in including the huge mt6625l until there is native driver available. Adding a supported USB or mPCIe wifi adapter to R2 seems like a reasonable workaround for now.

martinl, 2020-08-16:
I split the drivers subtree to per directory diffs for easier reviewing

frank-w, 2020-08-16:
Splitting wifi-driver make not much sense because all so far is needed to work,dirs alone did not...

martinl, 2020-08-16:
yes, just making it easier to read/review. I'm mildly interested in how to write wifi driver in 280k lines :slightly_smiling_face:

frank-w, 2020-08-16:
It will not be reviewed/upstreamed in current form...it can only be in mainline if completely rewritten

martinl, 2020-08-16:
I'm mostly planning to review it myself, to see which parts are needed for Armbian patches

frank-w, 2020-08-16:
It will be more interesting why wifi client mode does not work...see discussion in forum http://forum.banana-pi.org/t/bpi-r2-internal-wifi-bt-mt6625l-kernel/4307/299

frank-w, 2022-01-15:
No more answer from creator of issue

closed 2022-01-15 by frank-w