frank-w/BPI-Router-Linux #81: [R2] WiFi not working on self-made Alpine linux

black-ish, 2021-03-14:
Hello,
I don't know how to start rly so I hope I can get my issue across for some help.

I've read up quite a bit on it and I followed all steps very closely and got Alpine working with kernel 5.4-main and 5.10-main.

So after setting up everything I wante to try the on-board Wifi and the first thing that happened was weird.
I started "/usr/bin/wmt_loader" and the first thing I get is "/usr/bin/wmt_loader: No such file or directory" and I'm quite baffled here as the exectuable is in "/usr/bin" and also is +x alright.
So I looked around and found another, old, Alpine from Pantavisor and tried replacing the binaries with the ones from that Alpine, suddenly wmt_loader starts and does its thing.
So second step "/usr/bin/stp_uart_launcher -p /etc/firmware &" and same thing with wmt_loader, "/usr/bin/stp_uart_launcher: No such file or directory". I do the same thing with this binary as with wmt_loader and replace it with the one from the Pantavisor Alpine, and see there it starts, but it segfaults at the end (I don't know is this is normal).
I don't mind it as it did run and created the "/dev/wmtWifi" device. So next step is "echo A >/dev/wmtWifi" and I get "echo: write error: I/O error" as such I can never enable the ap0 interface.

dmesg shows the following:

``Mar 14 18:44:40 kernel: [97563.280914] [MTK-WIFI] WIFI_open: WIFI_open: major 155 minor 0 (pid 2119)
Mar 14 18:44:40 kernel: [97563.280977] [MTK-WIFI] WIFI_write: WIFI_write A
Mar 14 18:44:40 kernel: [97563.281060] vcn18: mode operation not allowed
Mar 14 18:44:40 kernel: [97563.282732] [WMT-CONSYS-HW][E]mtk_wcn_consys_hw_reg_ctrl(291):Read CONSYS chipId(0xfffff093)
Mar 14 18:44:40 kernel: [97563.376954] [WMT-CONF][E]wmt_conf_parse_pair(323):unknown field 'mt6620.defAnt'.
Mar 14 18:44:40 kernel: [97563.376966] [WMT-CONF][E]wmt_conf_parse_pair(323):unknown field 'mt6628.defAnt'.
Mar 14 18:44:40 kernel: [97563.376974] [WMT-CONF][E]wmt_conf_parse_pair(323):unknown field 'mt6630.defAnt'.
Mar 14 18:44:40 kernel: [97563.377201] [WMT-CORE][E]wmt_core_stp_init(648):WMT-CORE: no hif info!
Mar 14 18:44:40 kernel: [97563.377208] drivers/misc/mediatek/connectivity/common/conn_soc/core/wmt_core.c, 649, (0)
Mar 14 18:44:40 kernel: [97563.377214] [WMT-CORE][E]opfunc_pwr_on(918):WMT-CORE: wmt_core_stp_init fail (-1)
Mar 14 18:44:40 kernel: [97563.377219] drivers/misc/mediatek/connectivity/common/conn_soc/core/wmt_core.c, 919, (0)
Mar 14 18:44:40 kernel: [97563.377229] [WMT-CORE][E]wmt_core_ctrl(499):WMT-CORE: wmt_core_ctrl failed: id(5), type(1), value(0) iRet:(-1)
Mar 14 18:44:40 kernel: [97563.377235] drivers/misc/mediatek/connectivity/common/conn_soc/core/wmt_core.c, 501, (0)
Mar 14 18:44:40 kernel: [97563.377242] [WMT-CORE][E]wmt_core_ctrl(499):WMT-CORE: wmt_core_ctrl failed: id(5), type(0), value(0) iRet:(-1)
Mar 14 18:44:40 kernel: [97563.377246] drivers/misc/mediatek/connectivity/common/conn_soc/core/wmt_core.c, 501, (0)
Mar 14 18:44:40 kernel: [97563.377257] [WMT-CORE][E]wmt_core_ctrl(499):WMT-CORE: wmt_core_ctrl failed: id(3), type(0), value(0) iRet:(-2)
Mar 14 18:44:40 kernel: [97563.377262] drivers/misc/mediatek/connectivity/common/conn_soc/core/wmt_core.c, 501, (0)
Mar 14 18:44:40 kernel: [97563.377542] vcn18: mode operation not allowed
Mar 14 18:44:40 kernel: [97563.377579] vcn18: mode operation not allowed
Mar 14 18:44:40 kernel: [97563.379176] [WMT-CONSYS-HW][E]mtk_wcn_consys_hw_reg_ctrl(291):Read CONSYS chipId(0xfffff093)
Mar 14 18:44:40 kernel: [97563.476854] [WMT-CONF][E]wmt_conf_parse_pair(323):unknown field 'mt6620.defAnt'.
Mar 14 18:44:40 kernel: [97563.476866] [WMT-CONF][E]wmt_conf_parse_pair(323):unknown field 'mt6628.defAnt'.
Mar 14 18:44:40 kernel: [97563.476873] [WMT-CONF][E]wmt_conf_parse_pair(323):unknown field 'mt6630.defAnt'.
Mar 14 18:44:40 kernel: [97563.477095] [WMT-CORE][E]wmt_core_stp_init(648):WMT-CORE: no hif info!
Mar 14 18:44:40 kernel: [97563.477102] drivers/misc/mediatek/connectivity/common/conn_soc/core/wmt_core.c, 649, (0)
Mar 14 18:44:40 kernel: [97563.477108] [WMT-CORE][E]opfunc_pwr_on(918):WMT-CORE: wmt_core_stp_init fail (-1)
Mar 14 18:44:40 kernel: [97563.477113] drivers/misc/mediatek/connectivity/common/conn_soc/core/wmt_core.c, 919, (0)
Mar 14 18:44:40 kernel: [97563.477123] [WMT-CORE][E]wmt_core_ctrl(499):WMT-CORE: wmt_core_ctrl failed: id(5), type(1), value(0) iRet:(-1)
Mar 14 18:44:40 kernel: [97563.477128] drivers/misc/mediatek/connectivity/common/conn_soc/core/wmt_core.c, 501, (0)
Mar 14 18:44:40 kernel: [97563.477135] [WMT-CORE][E]wmt_core_ctrl(499):WMT-CORE: wmt_core_ctrl failed: id(5), type(0), value(0) iRet:(-1)
Mar 14 18:44:40 kernel: [97563.477140] drivers/misc/mediatek/connectivity/common/conn_soc/core/wmt_core.c, 501, (0)
Mar 14 18:44:40 kernel: [97563.477150] [WMT-CORE][E]wmt_core_ctrl(499):WMT-CORE: wmt_core_ctrl failed: id(3), type(0), value(0) iRet:(-2)
Mar 14 18:44:40 kernel: [97563.477155] drivers/misc/mediatek/connectivity/common/conn_soc/core/wmt_core.c, 501, (0)
Mar 14 18:44:40 kernel: [97563.477419] vcn18: mode operation not allowed
Mar 14 18:44:40 kernel: [97563.477455] vcn18: mode operation not allowed
Mar 14 18:44:40 kernel: [97563.479059] [WMT-CONSYS-HW][E]mtk_wcn_consys_hw_reg_ctrl(291):Read CONSYS chipId(0xfffff093)
Mar 14 18:44:40 kernel: [97563.576847] [WMT-CONF][E]wmt_conf_parse_pair(323):unknown field 'mt6620.defAnt'.
Mar 14 18:44:40 kernel: [97563.576859] [WMT-CONF][E]wmt_conf_parse_pair(323):unknown field 'mt6628.defAnt'.
Mar 14 18:44:40 kernel: [97563.576867] [WMT-CONF][E]wmt_conf_parse_pair(323):unknown field 'mt6630.defAnt'.
Mar 14 18:44:40 kernel: [97563.577101] [WMT-CORE][E]wmt_core_stp_init(648):WMT-CORE: no hif info!
Mar 14 18:44:40 kernel: [97563.577108] drivers/misc/mediatek/connectivity/common/conn_soc/core/wmt_core.c, 649, (0)
Mar 14 18:44:40 kernel: [97563.577114] [WMT-CORE][E]opfunc_pwr_on(918):WMT-CORE: wmt_core_stp_init fail (-1)
Mar 14 18:44:40 kernel: [97563.577119] drivers/misc/mediatek/connectivity/common/conn_soc/core/wmt_core.c, 919, (0)
Mar 14 18:44:40 kernel: [97563.577129] [WMT-CORE][E]wmt_core_ctrl(499):WMT-CORE: wmt_core_ctrl failed: id(5), type(1), value(0) iRet:(-1)
Mar 14 18:44:40 kernel: [97563.577134] drivers/misc/mediatek/connectivity/common/conn_soc/core/wmt_core.c, 501, (0)
Mar 14 18:44:40 kernel: [97563.577141] [WMT-CORE][E]wmt_core_ctrl(499):WMT-CORE: wmt_core_ctrl failed: id(5), type(0), value(0) iRet:(-1)
Mar 14 18:44:40 kernel: [97563.577146] drivers/misc/mediatek/connectivity/common/conn_soc/core/wmt_core.c, 501, (0)
Mar 14 18:44:40 kernel: [97563.577156] [WMT-CORE][E]wmt_core_ctrl(499):WMT-CORE: wmt_core_ctrl failed: id(3), type(0), value(0) iRet:(-2)
Mar 14 18:44:40 kernel: [97563.577160] drivers/misc/mediatek/connectivity/common/conn_soc/core/wmt_core.c, 501, (0)
Mar 14 18:44:40 kernel: [97563.577427] vcn18: mode operation not allowed
Mar 14 18:44:40 kernel: [97563.577462] [WMT-CORE][E]opfunc_func_on(1025):func(3) pwr_on fail(-2)
Mar 14 18:44:40 kernel: [97563.577467] drivers/misc/mediatek/connectivity/common/conn_soc/core/wmt_core.c, 1026, (0)
Mar 14 18:44:40 kernel: [97563.577514] [STP] mtk_wcn_stp_psm_enable:[W] STP Not Ready, Dont do Sleep/Wakeup
Mar 14 18:44:40 kernel: [97563.577525] [MTK-WIFI] WIFI_write: WMT turn on WIFI fail!
Mar 14 18:44:40 kernel: [97563.577732] [MTK-WIFI] WIFI_close: WIFI_close: major 155 minor 0 (pid 2119)
Mar 14 18:44:52 kernel: [97574.912583] [MTK-WIFI] WIFI_open: WIFI_open: major 155 minor 0 (pid 2119)
Mar 14 18:44:52 kernel: [97574.912645] [MTK-WIFI] WIFI_write: WIFI_write A
[... 35 lines cut ...]
Mar 14 18:44:52 kernel: [97575.206857] [WMT-CONF][E]wmt_conf_parse_pair(323):unknown field 'mt6628.defAnt'.
Mar 14 18:44:52 kernel: [97575.206864] [WMT-CONF][E]wmt_conf_parse_pair(323):unknown field 'mt6630.defAnt'.
Mar 14 18:44:52 kernel: [97575.207102] [WMT-CORE][E]wmt_core_stp_init(648):WMT-CORE: no hif info!
Mar 14 18:44:52 kernel: [97575.207108] drivers/misc/mediatek/connectivity/common/conn_soc/core/wmt_core.c, 649, (0)
Mar 14 18:44:52 kernel: [97575.207114] [WMT-CORE][E]opfunc_pwr_on(918):WMT-CORE: wmt_core_stp_init fail (-1)
Mar 14 18:44:52 kernel: [97575.207119] drivers/misc/mediatek/connectivity/common/conn_soc/core/wmt_core.c, 919, (0)
Mar 14 18:44:52 kernel: [97575.207130] [WMT-CORE][E]wmt_core_ctrl(499):WMT-CORE: wmt_core_ctrl failed: id(5), type(1), value(0) iRet:(-1)
Mar 14 18:44:52 kernel: [97575.207135] drivers/misc/mediatek/connectivity/common/conn_soc/core/wmt_core.c, 501, (0)
[... cut ...]
I don't know what else to do here. I want to use Alpine so another distro is out of the question for me, what can I do here?

I don' know if this is related but if I use the kernel that has NOT the ending "_nodt" the kernel panics at boot time claiming I didn't append the correct "root=" parameter. So I USE the kernel image WITH the ending of "nodt".

I'm not exactly new to this, but this is my first time going at a dsitro this way, since there is no up-to-date premade Alpine image for this board.

frank-w, 2021-03-14:
I guess "no such file..." on binaries are because libs are missing. You can show them with "ldd file". But with the other binaries it should work if they run. You can find needed libs in my buildroot-repo,if you want to try with debian binaries

https://github.com/frank-w/buildroot/tree/2020.11-bpi/rootfs_overlay

As for kernel "_nodt" stands for no devicetree,this kernel binary needs separate dtb to be loaded in uboot (for using with dt overlays). Here you can boot the normal uImage..don't care about nodt version

black-ish, 2021-03-14:
Ok, I tried some things with success.
I had to add the "gcompat" package which adds a small "ld-linux-armhf.so.3" in "/lib/" then had to replace that with the one from your link.
I have to use the "wmt_loader" from the Pantavisor Alpine package because the one I compiled or the one you have only gives me "permission denied" error.
Now it works. the ap0 interface is present. I will try hostpad next and update this thread a bit for maybe some others that want to know about how it works, if that is ok?.

frank-w, 2021-03-15:
i guess you can only use my package (files from buildroot rootfs overlay) and do not need the gcompat package as you replaced the file there

ok, please close it when it's running

black-ish, 2021-03-15:
Hostapd works now, some caveats are there that I noticed:
Restarting hostapd (because of channel change or other setting change) causes the WLAN ssid to not be broadcasted anymore, the interface has to be "restarted" with "echo 0 >/dev/wmtWifi" and then starting it again with "echo A >/dev/wmtWifi"
ACS seems to not be supported as it causes hostapd to get event 47(or 57?) which seems not supported.

I'll try to check which binaries function with which libs, but for now it works with how I described it, thank you very much again :)

closed 2021-03-15 by black-ish