frank-w/BPI-Router-Linux #87: SD-Card not found!

usermigallon, 2021-12-22:
Hi, I have this "SD-Card not found!"

frank-w, 2021-12-22:
can you please a bit more exactly? where did you get the error? on building/install or on running the kernel?

i guess while try to install via my build.sh

which operating system do you use? is BPI-BOOT/BPI-ROOT mounted to /media/$USER/? this is what the script checks

usermigallon, 2021-12-22:
1) pack
2) install to SD-Card
3) deb-package
4) upload
choice [1234]:2
Install Kernel to SD Card
generate branch vars...
getting git branch: 
* 5.4-main
kernbranch:5.4-main,gitbranch:-main
uImage-filename: uImage_5.4.150-main /dev/sde 
Name: uImage_5.4.150-main /dev/sde
Press [enter] to copy data to SD-Card...
SD-Card not found!

frank-w, 2021-12-22:
do you try to install to usb-stick/ssd/hdd?

build.sh uses /media/$USER/BPI-BOOT / BPI_ROOT to install, if your mountpoint differs to it you have to change occourances of this mountpoint with yours

usermigallon, 2021-12-22:
I have not install usb-stick/ssd/hdd

frank-w, 2021-12-22:
(where) is your sdcard mounted?

usermigallon, 2021-12-22:
/media/user/7F3D-3FFE or /dev/sde

frank-w, 2021-12-22:
OK, /media/user/7F3D-3FFE is your mountpoint, i guess because you miss the label in filesystem.

do you build for r2 or r64 (new bootchain)?

please show output of 

`mount | grep /dev/sde`
`fdisk -l /dev/sde`

frank-w, 2021-12-22:
you have no valid BPI image on your sdcard...it looks like a blanko sdcard.

you need a running system before installing new kernel

e.g. my debian buster/bullseye image you can found on bpi-forum

usermigallon, 2021-12-22:
I build r2

usermigallon, 2021-12-22:
my sd card is empty

usermigallon, 2021-12-22:
Need I to install first a Debian firstly install on my sd card?

frank-w, 2021-12-22:
you can use another image too, but you need bootable image first...this repo is only linux-kernel which is loaded by bootchain (preloader,bootheaders,uboot) so this needs to be installed first

usermigallon, 2021-12-22:
ok, thank you very much for your help, I will install first Debian os

usermigallon, 2021-12-22:
I need Debian or Ubuntu?

frank-w, 2021-12-22:
my kernel / build-script is only prepared for this. e.g. openwrt uses another way to load kernel and there it cannot be replaced that easy (rebuild openwrt needed).

which system do you want to use?

usermigallon, 2021-12-22:
ok openwrt , I try to install , Debian, Ubuntu, openwrt and lede, but all are incomplete

usermigallon, 2021-12-22:
I install openwrt in sd card, and I havev same SD-Card not found!

frank-w, 2021-12-22:
Openwrt is not compatible as i've wrote above and it will miss my changes (e.g. internal wifi for r2, some patches for r64 like bluetooth/pcie),

debian is the most actual OS as i have not make a newer ubuntu image. If you want ubuntu you can replace rootfs from debian-image with bootstrapped ubuntu, change rootpw and make some small changes like i wrote down in my wiki

usermigallon, 2021-12-22:
So I install Debian?

frank-w, 2021-12-22:
yes

frank-w, 2021-12-22:
which debian have you used? do you see the mountpoint /media/$USER/BPI-BOOT/BPI-ROOT?

usermigallon, 2021-12-22:
if I use this ./build.sh /media/user/a5dcd082-db07-4b55-a1bc-4193f3680cdf/ do not work too 

./build.sh /media/resu/a5dcd082-db07-4b55-a1bc-4193f3680cdf/
unknown command /media/resu/a5dcd082-db07-4b55-a1bc-4193f3680cdf/

frank-w, 2021-12-22:
you cannot pass mountpoint to build.sh, only predefined commands (shown with ./build.sh help)

the mountpoint looks like it is not a fat-partition. as i've said, the script looks for /media/$USER/BPI-BOOT

https://github.com/frank-w/BPI-R2-4.14/blob/5.4-main/build.sh#L316

so make sure your boot-partition has this label (my images do this), but i'm not use other images have this

so just show your partition table

usermigallon, 2021-12-22:
This is my sd
[](url)https://ibb.co/0cJBbWr

usermigallon, 2021-12-22:
Maybe have you got a Debian to suggest ?

usermigallon, 2021-12-22:
I bought this card because there were 4 rj45 ports, but I regret this card it is good for the trash

frank-w, 2021-12-22:
your image still show no image flashed to card...you have to use dd or similar tool to write image to card...not copy file to the partition

i posted link to my last debian image here: 
https://github.com/frank-w/BPI-R2-4.14/issues/87#issuecomment-999491218

usermigallon, 2021-12-22:
I use wrt3200acm, and I'm glad, I bought banana pi-r2 to be able to hack, but there is not a lot of documentation and it is not very clear.
Now I have this out put 
uImage:
/media/resu/BPI-BOOT/bananapi/bpi-r2/linux/uImage_5.4.150-main

by default this kernel-/dtb-file will be loaded (kernel-var in uEnv.txt):
it must take a while ?

usermigallon, 2021-12-22:
it's normal that it always stays in ?
by default this kernel-/dtb-file will be loaded (kernel-var in uEnv.txt):

frank-w, 2021-12-22:
no, maybe a bug....do you have a uEnv.txt in the linux-folder? if not create one and add

`kernel=uImage_5.4.150-main`

usermigallon, 2021-12-22:
No, I don't have uEnv.txt 
It ask me this at start up
install kernel with DT [yn]? y 
install kernel with separate DT [yn]? n
It right?

usermigallon, 2021-12-22:
The image I have download  [https://drive.google.com/file/d/1VbV_IaUy92p1bIrd74sahs77LQNSQEVd/view?usp=sharing](url) do not work.

frank-w, 2021-12-22:
then i missed creating an empty uEnv.txt in the image (it's not needed if there is no change compared to uboot builtin environment). but build.sh should create one if none exists.

yes for r2 default is kernel with builtin devicetree...else it needs to loaded separately that is a bit more complex....this is for experimenting with dt overlays.

usermigallon, 2021-12-22:
I try your image [https://drive.google.com/file/d/1VbV_IaUy92p1bIrd74sahs77LQNSQEVd/view?usp=sharing](url) 
without modifying any thing and do not work.

frank-w, 2021-12-22:
you have it unpacked (gunzip) and written img-file to sdcard (not to partition) with dd??

usermigallon, 2021-12-22:
1°) I use sd card connected to laptop, I have uart-cable but I do not use
 [https://www.aliexpress.com/item/33000531104.html?spm=a2g0o.productlist.0.0.74f230adrMGnv6&algo_pvid=0705413a-ffbe-4cc2-90c0-bced790dc3b6&algo_exp_id=0705413a-ffbe-4cc2-90c0-bced790dc3b6-9&pdp_ext_f=%7B%22sku_id%22%3A%2266952889056%22%7D](url) 
2°) I done `sudo dd if=bpi-r2-buster_5.4.58.img of=/dev/sde bs=4M status=progress`
3°) Do not work scan my network with angry scan Ubuntu, lights of my rj45 connection of my  bpi-r2 are  blinking but not any ip adress.

frank-w, 2021-12-22:
with an uart cable you are able to see anything if something not work ;)

i'm not sure which network-config i used in buster image...you can look/change in BPI-ROOT/etc/network/interfaces

frank-w, 2021-12-22:
have you got it working? wrong subnet of ethernet?

usermigallon, 2021-12-22:
I don't know  to use uart cable, can you give me tuto link?

usermigallon, 2021-12-22:
No, it doesn't work

frank-w, 2021-12-22:
https://www.fw-web.de/dokuwiki/doku.php?id=en:bpi-r2:debug-uart

usermigallon, 2021-12-22:
I will test tomorrow, and I will tell you what happen, thank you for your patient

usermigallon, 2021-12-23:
Hello, I just did what is in your tutorial and it works, it was an ip problem, but still have the same problem building, Have you got and idea ? [http://pastebin.fr/99695](url)

frank-w, 2021-12-23:
build looks successful...you mean the hang on display current kernel loaded? need to figure out why the hang occours basicly it is only a grep that should not hang

usermigallon, 2021-12-23:
it stops with the build and it puts
`uImage:
/media/resu/BPI-BOOT/bananapi/bpi-r2/linux/uImage_5.4.150-main

by default this kernel-/dtb-file will be loaded (kernel-var in uEnv.txt):`

frank-w, 2021-12-23:
it's no more build...it's after install, you can try to add an (empty) uEnv.txt in /media/user/BPI-BOOT/bananapi/bpi-r2/linux/

add line to this file:

kernel=uImage_5.4.150-main

usermigallon, 2021-12-23:
Works, thank you so much Franck

frank-w, 2021-12-23:
could you try if this build.sh fixes the problem?

https://github.com/frank-w/BPI-R2-4.14/blob/5.16-rc/build.sh

just rename the uEnv.txt you've created

usermigallon, 2021-12-25:
OK

closed 2023-03-04 by frank-w