frank-w/BPI-Router-Linux #88: Debian package > 5.4 have no kernet with dtb

Josue-T, 2022-02-01:
Hello,

While I was tring to upgrade my kernel from 5.4 to 5.10 I detected that in the debian package there are no kernel image with dtb. So in `/boot/bananapi/bpi-r2/linux/` I only have this file `uImage_nodt-5.10.90-bpi-r2-main` and the `dtb` directory.

So is it normal that we don't have the image with dtb ? And if yes how can I start with this image from uboot ?

frank-w, 2022-02-01:
do you use the deb from the choice after build or the pack_debs?

the deb-packing after choice should work

https://github.com/frank-w/BPI-R2-4.14/blob/5.10-main/build.sh#L535

the other is defined in the kbuild system and a bit more complex

https://github.com/frank-w/BPI-R2-4.14/blob/5.10-main/scripts/package/builddeb#L143

here the uImage with dtb and the dtb itself for the other uImage_nodt is missing

Josue-T, 2022-02-01:
Hello,

Thanks for your answer. Well I don't really know how you build the package, the package which has the issue is the package that I got here: https://github.com/frank-w/BPI-R2-4.14/releases/tag/CI-BUILD-5.10-main-5.10.90-2022-01-13_0818

Igot this file: linux-image-5.10.90-bpi-r2-main_5.10.90-bpi-r2-main-1_armhf.deb

frank-w, 2022-02-01:
Ok,you do not build yourself and use deb from releases...this is built with pack_debs.

I will upload fix tomorrow

Josue-T, 2022-02-02:
Thanks for the fix, and thanks a lot for all your work to maintain this kernel.

closed 2022-02-02 by Josue-T

frank-w, 2022-02-02:
last release contains these for boot partition
```
drwxr-xr-x root/root         0 2022-02-02 07:39 ./boot/bananapi/bpi-r2/linux/dtb/
-rw-r--r-- root/root     44389 2022-02-02 07:39 ./boot/bananapi/bpi-r2/linux/dtb/bpi-r2-5.10.90-bpi-r2-main.dtb
-rw-r--r-- root/root   8245660 2022-02-02 07:39 ./boot/bananapi/bpi-r2/linux/uImage_5.10.90-bpi-r2-main
-rw-r--r-- root/root   8210888 2022-02-02 07:39 ./boot/bananapi/bpi-r2/linux/uImage_nodt-5.10.90-bpi-r2-main
```

i wonder why your dtb directory was empty...

thx for report :)