frank-w/BPI-Router-Linux #90: Initramfs and bootargs support?

YannikSc, 2022-07-04:
Hello,

I'm currently trying to setup an alpine with a custom initramfs and with mainline u-boot.

So far I got u-boot to boot the kernel. The kernel however seems to not get any boot args, like `root=` and also the initramfs is not entered at all.

I've tried the `5.10-main` and `5.15-main` branches. Both running into a Kernel Panic, as no root is set/can be mounted.

Is this something hardcoded in the kernel to prevent problems caused by (faulty) bootloaders (I read, that sometime this can be a problem) or is it that u-boot, not passing the args properly the the kernel and not correctly reading the initramfs?

I built the kernel with the default configs and even tried to embed the initramfs, all without success. The default config seems to be correctly configured to support this however.

frank-w, 2022-08-10:
Imho you have to load your initrd and set root to /dev/ram

This is how i did for r3 (uboot before loading kernel):

```
BPI-R3> setenv rdaddr 0x48000000
BPI-R3> setenv root "/dev/ram0 rw"
BPI-R3> setenv bootopts "debug=7 initrd=${rdaddr},20M"
BPI-R3> fatload usb 0:1 ${rdaddr} rootfs.cpio.zst
```

frank-w, 2023-03-04:
No response from user,closing

closed 2023-03-04 by frank-w