frank-w/BPI-Router-Linux #94: Linux ACL's (Access Control Lists) does not work for ext4 for some reason.

kgara, 2022-10-07:
Have no idea why ACL's does not work, but it is kinda might be useful for network storage part of f-ty.
```
# dmesg | grep acl
...
[1334842.283671] EXT4-fs (sda1): acl option not supported
[1334842.356809] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts: acl
mxbpir201 renard # uname -a
Linux bpir201 4.14.27-bpi-r2-main #1 SMP Mon Mar 19 10:34:41 EET 2018 armv7l GNU/Linux
# setfacl -R -m group:ghome:rwX /mnt/md3tlocal/tmp/
setfacl: /mnt/md3tlocal/tmp/: Operation not supported
```
Any ideas? Kernel seems to be build with the support of EXT4 ACL's. Is it? 
```
$ ./build.sh importconfig
$ cat .config | grep _ACL
CONFIG_EXT4_FS_POSIX_ACL=y
# CONFIG_XFS_POSIX_ACL is not set
# CONFIG_BTRFS_FS_POSIX_ACL is not set
CONFIG_F2FS_FS_POSIX_ACL=y
CONFIG_FS_POSIX_ACL=y
CONFIG_TMPFS_POSIX_ACL=y
CONFIG_NFS_V3_ACL=y
CONFIG_NFSD_V2_ACL=y
CONFIG_NFSD_V3_ACL=y
CONFIG_NFS_ACL_SUPPORT=y
```
Thanks in advance.

frank-w, 2022-10-07:
Have not done anything with acls yet. I would look for ext4 in config...maybe another option is needed for it

kgara, 2022-10-15:
Any ideas? So far have to use another sbc nearby just for the sake of acl's...

frank-w, 2022-10-15:
acl does not depend on hardware....if you have another sbc/pc running with acl just compare kernel-config to get the possible option missing in my config

zcat /proc/config.gz

frank-w, 2023-04-13:
tested on my debian-bullseye system
```

root@bpi-r2:~# uname -a                                                         
Linux bpi-r2 6.3.0-rc1-bpi-r2-sfp #1 SMP Tue Mar  7 20:03:39 CET 2023 armv7l GNU
/Linux
root@bpi-r2:~# zgrep -i 'acl' /proc/config.gz                                   
# CONFIG_XILINX_EMACLITE is not set                                             
CONFIG_EXT4_FS_POSIX_ACL=y                                                      
# CONFIG_XFS_POSIX_ACL is not set                                               
# CONFIG_BTRFS_FS_POSIX_ACL is not set                                          
CONFIG_F2FS_FS_POSIX_ACL=y                                                      
CONFIG_FS_POSIX_ACL=y                                                           
CONFIG_TMPFS_POSIX_ACL=y                                                        
CONFIG_NFS_V3_ACL=y                                                             
CONFIG_NFSD_V3_ACL=y                                                            
CONFIG_NFS_ACL_SUPPORT=y
root@bpi-r2:~# apt install acl
root@bpi-r2:~# mount -o remount,acl /                                           
[   72.885881] EXT4-fs (mmcblk0p2): re-mounted 83746542-38aa-4ab4-ae7e-722e5775.

root@bpi-r2:~# setfacl -R -m group:users:rwX /system/etc/firmware/              
root@bpi-r2:~# getfacl /system/etc/firmware/                                    
getfacl: Removing leading '/' from absolute path names                          
# file: system/etc/firmware/                                                    
# owner: root                                                                   
# group: root                                                                   
user::rwx                                                                       
group::r-x                                                                      
group:users:rwx                                                                 
mask::rwx                                                                       
other::r-x 
```

maybe you need to set acl filesystem-option for your mountpoint (/mnt/md3tlocal/tmp/)? or maybe tmp directory is a tmpfs??

closed 2023-08-27 by frank-w