frank-w/BPI-Router-Linux #96: cryptodev not available on kernel 6.1

Josue-T, 2022-12-23:
Hello,

I discovered that cryptodev module was not enabled on kernel 6.1. Is it normal ? And is it possible to enable it ?

Thanks

frank-w, 2022-12-23:
Cryptodev needs additional work and 6.1 is really fresh (wanted to get kernel itself working for all devices first) :) just have not found time to add it (needs external code added).

Do you need it for r2 or other device?

As a note: 6.1 for r2 does not support internal wifi

frank-w, 2022-12-24:
can you please test branch 6.1-cryptodev?

at least for r2 compile works using

```
./build.sh importconfig
./build.sh
./build.sh cryptodev
```

Josue-T, 2022-12-24:
Hello,

Thanks for the answer. Yes it's for R2 (only).

Well ok for the version 6.1, and which kernel version support cryptodev ?

frank-w, 2022-12-24:
Why is it complicated? It is similar to older kernels...clone repo,change branch to 6.1-cryptodev and then build kernel and cryptodev module,install to your r2 and test it. I have no experience with cryptodev itself,only added code and make it compile

Josue-T, 2022-12-24:
oh sorry, it's just because I'm too busy these time.

frank-w, 2023-03-04:
Have done tests with it?

frank-w, 2023-10-02:
No response

closed 2023-10-02 by frank-w