frank-w/BPI-Router-Linux #97: iptables not available on kernel 6.1

gustavobsch, 2023-01-22:
Hi,

How can I add iptables support to mainline 6.1 kernel when building?

Thanks!

frank-w, 2023-01-22:
basicly

```
./build.sh importconfig # import my default config for the board defined in build.conf
./build.sh config #select iptables there (menuconfig)
./build.sh
```

gustavobsch, 2023-01-22:
Thanks, after building I'm getting the following error. I'm using the docker environment 

===========================================
1) pack
2) install to SD-Card
3) deb-package
4) upload
choice [1234]:1
Pack Kernel to Archive
generate branch vars...
getting git branch: 
* 6.1-main
kernbranch:6.1-main,gitbranch:-main
cleanup...
copy...
INSTALL_MOD_PATH: ../SD/BPI-ROOT/
make[1]: Entering directory '/home/builder/build'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

/home/builder/BPI-Router-Linux/Makefile:741: include/config/auto.conf: No such file or directory
make[1]: *** [/home/builder/BPI-Router-Linux/Makefile:806: include/config/auto.conf] Error 1
make[1]: Leaving directory '/home/builder/build'
make: *** [Makefile:231: __sub-make] Error 2
find: '../SD/BPI-ROOT/lib/modules/*': No such file or directory
WMT-Tools not available
pack...
-rw-rw-r-- 1 builder builder 17M Jan 22 14:15 /home/builder/SD/bpi-r3_6.1.0-main.tar.gz

frank-w, 2023-01-22:
try commenting out these 2 options in build.conf by puppting # in front of these lines:
```
builddir=../build
ramdisksize=4G
```

gustavobsch, 2023-01-22:
Thank you, that worked!

closed 2023-01-22 by gustavobsch