frank-w/BPI-Router-Linux #100: bpi-r3 eth1 dropping packets

gustavobsch, 2023-01-24:
Hi, 

I'm running 6.2 kernel so I can use VLAN on the eth1 Fiber SFP and I'm observing a lot of packets getting dropped. Only the received ones. The SFP and Fiber were moved from another router that it was working just fine. So I don't think it's the tranceiver or fiber cable.

Has anyone else seen similar behavior on the bpi-r3? 

# ifconfig eth1
eth1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::9065:f3ff:feec:b019  prefixlen 64  scopeid 0x20<link>
        ether 92:65:f3:ec:b0:19  txqueuelen 1000  (Ethernet)
        RX packets 11132122  bytes 10885212987 (10.1 GiB)
        RX errors 1  **dropped 875168**  overruns 0  frame 13784        <<<<<<< 875168 dropped packets
        TX packets 9577892  bytes 11333127550 (10.5 GiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 123  

Thanks!

frank-w, 2023-01-24:
I'm still doing basic tests and noticed that vlan aware bridge is not working on mt7531.

Only did ping tests with latest patches (series from felix). Have not yet seen packet drops.

Btw. I'm alone here,it will be better posting such questions in bananapi forum

gustavobsch, 2023-01-24:
Thank you Frank, for now I will use another port for trunking. I realize this is a new board

closed 2023-01-24 by gustavobsch

frank-w, 2023-01-27:
i see you have no rx_flow_control_packets...maybe your other side/switch needs to enable pause??

gustavobsch, 2023-01-28:
I did try enabling flow control on the other side, I also tried a different device and get the same result, lots of dropped packets. For now I will not use the SFP1