frank-w/BPI-Router-Linux #101: 6.1-main deb package is not compatible with Debian Bullseye

martinl, 2023-03-04:
```
dpkg --contents linux-image-6.1.12-bpi-r2-main_6.1.12-bpi-r2-main-2_armhf.deb
dpkg-deb: error: archive 'linux-image-6.1.12-bpi-r2-main_6.1.12-bpi-r2-main-2_armhf.deb' uses unknown compression for member 'control.tar.zst', giving up
```
Looks like zstd support was added in Debian 12:
https://unix.stackexchange.com/questions/669004/zst-compression-not-supported-by-apt-dpkg

frank-w, 2023-03-04:
Thanks for reporting,any idea how to fix it? As the debs are created by kernel build-chain i don't know how to change the compression

martinl, 2023-03-04:
Looks like the issue is that github workflow is using ubuntu to build the deb packages and ubuntu deb/dpkg is using zst by default.
Here are some recommendations how to change that, either by using command line or debian rules file:

https://stackoverflow.com/questions/71196872/how-do-i-build-a-deb-without-zstd-compression-in-ubuntu-21-10

Are the kernel deb packages built using build.sh pack_debs or before that?

martinl, 2023-03-04:
I'm currently looking into getting bpi-r2 support in armbian back to shape (without internal bt/wifi) and I have 5.10 running with u-boot 2022.07 from sd (https://github.com/armbian/build/pull/4873)

Armbian build system can build 6.1.x debs for bpi-r2 but it currently hangs at boot so I ended up here to find out how to fix it

frank-w, 2023-03-04:
if you use the deb package from github-releases this is done by pack-deps

https://github.com/frank-w/BPI-Router-Linux/blob/6.1-main/.github/workflows/build.yml#L76

i do not create the debian-rules file...all is done by kernel build chain...so we need to find a way to pass this to dpkg

```
override_dh_builddeb:
    dh_builddeb -- -Zgzip
```

i found this, maybe we can change it by some env variable:

scripts/package/builddeb:53:	dpkg-deb $dpkg_deb_opts ${KDEB_COMPRESS:+-Z$KDEB_COMPRESS} --build "$pdir" ..

ejka, 2023-03-04:
builddeb is only called in intdeb-pkg target and you use bindeb-pkg. To make bindeb-pkg work you need to patch scripts/package/mkdebian to add those two override lines to generated debian/rules.

frank-w, 2023-03-04:
could you try my way? i try to avoid patching kernel-source here

frank-w, 2023-03-04:
Verified older 6.1.12 main built by github actions has zst and last build has tar.gz files inside. So my way works too :)

closed 2023-03-04 by frank-w