frank-w/BPI-Router-Linux #102: mt7530 doesn't work on bpi-r3 if mtk_eth is built as module

ejka, 2023-03-09:
When both mt7530 and mtk_eth are built as modules, probing for mt7530 fails with error:
```
mtk_soc_eth 15100000.ethernet: mdio: MDIO timeout
mdio_bus mdio-bus: failed to read mt7530 register
mtk_soc_eth 15100000.ethernet: mdio: MDIO timeout
mdio_bus mdio-bus: failed to read mt7530 register
mt7530 mdio-bus:1f: chip ffff can't be supported
```
If mtk_eth is changed to built-in, everything works fine.
Tested on 6.1-main and 6.3-rc branches.
Any ideas how to debug this further?

frank-w, 2023-03-09:
You can add printk in code,but why do you want eth and switch driver as module? Device is useless when you have no real rootfs (initrd).

Else you should report this on kernel mailinglist...at least if 6.3-rc1 (without additional patches) is affected too

ejka, 2023-03-11:
I have found the root cause of the issue. Clock gates in ethsys can't be enabled, once they have been disabled and mtk_eth doesn't work. I've sent a mail describing problem to linux-mediatek. Not sure if it have reached the list yet, as I can't subscribe to it.
For now I use the following patch locally to prevent kernel from disabling the clocks:
```
diff --git a/drivers/clk/mediatek/clk-mt7986-eth.c b/drivers/clk/mediatek/clk-mt7986-eth.c
index 703872239ecc..a368426eddd7 100644
--- a/drivers/clk/mediatek/clk-mt7986-eth.c
+++ b/drivers/clk/mediatek/clk-mt7986-eth.c
@@ -62,19 +62,20 @@ static const struct mtk_gate_regs eth_cg_regs = {
 	.sta_ofs = 0x30,
 };
 
-#define GATE_ETH(_id, _name, _parent, _shift)                                  \
+#define GATE_ETH(_id, _name, _parent, _shift, _flags)	                        \
 	{                                                                      \
 		.id = _id, .name = _name, .parent_name = _parent,              \
 		.regs = &eth_cg_regs, .shift = _shift,                         \
 		.ops = &mtk_clk_gate_ops_no_setclr_inv,                        \
+		.flags = _flags,                                               \
 	}
 
 static const struct mtk_gate eth_clks[] __initconst = {
-	GATE_ETH(CLK_ETH_FE_EN, "eth_fe_en", "netsys_2x_sel", 6),
-	GATE_ETH(CLK_ETH_GP2_EN, "eth_gp2_en", "sgm_325m_sel", 7),
-	GATE_ETH(CLK_ETH_GP1_EN, "eth_gp1_en", "sgm_325m_sel", 8),
-	GATE_ETH(CLK_ETH_WOCPU1_EN, "eth_wocpu1_en", "netsys_mcu_sel", 14),
-	GATE_ETH(CLK_ETH_WOCPU0_EN, "eth_wocpu0_en", "netsys_mcu_sel", 15),
+	GATE_ETH(CLK_ETH_FE_EN, "eth_fe_en", "netsys_2x_sel", 6, CLK_IS_CRITICAL),
+	GATE_ETH(CLK_ETH_GP2_EN, "eth_gp2_en", "sgm_325m_sel", 7, CLK_IS_CRITICAL),
+	GATE_ETH(CLK_ETH_GP1_EN, "eth_gp1_en", "sgm_325m_sel", 8, CLK_IS_CRITICAL),
+	GATE_ETH(CLK_ETH_WOCPU1_EN, "eth_wocpu1_en", "netsys_mcu_sel", 14, CLK_IS_CRITICAL),
+	GATE_ETH(CLK_ETH_WOCPU0_EN, "eth_wocpu0_en", "netsys_mcu_sel", 15, CLK_IS_CRITICAL),
 };
 
 static void __init mtk_sgmiisys_0_init(struct device_node *node)
```
Not sure if it is correct way though as I don't have hardware documentation.

frank-w, 2023-03-11:
thx, but please send this to linux kernel mailinglist (scripts/get_maintainer.pl drivers/clk/mediatek/clk-mt7986-eth.c) and let discussion begin :)

frank-w, 2023-09-16:
@ejka have you send patch to mailing-list? looks like this is really a mt7986-only problem, so i guess you found the right solution

jcdutton, 2024-04-08:
Ah, if this is due to clocks being disabled after boot, there is a boot param to stop them being disabled. One can use the boot param until the driver supports enabling the clock it needs. So, I don't think one really needs to add critical to them.
Boot param is “clk_ignore_unused”.

frank-w, 2026-01-18:
My kernel is not openwrt...i have mt7530 builtin and not as module.

EmreYavuzalp, 2026-01-18:
> My kernel is not openwrt...i have mt7530 builtin and not as module.

I also have mt7530 as a built-in. Can't have it other way it's a switch.