frank-w/BPI-Router-Linux #103: bpi-r3 branch 6.3-rc missing SFP support

gustavobsch, 2023-04-08:
Hi Frank,

Testing the 6.3-rc branch I noticed the SFP's stopped working, any fix for this?

Thanks

frank-w, 2023-04-08:
which SFP do you use? which issue is with this?

there are several patches for sfp-support (mostly quirks for 2.5G rj45 sfp)

https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net-next.git/log/drivers/net/phy/sfp.c

network-stack is currently broken...but there are already fixes for it

#throughput for r3-gmac (wan sfp)
https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net-next.git/commit/drivers/net/ethernet/mediatek/mtk_eth_soc.c?id=193250ace270fecd586dd2d0dfbd9cbd2ade977f
https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net-next.git/commit/drivers/net/ethernet/mediatek/mtk_eth_soc.c?id=07b3af42d8d528374d4f42d688bae86eeb30831a

#mt7623 switch-throughput
https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net-next.git/commit/drivers/net/ethernet/mediatek/mtk_eth_soc.c?id=e669ce46740a9815953bb4452a6bc5a7fdc21a50

frank-w, 2023-04-12:
which SFP do you use exactly?

c6869008cc05 2023-03-14 net: sfp: support 2.5g copper SFP v3

should be make 2g5 sfp from aliexpress work...if you have the tplink then daniel prepared a quirk-patch for it

gustavobsch, 2023-04-12:
It's the tplink from aliexpress model TL-SM410U

I will give the 6.3-rc another try

frank-w, 2023-04-12:
Ok,the tp-link needs patch from daniel i have not yet in my tree...

gustavobsch, 2023-04-12:
Oh ok, got confused, do you have plans to add Daniels patch to your 6.3 tree?

frank-w, 2023-04-12:
have added the tp-link quirk, but daniel reports that it is not clean working and they search the issue

https://www.spinics.net/lists/netdev/msg892902.html

gustavobsch, 2023-04-12:
Thanks for the tp-link quirk patch, the SFP is working now in 6.3-rc. 

I will test the 6.2-main now, I noticed the fiber SFP is not detecting link.

gustavobsch, 2023-04-12:
> have added the tp-link quirk, but daniel reports that it is not clean working and they search the issue
> 
> https://www.spinics.net/lists/netdev/msg892902.html
> 
> btw. there is 6.2-main...better than rc to use
> 
> but i wonder why 6.2-rc works for you....
> 
> maybe you can try to revert the sfp-patch above...possibly it breaks your sfps

I couldn't find the 6.2-main branch

frank-w, 2023-04-13:
which fibre sfp? mine (H!Fibre in left cage) is working in 6.3-rc

sorry about 6.3-main, had not pushed it...done now

gustavobsch, 2023-04-13:
Thanks, I'll pull 6.3-main.

I tested two, I can see them but link is not detected 

Brocade 57-1000013-01 
Cisco  GLC-SX-MM=

frank-w, 2023-04-13:
6.2-main...6.3 is not final

frank-w, 2023-04-13:
Yes,there was a patch series from felix fietkau which cannot be applied...thought patch from vladimir fixed it but maybe it was another issue...felix patch disabled hw offloading on the gmac1

frank-w, 2023-04-14:
This should be the patch fixing the vlan-issue on gmac

https://github.com/frank-w/BPI-Router-Linux/commit/13826888199e17314db49fb416ee797fbadf38e0

frank-w, 2023-04-16:
i have pushed the vlan-fix to 6.3-pwm-branch you can cherry-pick to any other branch you want...it works so far on my bananapi-r3 and i send it as RFC/RFT to ML

https://patchwork.kernel.org/project/linux-mediatek/patch/20230416091038.54479-1-linux@fw-web.de/

gustavobsch, 2023-04-16:
I just build and booted from it and I can see the vlan tags in the packets. Amazing, thanks!

frank-w, 2023-04-16:
I hope there will be no sideeffects...i guess the patch disables rx vlan offload completely but better than broken vlan support

gustavobsch, 2023-04-16:
I've ran into this kernel panic twice now. I need to reboot to clear it. You think it's related to the a

```
[20795.193278] ------------[ cut here ]------------
[20795.197920] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 3 timed out
[20795.204916] WARNING: CPU: 1 PID: 0 at net/sched/sch_generic.c:525 dev_watchdog+0x26c/0x274
[20795.213182] Modules linked in: pps_ldisc tun nft_masq nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables libcrc32c nfnetlink pps_gpio mt7915e mt76_connac_lib mt76 mac80211 libarc4 cfg80211 fuse ip_tables x_tables
[20795.234045] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 6.3.0-rc6-bpi-r3-pwm #3
[20795.241167] Hardware name: Bananapi BPI-R3 (DT)
[20795.245686] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[20795.252633] pc : dev_watchdog+0x26c/0x274
[20795.256638] lr : dev_watchdog+0x26c/0x274
[20795.260637] sp : ffffffc00800be00
[20795.263938] x29: ffffffc00800be00 x28: ffffffc009426000 x27: ffffffc00800bee8
[20795.271061] x26: ffffffc009106008 x25: 0000000000000000 x24: ffffffc0094288c8
[20795.278183] x23: 0000000000000100 x22: ffffffc009426000 x21: 0000000000000003
[20795.285306] x20: ffffff8001e31000 x19: ffffff8001e31488 x18: ffffffffffffffff
[20795.292427] x17: ffffffc076a73000 x16: ffffffc008008000 x15: 076e076107720774
[20795.299550] x14: 0720073a07290768 x13: 0000000000000255 x12: 00000000ffffffea
[20795.306673] x11: 00000000ffffefff x10: 00000000ffffefff x9 : ffffffc00949a438
[20795.313796] x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : 0000000000000001
[20795.320921] x5 : 0000000000000002 x4 : 0000000000000000 x3 : 0000000000000002
[20795.328046] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffff8000134380
[20795.335168] Call trace:
[20795.337604]  dev_watchdog+0x26c/0x274
[20795.341262]  call_timer_fn+0x34/0x150
[20795.344917]  __run_timers.part.0+0x2f4/0x358
[20795.349177]  run_timer_softirq+0x3c/0x74
[20795.353089]  _stext+0x118/0x344
[20795.356220]  ____do_softirq+0x10/0x1c
[20795.359872]  call_on_irq_stack+0x24/0x4c
[20795.363782]  do_softirq_own_stack+0x1c/0x2c
[20795.367953]  __irq_exit_rcu+0xe0/0xfc
[20795.371606]  irq_exit_rcu+0x10/0x1c
[20795.375083]  el1_interrupt+0x38/0x54
[20795.378648]  el1h_64_irq_handler+0x18/0x24
[20795.382733]  el1h_64_irq+0x68/0x6c
[20795.386123]  default_idle_call+0x58/0x108
[20795.390121]  do_idle+0xa8/0x100
[20795.393253]  cpu_startup_entry+0x24/0x2c
[20795.397166]  secondary_start_kernel+0x128/0x14c
[20795.401686]  __secondary_switched+0xb8/0xbc
[20795.405859] ---[ end trace 0000000000000000 ]---
```

frank-w, 2023-04-17:
I think this is related to felix patch adding the transmi queues...but i have no idea how to fix...there is a issue on openwrt github repo

frank-w, 2023-04-17:
Do you know a way to trigger this bug?

frank-w, 2023-04-17:
Let issue open for tracing the transmit error

gustavobsch, 2023-04-17:
I don't, but it hasn't crashed today and yesterday when it crashed I was copying big files through SFP1/SFP2. 

I'll run iperf3 test and see if it crashes again

frank-w, 2023-04-17:
I have not seen it with iperf3 yet...thats my standard-test in both directions,but i guess the problem is when there are multiple streams which are spreaded over different tx queues over time

frank-w, 2023-04-19:
Have you seen this with 6.2 too? If not you can try the vlan-patch there and look if this is the cause but imho this cannot be the cause as this is not in tx path

gustavobsch, 2023-04-19:
I havent tested with 6.2, can I just checkout latest 6.2-main and build it?

I ran into the issue again yesterday and I think the issue has something to do with the right SFP, where I have the copper tplink SFP. 

I ran into it three times when traffic is flowing as shown below. 

```
                 ------------------>
       eth1                           lan4
-------------------------------------------------------
|   fiber/vlan   |   copper/untagged     |
-------------------------------------------------------
```

gustavobsch, 2023-04-19:
I can trigger it but unsure exactly how.. it just happens. I noticed it doesn't happen when I just stop using the right sfp2/lan4 

Both ends on both SFP's are 2.5g

I tried a copper 1g SFP on the right sfp2/lan4 and it wont take it seems like it only allows 2.5g modules
```

root@bpi-r3:~# dmesg
[ 2265.225821] sfp sfp-2: module AVAGO            ABCU-5700RZ      rev      sn AN1001A2CX       dc 100107  
[ 2286.381969] mt7530 mdio-bus:1f lan4: configuring for inband/2500base-x link mode
[ 2286.454562] mt7530 mdio-bus:1f lan4: validation with support 00,00000000,00000000,00000000 failed: -EINVAL
[ 2286.464391] sfp sfp-2: sfp_add_phy failed: -EINVAL
[ 2315.659816] mt7530 mdio-bus:1f lan4: configuring for inband/2500base-x link mode
[ 2315.798868] mt7530 mdio-bus:1f lan4: validation with support 00,00000000,00000000,00000000 failed: -EINVAL
[ 2315.808653] sfp sfp-2: sfp_add_phy failed: -EINVAL
root@bpi-r3:~# ethtool lan4
Settings for lan4:
	Supported ports: [ MII ]
	Supported link modes:   2500baseX/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: No
	Supported FEC modes: Not reported
	Advertised link modes:  2500baseX/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: No
	Advertised FEC modes: Not reported
	Speed: Unknown!
	Duplex: Unknown! (255)
	Auto-negotiation: off
	Port: MII
	PHYAD: 0
	Transceiver: internal
	Supports Wake-on: d
	Wake-on: d
	Link detected: no

```

frank-w, 2023-04-20:
My fiber sfp works in both cages and this is 1g only ..maybe your 1g module needs a quirk too...daniel reported from some sfps having checksum errors or having invalid data (e.g. supported modes) in eeprom

frank-w, 2023-04-22:
Can you trigger the issue when only working over lan4/sfp2 (writing/reading to/from r3).

gustavobsch, 2023-04-28:
I loaded 6.3.0-bpi-r3-main 24 hours ago and no crashes. When I initially ran into this issue it could crash after couple hours.

frank-w, 2023-05-03:
Have you same connections as before (sfps,rj45 ports,vlan)? Have not seen any fix for these problems

gustavobsch, 2023-05-03:
Something definitely changed, but I can't tell what.  
The SFP's and the cages where they are inserted hasn't changed.
It hasn't crashed since I reported the initial crash so I'm happy

gustavobsch, 2024-04-23:
Hi Frank,
 
Can you add the tp-link quirk patch to  6.6.25-bpi-r3-main? 

Thanks
Gustavo

frank-w, 2024-04-24:
Can you make a pull request or point to the patch?

gustavobsch, 2024-04-24:
I'm actually not sure that patch has the fix. The issue this time is the SFP not detecting link. 

[246390.041031] sfp sfp-2: module TP-LINK          TL-SM410U        rev 1.0  sn 121B0Y9000456    dc 211108  
[246390.050640] mt7530-mdio mdio-bus:1f lan4: unsupported SFP module: no common interface modes
root@bpi-r3:~# ethtool lan4
Settings for lan4:
	Supported ports: [ MII ]
	Supported link modes:   2500baseX/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  2500baseX/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Speed: Unknown!
	Duplex: Unknown! (255)
	Auto-negotiation: on
	Port: MII
	PHYAD: 0
	Transceiver: internal
	Supports Wake-on: d
	Wake-on: d
	Link detected: no

frank-w, 2024-04-24:
This looks like the main problem for all 2g5 sfp...look what i have done in 6.9-rc...i guess you need at least erics phylink patch and the tplink quirk let phylink know about 2g5...the phy-mapping i had done for the oem imho does not work for tplink