frank-w/BPI-Router-Linux #104: BPI-r3 expose pps

gustavobsch, 2023-04-12:
I want to setup a gps disciplined ntp server and I'm wondering how to setup a gpio to fee pps to on the bpi-r3 board.

I can see the kernel has pps and gpio modules builtin but  I think I need to setup an spcific pps gpio pin 'pps_gpio_pin' 

I an old thread where explaining how to do it on kernel 4.9 in your wiki below.

https://www.fw-web.de/dokuwiki/doku.php?id=en:bpi-r2:gpio

How can I make this process work with your kernel 6.2 or 6.3?

Thanks

frank-w, 2023-04-13:
I have not tested pps only added description from forum user to my wiki....

gustavobsch, 2023-04-13:
I tried adding the section below under gpio-keys to the file arch/arm64/boot/dts/mediatek/mt7986a-bananapi-bpi-r3.dts before compiling. Is this the right way to do it? I couldnt find other dts file in the file system 

pps {
   pinctrl-names = “default”;
   compatible = “pps-gpio”;
   gpios = <&pio 15 0>;
   status = “okay”;
};

frank-w, 2023-04-13:
file is right, but you have to do it at top-level, not under gpio-keys

closed 2023-04-13 by gustavobsch

gustavobsch, 2023-04-14:
Any chance you can add GPIO 65 as PPS to the bpir3 device tree? this pin it's perfectly located next to UART2, 3v3 and GND which are used to to connect to the GPS  

`
	gpio-keys {
		compatible = "gpio-keys";

		reset-key {
			label = "reset";
			linux,code = <KEY_RESTART>;
			gpios = <&pio 9 GPIO_ACTIVE_LOW>;
		};

		wps-key {
			label = "wps";
			linux,code = <KEY_WPS_BUTTON>;
			gpios = <&pio 10 GPIO_ACTIVE_LOW>;
		};
	};

	pps {
	   pinctrl-names = "default";
	   compatible = "pps-gpio";
	   gpios = <&pio 15 0>;
	   status = "okay";
	}; `

frank-w, 2023-05-01:
sry, missed this one...very few users use a gps module here, so i would leave this unconfigured, but you can form my repo and add it by yourself ;) but you should use only main-branches, because the rc-branches are rebased from time to time and may create conflicts if you want to update your source from my tree