frank-w/BPI-Router-Linux #108: Wireguard support

K-Fuchs, 2023-08-10:
One thing I'm missing for my BPI-R3 is Wireguard support and sadly, the DKMS Module does not seem to work either. I saw some closed Github Issues about Wireguard that where resolved. Would it be possible to add Wiregaurd back into the kernel?

frank-w, 2023-08-11:
Which kernelversion donyou want to use (last lts = 6.1 or last stable =6.4)

Which options do you want to have enabled exactly?

K-Fuchs, 2023-08-11:
Thanks for the quick reply.
To activate wireguard and  wg-quick with nftables in the kernel i narrowed down the options to:

```makefile
CONFIG_NET_CORE=y
CONFIG_WIREGUARD=y
CONFIG_IPV6=y
CONFIG_NET_UDP_TUNNEL=y
CONFIG_DST_CACHE=y
CONFIG_CRYPTO=y
CONFIG_CRYPTO_LIB_CURVE25519=y
CONFIG_CRYPTO_LIB_CHACHA20POLY1305=y
CONFIG_IP_MULTIPLE_TABLES=y
CONFIG_IPV6_MULTIPLE_TABLES=y
CONFIG_IP6_NF_RAW=m
CONFIG_IP_NF_RAW=m
CONFIG_NFT_FIB=m
CONFIG_NFT_FIB_IPV4=m
CONFIG_NFT_FIB_IPV6=m
CONFIG_NETFILTER_XT_MATCH_CONNMARK=y
NETFILTER_XT_TARGET_CT=y
```

closed 2023-08-13 by K-Fuchs

goddardduncan, 2025-06-09:
This was really helpful in getting wireguard going! Thanks for leaving the solution for how you resolved it.