frank-w/BPI-Router-Linux #109: WiFi doesn't work on kernel 5.15

MastMind, 2023-08-17:
I'd built the kernel, u-boot, buildroot rootfs and successfully flashed sd card, but when I try to run wmt_loader I've got next error:

# wmt_loader
init combo devic[   47.498018] [WMT-DETECT][I]wmt_detect_open:open major 154 minor 0 (pid 229)
e
[   47.506340] [WMT-DETECT][I]wmt_detect_unlocked_ioctl:cmd (-2147191037),arg(0)
Opened combo device
get device id : [   47.515616] [WMT-DETECT][I]wmt_detect_unlocked_ioctl:cmd (1074034433),arg(30243)
30243
[   47.524499] set current consys chipid (0x7623)
set device id : [   47.529684] [WMT-DETECT][I]wmt_detect_unlocked_ioctl:cmd (-2147191036),arg(30243)
30243
[   47.538557] [WMT-MOD-INIT][I]do_common_drv_init:start to do common driver init, chipid:0x00007623
[   47.553261] [WMT-CONF][E]wmt_conf_parse_pair(323):unknown field 'mt6620.defAnt'.
[   47.560732] [WMT-CONF][E]wmt_conf_parse_pair(323):unknown field 'mt6628.defAnt'.
[   47.568167] [WMT-CONF][E]wmt_conf_parse_pair(323):unknown field 'mt6630.defAnt'.
[   47.578021] [WMT-MOD-INIT][I]do_common_drv_init:finish common driver init
[   47.584842] [WCN-MOD-INIT][E]do_connectivity_driver_init(58):do common driver init failed, ret:-1
[   47.593778] [WLAN-MOD-INIT][I]do_wlan_drv_init:CONFIG_MTK_COMBO_WIFI is not defined
do module init: 30243
failed to init module
[   47.607679] [WMT-DETECT][I]wmt_detect_close:close major 154 minor 0 (pid 229)

The loaded mt7615e.ko via modprobe does no changes.

frank-w, 2023-08-17:
You mean r2 internal wifi? Do you have wifi firmware and configs in your rootfs? Buildroot with wifi firmwares is tricky...i suggest a normal rootfs and using my wifi.sh

MastMind, 2023-08-17:
Yes, I meant r2 board. I have WMT_SOC.cfg in /system/etc/firmware and wmt_tools (wmt_loader, wmt_loopback, stp_uart_launcher) in /usr/bin and all firmwares in /etc/firmware. I guess I have all necessary files. Maybe I need to load special drivers via modprobe before starting do these steps: https://github.com/BPI-SINOVOIP/BPI-R2-bsp/blob/Wifi_debug_branch/vendor/mediatek/connectivity/readme

MastMind, 2023-08-17:
I'd tried to run wifi.sh, but it finished with error too

frank-w, 2023-08-17:
The wifi.sh loads cfg80211 and the wifi module,which needs of course to be part of the initrd. Maybe you could try to boot one of my ubuntu/debian-images which are more recent than the bpi images/kernel

MastMind, 2023-08-17:
I'd started ap0 interface. At first I'd loaded next kernel modules (my lsmod output):
# lsmod
Module                  Size  Used by    Not tainted
stp_chrdev_bt          24576  0
wlan_gen2            1556480  0
cfg80211              782336  1 wlan_gen2
wmt_chrdev_wifi        20480  1 wlan_gen2

After that I'd tried to run wmt_tools steps:
# wmt_loader
init combo device
Opened combo device
get device id : 30243
set device id : 30243
do module init: 30243
failed to init module
# stp_uart_launcher -p /etc/firmware &
# open device node succeed.(Node:/dev/stpwmt, fd:3)
chip id is 0x7623
chiId:0x7623, setting to property(D▒)
chiId:0x7623
run SOC chip flow
create pwr on thread ok
enter power on connsys flowpatch name pre-fix:ROMv2_lm_patch
fwVersion:0x8a00
/etc/firmware/ROMv2_lm_patch_1_1_hdr.bin
fw Ver in patch: 0x8a00
read patch info:0x21,0x00,0x0e,0xf0
gpatchnum = [2]
gdwonseq = [1]
/etc/firmware/ROMv2_lm_patch_1_0_hdr.bin
fw Ver in patch: 0x8a00
read patch info:0x22,0x00,0x06,0x00
gdwonseq = [2]
cmd(srh_patch) resp(ok)

#
# echo 1 > /dev/wmt
#

And only after these action I could run wifi.sh and ap0 interface has been created. But I did not test connection to a new AP point, because I did not make a good dhcp configuration for that. I hope it will works.

frank-w, 2023-08-17:
Yes,startup of this wifi is complex,but i still use it with 5.15 on my main-router so i want to know if you have a hardware issue first...that seem not to be the case. So eizher you miss any config or firmware file or anything in startup...if you use wifi.sh in your initrd only the files-case can be the case or a more recent kernel breaks. But i use a recent 5.15...

MastMind, 2023-08-17:
**My image should start dhcp server and hostapd,so it should be ready to use here**

Maybe this is my fault. I'd built rootfs from there: https://github.com/frank-w/buildroot.git with replacement of several packages. And I just copied output rootfs to BPI-ROOT

frank-w, 2023-08-17:
The initrd is maybe not complete to bring up the r2 specific (or any other) wifi. It is designed as initrd to make basic tests on a device like bootup filesystem tests and network performance tests.

For a complete system i suggest using debian/ubuntu as base and the install your packages you need.

I have the bpi-router-image repo which creates the very basic image and there you can install the wifi-related parts

MastMind, 2023-08-18:
I found the best way for bringing up an access point (without wifi.sh):

1. Run wmt_loader
2. stp_uart_launcher -p /etc/firmware &
3. modprobe wlan_gen2 (we need only that driver)
4. echo A >/dev/wmtWifi (turn ap0 interface into AP mode)

It works. I could set up NAT with iptables and now my BPI-R2 board works as a regular home WiFi router.

closed 2023-08-18 by MastMind