frank-w/BPI-Router-Linux #111: bpi-r2, kernel 5.15.129 - compilation errors (mediatek) - arm-linux-gnueabihf-gcc (Debian 13.2.0-2)

dto-23, 2023-10-02:
Hi,

compilation of source code fails with some errors:

drivers/misc/mediatek/connectivity/common/conn_soc/core/stp_core.c:1331:5: error: conflicting types for 'mtk_wcn_stp_psm_notify_stp' due to enum/integer mismatch; have 'int(const MTKSTP_PSM_ACTION_T)' [-Werror=enum-int-mismatch]
 1331 | int mtk_wcn_stp_psm_notify_stp(const MTKSTP_PSM_ACTION_T action)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from drivers/misc/mediatek/connectivity/common/conn_soc/core/stp_core.c:16:
./drivers/misc/mediatek/connectivity/common/conn_soc/core/include/stp_core.h:532:12: note: previous declaration of 'mtk_wcn_stp_psm_notify_stp' with type 'int(const UINT32)' {aka 'int(const unsigned int)'}
  532 | extern int mtk_wcn_stp_psm_notify_stp(const UINT32 action);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  fs/xfs/xfs_symlink.o
  CC [M]  net/netfilter/nft_rt.o
  CC [M]  net/netfilter/nft_exthdr.o
cc1: all warnings being treated as errors
make[7]: *** [scripts/Makefile.build:289: drivers/misc/mediatek/connectivity/common/conn_soc/core/stp_core.o] Error 1
make[6]: *** [scripts/Makefile.build:552: drivers/misc/mediatek/connectivity/common/conn_soc/core] Error 2
make[5]: *** [scripts/Makefile.build:552: drivers/misc/mediatek/connectivity/common/conn_soc] Error 2
make[4]: *** [scripts/Makefile.build:552: drivers/misc/mediatek/connectivity/common] Error 2
make[3]: *** [scripts/Makefile.build:552: drivers/misc/mediatek/connectivity] Error 2
make[2]: *** [scripts/Makefile.build:552: drivers/misc/mediatek] Error 2
make[1]: *** [scripts/Makefile.build:552: drivers/misc] Error 2
make: *** [Makefile:1905: drivers] Error 2

and

  CC [M]  drivers/misc/mediatek/connectivity/wlan/gen2/os/linux/gl_proc.o
In file included from ./include/asm-generic/preempt.h:5,
                 from ./arch/arm/include/generated/asm/preempt.h:1,
                 from ./include/linux/preempt.h:78,
                 from ./include/linux/spinlock.h:55,
                 from ./include/linux/debugobjects.h:6,
                 from ./include/linux/timer.h:8,
                 from ./drivers/misc/mediatek/connectivity/wlan/gen2/os/linux/include/gl_os.h:531,
                 from ./drivers/misc/mediatek/connectivity/wlan/gen2/include/precomp.h:171,
                 from drivers/misc/mediatek/connectivity/wlan/gen2/os/linux/gl_proc.c:65:
In function 'check_copy_size',
    inlined from 'copy_from_user' at ./include/linux/uaccess.h:191:6,
    inlined from 'procDbgLevelWrite' at drivers/misc/mediatek/connectivity/wlan/gen2/os/linux/gl_proc.c:511:6:
./include/linux/thread_info.h:227:25: error: call to '__bad_copy_to' declared with attribute error: copy destination size is too small
  227 |                         __bad_copy_to();
      |                         ^~~~~~~~~~~~~~~
In function 'check_copy_size',
    inlined from 'copy_from_user' at ./include/linux/uaccess.h:191:6,
    inlined from 'procTxDoneCfgWrite' at drivers/misc/mediatek/connectivity/wlan/gen2/os/linux/gl_proc.c:591:6:
./include/linux/thread_info.h:227:25: error: call to '__bad_copy_to' declared with attribute error: copy destination size is too small
  227 |                         __bad_copy_to();
      |                         ^~~~~~~~~~~~~~~

frank-w, 2023-10-02:
hi, which compiler (incl. version) do you use? gcc9.5 (ubuntu 22.04) is fine for me...driver for bpi-r2 internal wifi is a pain

dto-23, 2023-10-02:
;-) ...

Using gcc 13.2.0 on Debian 6.5.3 (Kali 2023.3) ....

frank-w, 2024-01-26:
sorry missed this issue, can you try older gcc? i do not support the r2 internal wifi driver anymore as long as i can compile it anyhow on an actual ubuntu system (currently 22.04 with gcc 11.4)

dto-23, 2024-02-03:
Okay, thanks a lot - compilation with (Debian) gcc 11.4 works (without errors and warnings).

closed 2024-02-03 by frank-w