frank-w/BPI-Router-Linux #114: i2c rtcs

mithrandir42, 2024-01-05:
Please enable the usual i2c (these could be attached to the router boards) real time clocks in defconfig. Thanks!

CONFIG_RTC_DRV_DS1307

There are further i2c rtc drivers, others might be interested in, i.e.:

CONFIG_RTC_DRV_M41T80
CONFIG_RTC_DRV_RV3029C2
CONFIG_RTC_DRV_RV3032
CONFIG_RTC_DRV_ISL1208
CONFIG_RTC_DRV_DS1672
CONFIG_RTC_DRV_FM3130
CONFIG_RTC_DRV_ABEOZ9
CONFIG_RTC_DRV_PCF85063
CONFIG_RTC_DRV_ISL12022
CONFIG_RTC_DRV_S35390A
CONFIG_RTC_DRV_MAX6900
CONFIG_RTC_DRV_ISL12026
CONFIG_RTC_DRV_RX8025
CONFIG_RTC_DRV_ABX80X
CONFIG_RTC_DRV_RV3028
CONFIG_RTC_DRV_PCF8523
CONFIG_RTC_DRV_RX8581
CONFIG_RTC_DRV_BQ32K
CONFIG_RTC_DRV_SD3078
CONFIG_RTC_DRV_PCF8563
CONFIG_RTC_DRV_PCF8523
CONFIG_RTC_DRV_RX8010
CONFIG_RTC_DRV_DS3232
CONFIG_RTC_DRV_HYM8563
CONFIG_RTC_DRV_PCF2127
CONFIG_RTC_DRV_ABB5ZES3
CONFIG_RTC_DRV_RS5C372
CONFIG_RTC_DRV_X1205
CONFIG_RTC_DRV_RV8803
CONFIG_RTC_DRV_EM3027
CONFIG_RTC_DRV_NCT3018Y
CONFIG_RTC_DRV_RX6110

Don't know, for which of these it might make sense to enable also.

frank-w, 2024-01-26:
i do not want to enable all available drivers, only the needed ones...i try to keep the defconfig as small as possible to reduce compile time for me (i have to compile kernel very often for tests across different kernel versions) and pipeline/actions. if you have a specific driver i can add it, but not all possible

closed 2024-01-26 by frank-w

mithrandir42, 2024-01-26:
Thanks, the needed one for me would be CONFIG_RTC_DRV_DS1307. Would be great if you could add it.