frank-w/BPI-Router-Linux #115: [BPI-R3][Kernel 6.1] LVM issue

guedz, 2024-01-26:
Hello,

When i try to create a LV i have an issue :

```bash
root@bpi-r3:~# vgs
  VG      #PV #LV #SN Attr   VSize   VFree  
  os        1   0   0 wz--n- <32.00g <32.00g
root@bpi-r3:~# lvs
root@bpi-r3:~# lvcreate -n opt os -L 1G
  /dev/mapper/control: open failed: No such device
  Failure to communicate with kernel device-mapper driver.
  Check that device-mapper is available in the kernel.
  Incompatible libdevmapper 1.02.185 (2022-05-18) and kernel driver (unknown version).
  striped: Required device-mapper target(s) not detected in your kernel.
  Run `lvcreate --help' for more information.
```

Have you an idea about this issue ?
Thanks for your help

Best regards

frank-w, 2024-01-26:
I guess i have not added related kernel options for r3 :)

frank-w, 2024-01-26:
Try adding these options

https://github.com/frank-w/BPI-Router-Linux/blob/6.1-main/arch/arm/configs/mt7623n_evb_fwu_defconfig#L384

To end of this file

https://github.com/frank-w/BPI-Router-Linux/blob/6.1-main/arch/arm64/configs/mt7986a_bpi-r3_defconfig

Running importconfig again and building