frank-w/BPI-Router-Linux #116: BPI-R3 Mini

mrbluecoat, 2024-02-12:
Quick question: does your R3 solution also work for the R3 Mini? https://wiki.banana-pi.org/Banana_Pi_BPI-R3_Mini  

What is the best way to install Debian on that device?

mrbluecoat, 2024-02-12:
One last update: looks like it's still in development: https://forum.banana-pi.org/t/bpi-r3-mini-review-and-mainline-immortalwrt-firmware-support/16526/56

Thanks for your efforts on this!

P.S. I was able to get Arch Linux loaded via https://github.com/ericwoud/buildR64arch in case that repo helps.

frank-w, 2024-02-12:
Have not included r3mini in my dependend pipelines (uboot and kernel) which are needed by imagebuilder. You can create r3 image and replace uboot+kernel manually you've built,or at least for kernel you can use the skipkerneldownload and use selfcompiled kernel as source

E.g. build kernel from 6.6-r3mini branch,and pack it to tar.gz. this file can be placed in the folder where you have cloned image-repo and then you create this config:

sourcefiles_bpi-r3.conf
```
skipkerneldownload=1
kernelfile=kernelfile.tar.gz
```
And build image for r3.

This you can flash to emmc of r3mini and then you need to replace uboot with emmc version.

One thing you need to change for uboot for r3mini:

Create a uEnv.txt in boot-partition with this content:

`setbootconf=setenv bootconf "#conf-emmc-mini"`

Have this already in uboot branch 2024-04-bpi but this is not yet built...

mrbluecoat, 2024-02-16:
> R3mini is currently only supported in specific r3mini branches as phy driver is still in development.

For those interested in watching the progress: https://forum.banana-pi.org/t/bpi-r3-mini-airoha-en8811h-2-5g-phy-driver/16590/144

This is way past my technical skills -- kudos to @frank-w and @ericwoud !

mrbluecoat, 2024-02-16:
Could we use the PHY drivers from the official OpenWRT image released today? https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=b03d3644cfe9eb8db2a3441c55c84b076ba7cd08

```
If you like to have Ethernet support inside U-Boot (eg. to boot via
TFTP) you also need to write the PHY firmware to /dev/mmcblk0boot1:
echo 0 > /sys/block/mmcblk0boot1/force_ro
dd if=/lib/firmware/airoha/EthMD32.dm.bin of=/dev/mmcblk0boot1
dd if=/lib/firmware/airoha/EthMD32.DSP.bin bs=16384 seek=1 of=/dev/mmcblk0boot1
```

frank-w, 2024-02-16:
My uboot-repo also have a r3mini branch where i have modified openwrt driver (with wifi-firmware builtin) but same issue that tftp only works once.

Flashing firmware to boot1 has not worked for me...flashed in uboot and got write error

frank-w, 2024-04-24:
I managed to flash firmware to boot0

https://forum.banana-pi.org/t/r3mini-uboot-getting-ethernet-nvme-to-work/17165/21?u=frank-w

Can this issue now be closed?

closed 2024-04-24 by mrbluecoat