frank-w/BPI-Router-Linux #119: Wifi advanced features do not work on your 6.9-rc

jcdutton, 2024-03-30:
Notice the hostapd.conf file extract below. All the lines that are commented out with #  are not recognised on 6.9-rc but are recognised on openwrt for the BPI-R3.
I don't know if this is a hostapd problem or a 6.9-rc problem.

driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=GB
ieee80211d=1
hw_mode=g
supported_rates=60 90 120 180 240 360 480 540
basic_rates=60 120 240
beacon_int=100
channel=acs_survey

#num_global_macaddr=1
ieee80211n=1
#ht_coex=0
ht_capab=[LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]
#ieee80211ax=1
#he_su_beamformer=1
#he_su_beamformee=1
#he_mu_beamformer=1
#he_bss_color=128
#he_spr_sr_control=3
#he_default_pe_duration=4
#he_rts_threshold=1023
#he_mu_edca_qos_info_param_count=0
#he_mu_edca_qos_info_q_ack=0
#he_mu_edca_qos_info_queue_request=0
#he_mu_edca_qos_info_txop_request=0
#he_mu_edca_ac_be_aifsn=8
#he_mu_edca_ac_be_aci=0
#he_mu_edca_ac_be_ecwmin=9
#he_mu_edca_ac_be_ecwmax=10
#he_mu_edca_ac_be_timer=255
#he_mu_edca_ac_bk_aifsn=15
#he_mu_edca_ac_bk_aci=1
#he_mu_edca_ac_bk_ecwmin=9
#he_mu_edca_ac_bk_ecwmax=10
#he_mu_edca_ac_bk_timer=255
#he_mu_edca_ac_vi_ecwmin=5
#he_mu_edca_ac_vi_ecwmax=7
#he_mu_edca_ac_vi_aifsn=5
#he_mu_edca_ac_vi_aci=2
#he_mu_edca_ac_vi_timer=255
#he_mu_edca_ac_vo_aifsn=5

frank-w, 2024-03-30:
He options are for ax mode and need afaik self compiled hostapd in debian based systems,see thread in bpi forum

https://forum.banana-pi.org/t/hostapd-wifi-speed-not-reach-the-maximum-in-debian11/17642/14?u=frank-w

jcdutton, 2024-03-30:
I compiled my own hostapd, and now it lets me uncomment "ieee80211ax=1"
It still does not like the rest of the parameters, with the kernel saying  they are invalid or out of range.
The wifi link is at least up now and forwarding packets.
My aim is to get your kernel, that is close to mainline, working with all the same parameters that openwrt can handle.

frank-w, 2024-03-30:
Maybe this helps: https://community.solid-run.com/t/hostapd-and-wifi-6e/586/10

frank-w, 2024-04-05:
Have you made progress here? How have you compiled hostapd (debian sources or crosscompile)?