frank-w/BPI-Router-Linux #121: mt7615 wifi doesn't work properly on kernel 5.15

MastMind, 2024-04-20:
Hi

I'd tried to extend wifi abilities of my bananapi-r2 board. For this purposes I bough an external PCIe wifi module mt7615. Also I'd compiled mt7615e driver as kernel module. I've got the next issue: when I load the kernel module everything is fine in dmesg output:

```
[  300.866646] mt7615e 0000:01:00.0: enabling device (0140 -> 0142)
[  300.872971] mt7615e 0000:01:00.0: Load eeprom: /lib/firmware/mediatek/mt7615e_rf.bin
[  300.881994] mt7615e 0000:01:00.0: Load eeprom OK, count 1200 byte
[  300.893024] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[  300.902178] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[  300.923577] mt7615e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20180518100604a
[  300.923577] 
[  300.989698] mt7615e 0000:01:00.0: N9 Firmware Version: _reserved_, Build Time: 20200814163649
[  301.009874] mt7615e 0000:01:00.0: CR4 Firmware Version: _reserved_, Build Time: 20181207140436
```

And I can see _wlan0_ and _wlan1_ interfaces in the list of networking devices. But when I try to bring up one of them (or just gather information with _iw list_ command) I've got an segfault error in command line and the next error messages in dmesg output:

```
[ 1604.251263] 8<--- cut here ---
[ 1604.254356] Unable to handle kernel NULL pointer dereference at virtual address 000000a6
[ 1604.262511] pgd = d82e545d
[ 1604.265232] [000000a6] *pgd=fb26f835
[ 1604.268827] Internal error: Oops: 817 [#1] SMP ARM
[ 1604.273626] Modules linked in: mt7615e mt7615_common mt76_connac_lib mt76 mac80211 libarc4 cfg80211 tun xt_nat xt_MASQUERADE iptable_nat nf_nat xt_tcpudp xt_state xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 iptable_filter ip_tables x_tables [last unloaded: mt7615e]
[ 1604.298243] CPU: 2 PID: 474 Comm: ip Not tainted 5.15.117-bpi-r2-main #4
[ 1604.304952] Hardware name: Mediatek Cortex-A7 (Device Tree)
[ 1604.310526] PC is at cfg80211_mgd_wext_connect+0x68/0x18c [cfg80211]
[ 1604.317800] LR is at rtnl_is_locked+0x20/0x24
[ 1604.322178] pc : [<bf0ec64c>]    lr : [<c0ac4400>]    psr: 20000013
[ 1604.328450] sp : c5b4f6f8  ip : c5b4f6e8  fp : c5b4f71c
[ 1604.333678] r10: 00000000  r9 : c5b4fbd8  r8 : c5b9262c
[ 1604.338906] r7 : c6168000  r6 : c6168000  r5 : 00000002  r4 : c5b92608
[ 1604.345440] r3 : c5b928d0  r2 : ffffffff  r1 : c5b92000  r0 : 00000001
[ 1604.351972] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[ 1604.359115] Control: 10c5387d  Table: 8614006a  DAC: 00000051
[ 1604.364864] Register r0 information: non-paged memory
[ 1604.369934] Register r1 information: slab kmalloc-cg-8k start c5b92000 pointer offset 0 size 8192
[ 1604.378843] Register r2 information: non-paged memory
[ 1604.383900] Register r3 information: slab kmalloc-cg-8k start c5b92000 pointer offset 2256 size 8192
[ 1604.393058] Register r4 information: slab kmalloc-cg-8k start c5b92000 pointer offset 1544 size 8192
[ 1604.402216] Register r5 information: non-paged memory
[ 1604.407272] Register r6 information: non-slab/vmalloc memory
[ 1604.412938] Register r7 information: non-slab/vmalloc memory
[ 1604.418602] Register r8 information: slab kmalloc-cg-8k start c5b92000 pointer offset 1580 size 8192
[ 1604.427760] Register r9 information: non-slab/vmalloc memory
[ 1604.433427] Register r10 information: NULL pointer
[ 1604.438221] Register r11 information: non-slab/vmalloc memory
[ 1604.443973] Register r12 information: non-slab/vmalloc memory
[ 1604.449724] Process ip (pid: 474, stack limit = 0xb98d6c7a)
[ 1604.455306] Stack: (0xc5b4f6f8 to 0xc5b50000)
[ 1604.459670] f6e0:                                                       c0aba79c c6168a74
[ 1604.467860] f700: 00000010 c5b92000 c5b92608 c61681c0 c5b4f8cc c5b4f720 bf0844ec bf0ec5f0
[ 1604.476049] f720: c0149c94 c0149700 c5b92000 c6168520 c6168a70 c5b92188 c5b4f754 c5b4f748
[ 1604.484236] f740: bf1ca43c c0149c48 c5b4f774 c5b4f758 bf1a5344 bf1ca400 c5b92000 bf20a88c
[ 1604.492424] f760: 00000000 c5b92188 c5b4f794 c0d62700 c5b92000 00000000 c5b4f794 c5b4f788
[ 1604.500613] f780: c0d62700 c01309ec c5b4f7c4 c5b4f798 c0abb86c c0d626d0 00000000 00000000
[ 1604.508801] f7a0: 00000000 00000000 c2026d80 c559d000 c5b92000 c160ef80 c5b4f81c c5b4f7c8
[ 1604.516989] f7c0: c0baebd8 c0abb7fc 00000000 010000e0 c5b4f7ec 005e0001 00000100 00000000
[ 1604.525176] f7e0: 00000000 00000000 00000000 00000000 00000000 c0bb0560 00000000 010000e0
[ 1604.533364] f800: c5b4f81c c5b4f810 c0d62700 c01309ec c5b4f84c c5b4f820 c0badce8 c0d626d0
[ 1604.541552] f820: c160f2d4 dc96fa09 c559d000 c0d626c0 c59de380 c59de380 c5b4f854 c5b4f848
[ 1604.549741] f840: c0d626c0 c01309ec c5b4f894 c5b4f858 c0c0ab6c c0d62698 c5b92000 00000000
[ 1604.557929] f860: c5b4f8cc c5b4f870 c0ba83f0 00000001 fffffff2 c5b92000 c59de400 c160ef80
[ 1604.566118] f880: 00000000 c0154440 c5b4f888 c5b4f888 c0c0f508 c0154440 c5b4f8cc dc96fa09
[ 1604.574306] f8a0: 00c58970 dc96fa09 bf215234 ffffffea 00000001 c5b4f918 c54f1410 c5b4fbd8
[ 1604.582494] f8c0: c5b4f8ec c5b4f8d0 c0154440 bf08408c c5b4f918 00000001 c160ef80 00001002
[ 1604.590682] f8e0: c5b4f914 c5b4f8f0 c0aa71b8 c0154408 00000000 c5b4f900 c5b92000 c5b92000
[ 1604.598870] f900: 00000001 00000000 c5b4f93c c5b4f918 c0ab7b74 c0aa7164 c5b92000 00000000
[ 1604.607058] f920: 00000000 dc96fa09 c5b92000 00000000 c5b4f95c c5b4f940 c0ab84f4 c0ab7b18
[... 87 lines cut ...]
[ 1605.286494]  r10:c5b4fe84 r9:c5b4fe44 r8:00000000 r7:00000000 r6:c39ee080 r5:c5b4ff48
[ 1605.294329]  r4:00000000
[ 1605.296862] [<c0a87870>] (___sys_sendmsg) from [<c0a87bcc>] (sys_sendmsg+0x80/0xc8)
[ 1605.304540]  r10:00000128 r9:c5b4e000 r8:c01002a4 r7:be9ec694 r6:00000000 r5:c39ee080
[ 1605.312375]  r4:00000000
[ 1605.314908] [<c0a87b4c>] (sys_sendmsg) from [<c0100060>] (ret_fast_syscall+0x0/0x48)
[ 1605.322668] Exception stack(0xc5b4ffa8 to 0xc5b4fff0)
[ 1605.327727] ffa0:                   00000003 00089000 00000003 be9ec694 00000000 00000000
[ 1605.335916] ffc0: 00000003 00089000 00089284 00000128 00000000 00000001 be9eceb5 00063805
[ 1605.344102] ffe0: 00089114 be9ec650 00060c7c b6ee3a2c
[ 1605.349160]  r7:00000128 r6:00089284 r5:00089000 r4:00000003
[ 1605.354827] Code: e5843204 0a000016 e2843fb2 e1d320d0 (e58520a4) 
[ 1605.360989] ---[ end trace a3b7f10f219c14e2 ]---
```

And device can't be rebooted after that. But wired network is working fine. Also I have the next _lspci_ output:

```
00:00.0 PCI bridge: MEDIATEK Corp. Device 0801 (rev 01)
00:01.0 PCI bridge: MEDIATEK Corp. Device 0801 (rev 01)
01:00.0 Unclassified device [0002]: MEDIATEK Corp. Device 7615
02:00.0 IDE interface: ASMedia Technology Inc. ASM1061 SATA IDE Controller (rev 02)
```

Linux can't identify the mt7615 board as a network device. With loaded mt7615e or not. I have _Linux bpi-r2 5.15.117-bpi-r2-main #4 SMP Thu Aug 17 11:51:19 MSK 2023 armv7l GNU/Linux_ kernel which built from this repo. What can I do in this situation? It seems like wrong firmware binaries.

frank-w, 2024-04-20:
i don't think it is firmware-related

`Unable to handle kernel NULL pointer dereference at virtual address 000000a6`
...
```
[ 1604.310526] PC is at cfg80211_mgd_wext_connect+0x68/0x18c [cfg80211]
[ 1604.317800] LR is at rtnl_is_locked+0x20/0x24
```
seems it is happening in cfg80211_mgd_wext_connect, for more you can add some debug prints to find the right location

try to update code first to latest 5.15 (./build.sh updatesrc), maybe it is already fixed

MastMind, 2024-04-20:
Thank you very much.
The latest update works well. I could run 2.4Ghz interface (I didn't tested 5Ghz yet but I guess it works fine too)

closed 2024-04-22 by MastMind