frank-w/BPI-Router-Linux #122: arm64 linux-headers-6.6.25-bpi-r3-main package ships with x86-64 files

gustavobsch, 2024-04-22:
Hi Frank,

I downloaded the latest 6.6-main package and I'm trying to dkms install a module having the following issue. Any idea how to resolve it?

`root@bpi-r3:/tmp# dpkg -i linux-headers-6.6.25-bpi-r3-main_6.6.25-gb54960c0ce55-8_arm64.deb 
Selecting previously unselected package linux-headers-6.6.25-bpi-r3-main.
(Reading database ... 122574 files and directories currently installed.)
Preparing to unpack linux-headers-6.6.25-bpi-r3-main_6.6.25-gb54960c0ce55-8_arm64.deb ...
Unpacking linux-headers-6.6.25-bpi-r3-main (6.6.25-gb54960c0ce55-8) ...
Setting up linux-headers-6.6.25-bpi-r3-main (6.6.25-gb54960c0ce55-8) ...
root@bpi-r3:/tmp#
root@bpi-r3:/tmp#
root@bpi-r3:/tmp# file /usr/src/linux-headers-6.6.25-bpi-r3-main/scripts/basic/fixdep 
fixdep: ELF 64-bit LSB pie executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=b7802d3f29393d5a235b4456107c5bf5489d9b3b, for GNU/Linux 3.2.0, not stripped
`

Thanks
Gustavo

gustavobsch, 2024-04-24:
Seems like I can re-generate the content of this folder by running make. However now I'm getting this error. Any idea how to generate the Kconfig file?

```
root@bpi-r3:/usr/src/linux-headers-6.6.25-bpi-r3-main# make
  SYNC    include/config/auto.conf.cmd
can't find file Kconfig
make[3]: *** [scripts/kconfig/Makefile:77: syncconfig] Error 1
make[2]: *** [Makefile:697: syncconfig] Error 2
make[1]: *** [/usr/src/linux-headers-6.6.25-bpi-r3-main/Makefile:798: include/config/auto.conf.cmd] Error 2
make: *** [Makefile:234: __sub-make] Error 2

```

frank-w, 2024-04-24:
I'm not sure what you trying to do...headers are only h files,no full kernel code...these are only for compiling 3rd party apps needing kernel apis (if you try to build a userspace app using gpio or similar).

Kconfig files are source for menuconfig and only in full source and not generable

frank-w, 2024-04-24:
It is possible that the debs are not all built right, libc,headers and dbg packages are built automaticly with the kbuild system...

gustavobsch, 2024-04-24:
I ended up installing debian's official linux-kbuild-6.7.9 for arm64 package and using the scripts in it

closed 2024-04-24 by gustavobsch