frank-w/BPI-Router-Linux #123: Buidl Fail , dt-bindings not found

1377382065, 2024-05-19:
when i try build kernel 5.10-main in bpi r2 
 i get error message
```
 SYSHDR  arch/arm/include/generated/uapi/asm/unistd-common.h
  SYSHDR  arch/arm/include/generated/uapi/asm/unistd-oabi.h
  SYSHDR  arch/arm/include/generated/uapi/asm/unistd-eabi.h
  UPD     include/config/kernel.release
  UPD     include/generated/uapi/linux/version.h
  UPD     include/generated/utsrelease.h
  HOSTCC  scripts/dtc/dtc.o
  HOSTCC  scripts/dtc/flattree.o
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/dtc/data.o
  HOSTCC  scripts/dtc/livetree.o
  HOSTCC  scripts/dtc/treesource.o
  HOSTCC  scripts/dtc/srcpos.o
  HOSTCC  scripts/dtc/checks.o
  HOSTCC  scripts/dtc/util.o
  LEX     scripts/dtc/dtc-lexer.lex.c
  YACC    scripts/dtc/dtc-parser.tab.[ch]
  HOSTCC  scripts/dtc/dtc-lexer.lex.o
  HOSTCC  scripts/dtc/dtc-parser.tab.o
  HOSTLD  scripts/dtc/dtc
  HOSTCC  scripts/kallsyms
  HOSTCC  scripts/recordmcount
  HOSTCC  scripts/sorttable
  DTC     arch/arm/boot/dts/mt2701-evb.dtb
arch/arm/boot/dts/mt2701-evb.dts:9:10: fatal error: dt-bindings/gpio/gpio.h: No such file or directory
    9 | #include <dt-bindings/gpio/gpio.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.lib:326: arch/arm/boot/dts/mt2701-evb.dtb] Error 1
make: *** [Makefile:1387: dtbs] Error 2
make: *** Waiting for unfinished jobs....
  HOSTCC  scripts/asn1_compiler
  HOSTCC  scripts/extract-cert

```

frank-w, 2024-05-19:
cannot reproduce this...file is there and under git control

```
frank@frank-G5:/media/data_ext/git/kernel/BPI-R2-4.14 (5.10-main) [0M5U]
$ ls -l include/dt-bindings/gpio/gpio.h
-rw-rw-r-- 1 frank frank 1170 Mai 19 11:19 include/dt-bindings/gpio/gpio.h
frank@frank-G5:/media/data_ext/git/kernel/BPI-R2-4.14 (5.10-main) [0M5U]
$ git logone include/dt-bindings/gpio/gpio.h
ede033e1e863 2019-02-07 dt-bindings: gpio: document the new pull-up/pull-down flags Thomas Petazzoni 
e10f72bf4b3e 2017-11-30 gpio: gpiolib: Generalise state persistence beyond sleep Andrew Jeffery 
....
```
https://github.com/frank-w/BPI-Router-Linux/blob/5.10-main/include/dt-bindings/gpio/gpio.h

you can try

`git checkout include/dt-bindings/gpio/gpio.h`

or something is wrong in your build-environment...do you use build.sh from your source root?

1377382065, 2024-05-19:
yes , i sure file exist
Here is my complete operation: 

```
niko@ubuntu-compile:~/BPI-Router-Linux$ pwd
/home/niko/BPI-Router-Linux
niko@ubuntu-compile:~/BPI-Router-Linux$ git branch 
* 5.10-main
  5.15-main
niko@ubuntu-compile:~/BPI-Router-Linux$ git status 
On branch 5.10-main
Your branch is up to date with 'origin/5.10-main'.

nothing to commit, working tree clean
niko@ubuntu-compile:~/BPI-Router-Linux$ 
```

```
./build.sh importconfig
./build.sh config
./build.sh

arch/arm/boot/dts/mt2701-evb.dts:9:10: fatal error: dt-bindings/gpio/gpio.h: No such file or directory
    9 | #include <dt-bindings/gpio/gpio.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.lib:326: arch/arm/boot/dts/mt2701-evb.dtb] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from arch/arm/boot/dts/mt6589-aquaris5.dts:9:
arch/arm/boot/dts/mt6589.dtsi:8:10: fatal error: dt-bindings/interrupt-controller/irq.h: No such file or directory
    8 | #include <dt-bindings/interrupt-controller/irq.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.lib:326: arch/arm/boot/dts/mt6589-aquaris5.dtb] Error 1
In file included from arch/arm/boot/dts/mt6580-evbp1.dts:9:
arch/arm/boot/dts/mt6580.dtsi:8:10: fatal error: dt-bindings/interrupt-controller/irq.h: No such file or directory
    8 | #include <dt-bindings/interrupt-controller/irq.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.lib:326: arch/arm/boot/dts/mt6580-evbp1.dtb] Error 1
make: *** [Makefile:1387: dtbs] Error 2
make: *** Waiting for unfinished jobs....

```

but 

```
niko@ubuntu-compile:~/BPI-Router-Linux$ ls ./include/dt-bindings/interrupt-controller/irq.h
./include/dt-bindings/interrupt-controller/irq.h
niko@ubuntu-compile:~/BPI-Router-Linux$ 
```

frank-w, 2024-05-19:
Something is strange with your buildchain...you did the right steps and i compiled 5.10 sucessfully on my ubuntu 22.04 host

1377382065, 2024-05-19:
i same use ubuntu 22.04
```
niko@ubuntu-compile:~/BPI-Router-Linux$ uname  -a 
Linux ubuntu-compile 5.15.0-107-generic #117-Ubuntu SMP Fri Apr 26 12:26:49 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
niko@ubuntu-compile:~/BPI-Router-Linux$ cat /proc/version
Linux version 5.15.0-107-generic (buildd@lcy02-amd64-012) (gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #117-Ubuntu SMP Fri Apr 26 12:26:49 UTC 2024
niko@ubuntu-compile:~/BPI-Router-Linux$ 
```
but i get same error when i compile in bpi r2 , i use basis image https://drive.google.com/drive/folders/1oP7jy1KrrIOifvImo2nQ59wx3_9hHkgk

1377382065, 2024-05-19:
i use basis image https://drive.google.com/drive/folders/1oP7jy1KrrIOifvImo2nQ59wx3_9hHkgk
but not support ipset , and i not found this kernel version source code , i just need ipset support

frank-w, 2024-05-19:
why not using later kernel-version? r2 itself also works in 6.6 (latest LTS), but internal wifi (mt6625l) is last supported in 5.15

but strange anyway why it does not compile for you...i updated the tree and pushed it...lets see if pipeline fails too...

on my current system i use this gcc:

```
$ arm-linux-gnueabihf-gcc -v
Using built-in specs.
COLLECT_GCC=arm-linux-gnueabihf-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc-cross/arm-linux-gnueabihf/9/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 9.5.0-1ubuntu1~22.04' --with-bugurl=file:///usr/share/doc/gcc-9/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++,gm2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --disable-libquadmath-support --enable-plugin --enable-default-pie --with-system-zlib --without-target-system-zlib --enable-libpth-m2 --enable-multiarch --disable-sjlj-exceptions --with-arch=armv7-a --with-fpu=vfpv3-d16 --with-float=hard --with-mode=thumb --disable-werror --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=arm-linux-gnueabihf --program-prefix=arm-linux-gnueabihf- --includedir=/usr/arm-linux-gnueabihf/include --with-build-config=bootstrap-lto-lean --enable-link-mutex
Thread model: posix
gcc version 9.5.0 (Ubuntu 9.5.0-1ubuntu1~22.04)
```

1377382065, 2024-05-19:
i download release souce code  and build success !  
maybe git clone loss some file ?

closed 2024-05-19 by 1377382065

frank-w, 2024-05-19:
i can also build successful and as the file is there there is something broken in build-process, you can try to reset your tree "ARCH=arm make mrproper" and maybe do a git hard reset (git reset --hard origin/5.10-main)

release source-code is taken from git :p see pipeline...it does a git clone first