frank-w/BPI-Router-Linux #124: [BPI-R4] Docker support

arosenhagen, 2024-07-30:
Is there a way to get Docker working on a BPI-R4 with current Debian Bookworm Image? I tried different kernel versions but all stuck on the following error, when trying to start a docker-container:

`docker: Error response from daemon: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error setting cgroup config for procHooks process: bpf_prog_query(BPF_CGROUP_DEVICE) failed: invalid argument: unknown.`

Adding `systemd.unified_cgroup_hierarchy=0` to kernel parameters didn't help either.

frank-w, 2024-07-30:
you need CONFIG_CGROUP_BPF selected in kernel, seems my r4 branches do not have it yet...

arosenhagen, 2024-07-30:
if you give me a bit of a direction of where to actually add it I may build it on my own and test it ;-)

frank-w, 2024-07-30:
pushed to 6.9-main/6.10-main branches so you can wait for pipeline to be complete or compile yourself (instructions in readme.md on default branch)

closed 2024-07-31 by arosenhagen

frank-w, 2024-07-31:
Docker works now?

arosenhagen, 2024-08-01:
yes - like a charm. thanks for the fast adjustment!