frank-w/BPI-Router-Linux #125: please install libncurses5-dev (Ubuntu 24.04 )

amatsumara, 2024-08-10:
Trying to build image on Ubuntu 24.04
```
./build.sh                                                                                                                         (5.15-main)BPI-Router-Linux
Build Kernel
Cleanup Kernel Build
needed: make u-boot-tools bc gcc libc6-dev libncurses5-dev ccache libssl-dev
please install libncurses5-dev

```
But `libncurses-dev` already present in the system:
```
apt install libncurses5-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libncurses-dev' instead of 'libncurses5-dev'
libncurses-dev is already the newest version (6.4+20240113-1ubuntu2).
0 to upgrade, 0 to newly install, 0 to remove and 4 not to upgrade.
```

frank-w, 2024-08-10:
Yes i already fixed this in some trees...which branch do you use?

https://github.com/frank-w/BPI-Router-Linux/commit/2ba2cbbaf46c1eef7843773f9ae4299179f6c066

amatsumara, 2024-08-10:
> Yes i already fixed this in some trees...which branch do you use?
> 
> [2ba2cbb](https://github.com/frank-w/BPI-Router-Linux/commit/2ba2cbbaf46c1eef7843773f9ae4299179f6c066)

I tried to build 5.15-main

frank-w, 2024-08-11:
Have updated the lts main trees with the fix and updated kernel version...should work now

closed 2024-08-11 by amatsumara