frank-w/BPI-Router-Linux #129: Add CAKE scheduler to tc

danielschonfeld, 2024-08-31:
Same idea as adding IFB, CAKE is more modern and useful for SQM configurations.

Please consider adding this
https://github.com/frank-w/BPI-Router-Linux/blob/89214dc3d3de95927217cd0102451a591949a8f7/net/sched/Kconfig#L281

frank-w, 2024-08-31:
Can you check if more options make sense for tc / queuing so i do not add options one-by-one

danielschonfeld, 2024-08-31:
I’m not an expert on this but mostly I think this project (
https://github.com/tohojo/sqm-scripts) is the de facto “standard”. I think
once we add cake that should be it but I’m not promising anything.

Apologies for the piece meal requests!

danielschonfeld, 2024-09-01:
Thank you for adding that scheduler.  I just tested it and that's all one needs for sqm-scripts so I believe we can close this issue.

👍

closed 2024-09-02 by frank-w