frank-w/BPI-Router-Linux #131: Which branch for BPI-R4 with 6-Antenna BE14 Wifi Board?

BtbN, 2024-09-07:
Title says it all.
There are three potential branches of the 6.10 kernel. The main one, the mt76 or the mt76_eht.
They all seem to contain relevant patches, but the mt76_eht seems to revert support for the 2+3+3 module, which I think is the BE14?
I've also successfully rebased the mt76_eht on top of main, and excluded the commit that removes support. But no idea if that is in any way correct.

frank-w, 2024-09-07:
Why do you think the 233 patch is reverted? I only reverted one patch from the eht series as it causes a warn on bootup

6.10-main or 6.11-rc should be fine

frank-w, 2024-09-07:
6.11-rc contains the eht patches except 1 reverted...6.10-main has only be14 support