frank-w/BPI-Router-Linux #133: BPI R4 ethernet fails under heavy load on 6.11

wteiken, 2024-11-25:
After a while of heavy load I have seen the BPI R4 ethernet stop working. The network can no longer send, and after a few seconds network watchdog messages show up:

2024-11-24T05:03:30.377477-05:00 nostromo kernel: [22865.993414] mtk_soc_eth 15100000.ethernet en-lan0: NETDEV WATCHDOG: CPU: 0: transmit queue 1 timed out 5710 ms
2024-11-24T05:03:30.377532-05:00 nostromo kernel: [22866.003459] mtk_soc_eth 15100000.ethernet en-lan0: transmit timed out
2024-11-24T05:03:30.377543-05:00 nostromo kernel: [22866.009909] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 0: transmit queue 3 timed out 5690 ms

Happens on two different boards, so not an issue with a specific board.

This happened very rarely with the BPI provided 6.1 kernel (every few weeks), but significantly more often with the 6.11 (almost every time when a a backup runs and uploads larger amounts of data over a long period of time.But it also happened on a slower internet connection, so can't be a pure throughput issue.

I'll try to check if the network card is still receiving traffic, or if both send and received are done.

wteiken, 2024-11-25:
I have seen it a few minutes into the heavy load or a few hours in. So no clear trigger identified yet. I'll test a few things based on the thread.

frank-w, 2024-11-25:
have added this patch to 6.11-main tree

https://patchwork.kernel.org/project/linux-mediatek/patch/20241015081755.31060-1-nbd@nbd.name/

6.12-main already contains it

maybe it helps a bit...i also try to port daniels patch to 6.12

wteiken, 2024-11-26:
Thanks, I just installed the new kernel including the patches. Will see how it fares.

wteiken, 2024-11-26:
Failed fairly quickly (20 minutes of ~250mbit routed) with the new patches, so it does not seem to make a difference. Traffic is being routed between the 1st SFP and one of the eth ports on the DSA. The SFP has multiple VLANs, not sure if that may have an impact.

Confirmed that the system is still receiving network traffic, it just can't send anymore (i.e., I can do a tcpdump and see the traffic on the device.

These are the settings that are "on":
rx-checksumming: on
tx-checksumming: on
	tx-checksum-ipv4: on
	tx-checksum-ipv6: on
scatter-gather: on
	tx-scatter-gather: on
tcp-segmentation-offload: on
	tx-tcp-segmentation: on
	tx-tcp-ecn-segmentation: on
	tx-tcp-mangleid-segmentation: on
	tx-tcp6-segmentation: on
generic-segmentation-offload: on
generic-receive-offload: on
tx-vlan-offload: on
hw-tc-offload: on

I do not see anything else in the logs that indicates issues before the crash.

frank-w, 2024-11-26:
Which base image do you use? Imho my debian image has tx offload disabled by default...have you tried disabling it?

wteiken, 2024-11-26:
It's a standard debian install with a custom kernel. I'll try disabling tx offload over the weekend (will try the 6.1 for a few days to double check it is still more stable / ensure nothing else causes the issue).

frank-w, 2024-11-26:
Debian install? How uave you managed to install debian without a base image (atf,uboot,etc.)?

wteiken, 2024-11-26:
Ah, misunderstood. The original images are from the BPI repo, that is also the source for the 6.1 kernel I am using (with later patches installed to 6.1.119).

frank-w, 2024-11-26:
Hi,no i have not changed to ubi yet as the setup is much different and i'm not experienced in it

wteiken, 2024-11-26:
Ah, I though that combined one was then called bl33. But yes, I am talking about the bl31.bin that also goes into the fip.

When I glanced at your commits that happened after I built my u-boot I saw some comment about enabling UBI. My bl2 is configured to read the fip from UBI. u-boot also reads the env from UBI. But I assume that should not really make a difference for this specific issue.

frank-w, 2024-11-27:
sorry for offtopic, right, the ATF should be not resposible for the network breakage, but the basic image may...from my experience the bpi-images are heavily modified and acting unstable in some cases. thats why i build my own debian images which are build from scratch (atf+uboot+kernel built from source) and getting rootfs from debootstrap with some basic configs changed in a transparent way (bpi-router-images repo).

as said before there are some networking issues known (e.g. 100mbit clients may cause some issues and there is the timeout-issue not completely solved due to missing reproducing steps - it just happening for some users). i have not be able to reproduce it, but seems like tx-offloading is one rootcause...maybe not initialized completely correct, but disabling it seems to solve it...not the right way, but helps to find the rootcause...the mtk ethernet driver is huge and digging for such error without reproducing steps and having no clue why it happens is nearly impossible

wteiken, 2024-11-27:
Yes, the debian image is installed from scratch, I only used the BPI image to get the boot loaders set up.

I'll try disabling TX offloading and see what that does.

wteiken, 2024-11-30:
Did a few experiments, it seems turning TX off does seem to help (running a transfer for close to 3 hours now, no issues yet). And it only happens on the transmitting system, so that would be aligned with an issue on TX offload.

As a test I also only disabled TX offload on the sending system, the receiving system still uses TX offload and no issues. So some suspicion this only causes issues with TX offload for large packets or under heavy load.

I'll continue running the test to see if it just takes a bit longer until the crash.

The other observation is that this seems to  happen only with the DSA ethernet on the R4. I transferred data for a number of hours on the SFP module and no issues.

wteiken, 2024-12-01:
Not yet sure what is going on exactly, but I realized I did not have irqbalance running before. After enabling irqbalance the issue seems to also at least be a lot less prominent (transferred data for ~4h without issues). Wondering if it that points to a timing issue with the TX offload...
But will run some more tests to confirm.

frank-w, 2024-12-02:
added these 2 to 6.12-main (first one is a bit changed as dummy_dev is allocated a bit differently)

frank-w, 2024-12-07:
This seems to need a param (e.g. ignore_dma_hangs=1) in cmdline...i guess it is not set by default in sdk build

frank-w, 2024-12-07:
Or like this,but it seems like a debug setting,not productive change

wteiken, 2024-12-08:
I have not yet been able to reproduce the issue since I enabled irqbalancer despite the workload that tended to trigger it having run multiple times. So that definitely change the likelihood of it happening.

Will try do disable irqbalancer just to double check...

frank-w, 2024-12-09:
You're right,imho it should be mtk_max_devs (3)

There are some more 6.6 patches (also for debugfs).

https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/69faae33c29729926136d9acba6de4d8db23aba2/autobuild/unified/global/24.10/files/target/linux/mediatek/patches-6.6?autodive=0/

Not sure about netsys_v1 imho these are older devices like mt7622/mt7623. Not if these changes are needed for them.

Which base is your patch in gist above? 6.6 or 6.11?

frank-w, 2024-12-09:
Ok

Some interesting should be this:

https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/69faae33c29729926136d9acba6de4d8db23aba2/autobuild/unified/global/24.10/files/target/linux/mediatek/patches-6.6/999-2703-net-ethernet-mtk_eth_soc-revise-hardware-configurati.patch?autodive=0%2F

frank-w, 2024-12-09:
> This patch is old.

Possibly, but it looks like not applied. E.g. this position:

https://github.com/frank-w/BPI-Router-Linux/blob/6.12-main/drivers/net/ethernet/mediatek/mtk_eth_soc.c#L4190

frank-w, 2024-12-11:
I'm in contact with mtk regarding rss/lro,not sure if the contact is same...can you send me email and i follow it to mtk.

frank-w, 2024-12-16:
Iit would be interesting if the hang_count is above 0 before it is set to zero..

frank-w, 2024-12-28:
Could you contact me by email to evaluate next steps?

wteiken, 2025-01-08:
Another update: I changed my backup client, and with the new traffic pattern it seems it is setting the bug off fairly reliably every 5-10 minutes. As mentioned the system recovers after  an ~5 second hang.

A sample log:
[118932.223231] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 2: transmit queue 3 timed out 5430 ms
[118932.233089] mtk_soc_eth 15100000.ethernet eth0: transmit timed out
[118932.303226] mtk_soc_eth 15100000.ethernet en-lan0: NETDEV WATCHDOG: CPU: 0: transmit queue 1 timed out 5510 ms
[118932.437869] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[118932.437958] mtk_soc_eth 15100000.ethernet en-lan0: Link is Down
[118932.884647] mtk_soc_eth 15100000.ethernet: warm reset failed
[118932.911327] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/internal link mode
[118932.911376] mtk_soc_eth 15100000.ethernet en-lan0: configuring for inband/10gbase-r link mode
[118932.911379] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 10Gbps/Full - flow control rx/tx
[118932.960541] mtk_soc_eth 15100000.ethernet en-lan0: Link is Up - 10Gbps/Full - flow control off

Another thing I noticed: I see the following message once this happens once on a regular basis:
[118961.343310] page_pool_release_retry() stalled pool shutdown: id 6, 33 inflight 1809 sec

It seems this does not get resolved when the network recovers.

Given the timing I assume these two are connected.

frank-w, 2025-01-08:
i added the padding-fix to 6.12-main with changed condition to MTK_MIN_TX_LENGTH to always pad to this value (not only if length smaller 40 bytes)...maybe it helps you here...

wteiken, 2025-01-13:
Unfortunately neither patch seemed to help. The interface is still getting stuck, and the pool is not freed (double checked that I actually installed the new kernel...).

As far as I can tell this only ever happens on the upload routing side (incoming traffic on 10G SFP, outgoing traffic on one of the eths behind the internal switch). Upload throughput is ~400mbit. So the download side would have smaller packets with acks (which I think would be smaller than MTK_MIN_TX_LENGTH) so I was hoping this is it, but no such luck.
But also, both the lan side and the wan side queue seem to always get stuck at the same time, so the issue may be somewhere else.

I tried changing the upload client parallelism, I see some mild evidence that with less parallel uploads this happens more often (more processes/threads mean hang every 5-10 min, less processes/threads mean hang every 2-4 min) despite the total throughput being roughly the same. But I did not test this extensively, so it could have been a fluke.
I have never seen this with long downloads, but given this also does not happen with every upload client not sure how reliable that is.

When I'll get a chance I'll try to do the upload from the bpi directly (avoid the routing) to see if it still happens. That would at least give some indication if this is an issue directly or indirectly caused by traffic on two interfaces at the same time.
I'll also try downloads with the same client and see if that triggers the same effect in the other direction.

wteiken, 2025-01-13:
Not yet sure what to make of it, but I got around to setting up IPv6 on the machine running the backup client and now the backup ran through without any issues. I will have to double check, but disabling IPv6 was among the changes (in addition to a different backup client) that caused the issue to reappear after it disappeared when enabling irqbalancer.

So up to now the score seems to be: if traffic is sent via IPv6 and irqbalancer is active then I do not see the issue. Not really sure how IPv6 would impact it though. Traffic patterns would be somewhat different, and of course there is no masquerading, but would that impact anything in the network driver?

I'll disable IPv6 and retry to see if the issue reappears when I get a chance.

wteiken, 2025-01-14:
Confirmed that switching back to IPv4 made the issue reappear.

After applying patch 1 (2&3 are already in 6.12-main) I could not reproduce the issue. So that patch may have been it.

I'll monitor over the next few days.

Thanks!

wteiken, 2025-01-15:
Another two days without issue, so patch 1 seems to have done the trick for me.

wteiken, 2025-01-15:
Closing for the moment as the issue seems resolved. Thanks!

closed 2025-01-15 by wteiken

Headcrabed, 2025-03-23:
> [@wteiken](https://github.com/wteiken) I think that you should import all mtk fixes from openwrt.
> 
> 1. [1c47744](https://github.com/frank-w/BPI-Router-Linux/commit/1c47744156bf9aaec28cb62ec64e1303d646f822)
> 2. https://github.com/openwrt/openwrt/blob/5ce1af9539da667f62dc3cb426f9ec36e53fb54e/target/linux/generic/pending-6.6/734-net-ethernet-mediatek-enlarge-DMA-reserve-buffer.patch
> 3. https://github.com/openwrt/openwrt/blob/5ce1af9539da667f62dc3cb426f9ec36e53fb54e/target/linux/generic/pending-6.6/730-net-ethernet-mtk_eth_soc-reset-all-TX-queues-on-DMA-.patch
> 4. Also, you can swap these two lines.
>    https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/refs/heads/master/21.02/files/target/linux/mediatek/files-5.4/drivers/net/ethernet/mediatek/mtk_eth_soc.c#2891
> 
> ```
> @@ -2468,10 +2468,10 @@ static int mtk_poll_tx_qdma(struct mtk_eth *eth, int budget,
>  	while ((cpu != dma) && budget) {
>  		u32 next_cpu = desc->txd2;
>  
> -		desc = mtk_qdma_phys_to_virt(ring, desc->txd2);
>  		if ((desc->txd3 & TX_DMA_OWNER_CPU) == 0)
>  			break;
>  
> +		desc = mtk_qdma_phys_to_virt(ring, desc->txd2);
>  		tx_buf = mtk_desc_to_tx_buf(ring, desc,
>  					    eth->soc->tx.desc_size);
>  		if (!tx_buf->data)
> ```
> 
>  5. https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/3a5223473e086a4b54a2b9a44df7d9ddcc2bc75a
> 
> ```
> @@ -3603,9 +3603,6 @@ static int mtk_open(struct net_device *dev)
>  			}
>  			mtk_gdm_config(eth, target_mac->id, gdm_config);
>  		}
> -		/* Reset and enable PSE */
> -		mtk_w32(eth, RST_GL_PSE, MTK_RST_GL);
> -		mtk_w32(eth, 0, MTK_RST_GL);
>  
>  		napi_enable(&eth->tx_napi);
>  		napi_enable(&eth->rx_napi);
> ```

@frank-w seems there are still some modifications in this reply not applied to your current code, can you put them in ?

wirr00, 2025-04-03:
Not sure if that also applies to your context, but disabling flow control (rx) on the switch interfaces resolved this issue on kernel 6.6.85 (OpenWRT Snapshot 25-04-02).