frank-w/BPI-Router-Linux #134: [R2] mt6323 issues

alxchk, 2024-12-17:
Hi frank,
Long time R2 worked for me without issues as my primary router, however the day has come and something made it dead ~_~

So, long story short, one day I had storm of messages with 
```
mt-pmic-pwrap 1000d000.pwrap: unexpected interrupt int=0x2
mt-pmic-pwrap 1000d000.pwrap: unexpected interrupt int=0x2
mt-pmic-pwrap 1000d000.pwrap: unexpected interrupt int=0x2
mt-pmic-pwrap 1000d000.pwrap: unexpected interrupt int=0x2
```
While system still works, it's infinite loop of such messages, so sooner or later it hangs.
Usually reboot fixed that for a week or so.
With time, things became worser, and now it's not even possible to boot without issues.

However,  I figured out, that 
1) If pmic/pwrap drivers disabled it works somehow
2) If CPU freq policy set to powersave it usually boots (however may silently poweroff)
3)  Usually errors and power issues follows CPU frequency change + SATA/USB active

So, it's clear that there are some kind of hardware issue which I'm not sure how I can debug.
Still, if "unexpected interrupt int=0x2" may be fixed somehow, this thing still may work (at least until new R4 will arrive)

This message is from `drivers/soc/mediatek/mtk-pmic-wrap.c`, function pwrap_interrupt
```
rdata = pwrap_readl(wrp, PWRAP_INT_FLG);
dev_err(wrp->dev, "unexpected interrupt int=0x%x\n", rdata);
pwrap_writel(wrp, 0xffffffff, PWRAP_INT_CLR);
```
There are no documentation about mt6323, so it's not clear, is that way of clearing error okay or not.
And what "int=0x2" actually means.

I tried to  put various things to int_en_all in hope to at least request not to call this interrupt without any success
```
static const struct pmic_wrapper_type pwrap_mt7622 = {
        .regs = mt7622_regs,
        .type = PWRAP_MT7622,
        .arb_en_all = 0xff,
        .int_en_all = ~(u32)BIT(31), 
        .int1_en_all = 0,
        .spi_w = PWRAP_MAN_CMD_SPI_WRITE,
        .wdt_src = PWRAP_WDT_SRC_MASK_ALL,
        .caps = PWRAP_CAP_RESET | PWRAP_CAP_DCM,
        .init_reg_clock = pwrap_common_init_reg_clock,
        .init_soc_specific = pwrap_mt7622_init_soc_specific,
};
```

How do you think, is there any way to get more information about that?

frank-w, 2024-12-17:
i have no more info about it...maybe revert the changes for internal wifi in pwrap driver...

alxchk, 2024-12-17:
Well, that happend at pretty old kernel first, so it's not because of some kernel update I think. 
> why do you add this struct for mt7622 (r64)?

This struct is from original sources, I had not added anything there. Just tried to remove bits from int_en_all field because had hope, that may mask interrupt (blind attempt, as there is no datasheet around)

frank-w, 2024-12-17:
Yes but maybe it is related to wifi and maybe you can ressurect your board when disabling internal wifi power.

You have to change struct for mt7623 (not mt7622), but be careful...you're working on power system and maybe adding more damage

alxchk, 2024-12-17:
Do you have idea how properly disable wifi? I.e. I can disable driver in .config, however not sure it will be enough. 
As for now I'm not activating wifi (at least userspace part is disabled). However I don't know was it enabled by regulator or not.