frank-w/BPI-Router-Linux #137: Problem with BPI MT7615 module and eeprom

agevlakh, 2025-01-31:
Hi!
I'm sorry to write here, but I really need your help. I want to set up a Banana Pi R2 wi-fi module from BPi mt7615. At the same time, I am using OpenWRT 24.10-rc7. I ran into a problem with a poor signal of 6 dB. I found a topic about eeprom that needs to be downloaded from a file, downloaded the OpenWRT code via git, found the eeprom.c file in the mt76 folder, and changed it, as described in your example. I changed the file in 3 places: it is present in the linux folders-6.7.73, backports-6.12.6, and in the toolchain folder. Next, in menuconfig, I selected mt76 as the module (M) (I also tried with an asterisk, but the results were the same (*)), as well as the mt7615 drivers. Unfortunately, after compiling, flashing and adding the mt7615e_rf file.bin to the /lib/firmware/mediatek folder, nothing happened in the logs, there is no information that the eeprom module is being searched for. I don't understand what I did wrong. Please tell me what I'm doing wrong, I didn't know how else to contact you. Thank you very much in advance...

frank-w, 2025-01-31:
Also have only read about the tx power change in eeprom,but have not done this so cannot guide you here.
The eeprom file loading needed additional patch,i'm not sure if i added it to the branch you use or openwrt.

agevlakh, 2025-01-31:
Thanks for the answer. I just want to increase the signal strength. Wifi module work done, but weak signal. Tell me, do you use this patch? Maybe I can apply it to the openwrt source code? I only saw information about editing eeprom.c and adding bin file in mediatek folder

frank-w, 2025-01-31:
3afa402eaa58 2020-12-16 mt76: allow loading eeprom from file

seems 6.6-main missing this patch, 6.1-main has this:

f2d182e5aa22 2020-12-16 mt76: allow loading eeprom from file

agevlakh, 2025-01-31:
Thank you, but I just described above how I applied this patch to the openwrt kernel. But for some reason it doesn't work. Maybe I should try to name the stable version? It has kernel 6.1, if I'm not mistaken. Maybe it's a kernel version issue?

frank-w, 2025-01-31:
i do not see your changes, but you need to add the function loading from file and add its call the right way to the mt76_get_of_eeprom function and change the eeprom binary at some offsets to get higher tx-power. using the original file does not change it. not sure if there was somewhere a already modified eeprom file

is the file loaded correctly? maybe adding some debug lines to code to make sure

agevlakh, 2025-01-31:
I'm not a programmer, but I'll try to explain what I did again:
1) downloaded openwrt source codes
2) changed the eeprom.c file in the mt76 folder
3) compiled the firmware
4) installed the firmware
5) put the mt7615_rf.bin file in the /lib/firmware/mediatek folder

My logs after that:
```
root@bpir2:~# logread | grep mt7615
Fri Jan 31 15:54:36 2025 kern.info kernel: [   11.877095] mt7615e 0000:01:00.0: enabling device (0140 -> 0142)
Fri Jan 31 15:54:36 2025 kern.info kernel: [   11.884618] mt7615e 0000:01:00.0: Invalid MAC address, using random address 9a:01:8c:44:9e:88
Fri Jan 31 15:54:36 2025 kern.info kernel: [   11.895313] mt7615e 0000:01:00.0: registering led 'mt76-phy0'
Fri Jan 31 15:54:36 2025 kern.info kernel: [   12.027159] mt7615e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20180518100604a
Fri Jan 31 15:54:36 2025 kern.info kernel: [   12.152658] mt7615e 0000:01:00.0: N9 Firmware Version: _reserved_, Build Time: 20200814163649
Fri Jan 31 15:54:36 2025 kern.info kernel: [   12.184183] mt7615e 0000:01:00.0: CR4 Firmware Version: _reserved_, Build Time: 20190415154149
Fri Jan 31 15:54:43 2025 kern.info kernel: [   20.658950] mt7615e 0000:01:00.0: registering led 'mt76-phy1'
Fri Jan 31 15:54:43 2025 kern.info kernel: [   20.678816] mt7615e 0000:01:00.0 phy0-ap0: entered allmulticast mode
Fri Jan 31 15:54:43 2025 kern.info kernel: [   20.685590] mt7615e 0000:01:00.0 phy0-ap0: entered promiscuous mode
Fri Jan 31 15:54:45 2025 kern.info kernel: [   22.617317] mt7615e 0000:01:00.0 phy1-ap0: entered allmulticast mode
Fri Jan 31 15:54:45 2025 kern.info kernel: [   22.624110] mt7615e 0000:01:00.0 phy1-ap0: entered promiscuous mode
Fri Jan 31 15:59:40 2025 kern.info kernel: [   32.071112] mt7615e 0000:01:00.0 phy0-ap0: left allmulticast mode
Fri Jan 31 15:59:40 2025 kern.info kernel: [   32.077277] mt7615e 0000:01:00.0 phy0-ap0: left promiscuous mode
Fri Jan 31 15:59:41 2025 kern.info kernel: [   33.188621] mt7615e 0000:01:00.0 phy0-ap0: entered allmulticast mode
Fri Jan 31 15:59:41 2025 kern.info kernel: [   33.195444] mt7615e 0000:01:00.0 phy0-ap0: entered promiscuous mode
```
I think th patch not working  I cannot see phrases like load eeprom or similar

frank-w, 2025-01-31:
i cannot help with openwrt, here you have to ask in openwrt forum. not sure if eeprom.c is maybe resettet on build or not compatible, but this should result in build-errors

agevlakh, 2025-02-02:
I try again compile your patch and have error:

`/mnt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-mediatek_mt7623/mt76-2024-04-03-1e336a85/eeprom.c:162:1: error: static declaration of 'mt76_find_power_limits_node' follows non-static declaration
  162 | mt76_find_power_limits_node(struct mt76_dev *dev)
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mnt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-mediatek_mt7623/mt76-2024-04-03-1e336a85/eeprom.c:10:
/mnt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-mediatek_mt7623/mt76-2024-04-03-1e336a85/mt76.h:1605:1: note: previous declaration of 'mt76_find_power_limits_node' with type 'struct device_node *(struct mt76_dev *)'
 1605 | mt76_find_power_limits_node(struct mt76_dev *dev);
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/mnt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-mediatek_mt7623/mt76-2024-04-03-1e336a85/eeprom.c:214:1: error: static declaration of 'mt76_find_channel_node' follows non-static declaration
  214 | mt76_find_channel_node(struct device_node *np, struct ieee80211_channel *chan)
      | ^~~~~~~~~~~~~~~~~~~~~~
/mnt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-mediatek_mt7623/mt76-2024-04-03-1e336a85/mt76.h:1607:1: note: previous declaration of 'mt76_find_channel_node' with type 'struct device_node *(struct device_node *, struct ieee80211_channel *)'
 1607 | mt76_find_channel_node(struct device_node *np, struct ieee80211_channel *chan);
      | ^~~~~~~~~~~~~~~~~~~~~~
make[5]: *** [scripts/Makefile.build:289: /mnt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-mediatek_mt7623/mt76-2024-04-03-1e336a85/eeprom.o] Error 1
make[4]: *** [Makefile:1907: /mnt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-mediatek_mt7623/mt76-2024-04-03-1e336a85] Error 2
make[4]: Leaving directory '/mnt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-mediatek_mt7623/linux-5.15.167'
make[3]: *** [Makefile:638: /mnt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-mediatek_mt7623/mt76-2024-04-03-1e336a85/.built] Error 2
make[3]: Leaving directory '/mnt/openwrt/package/kernel/mt76'
time: package/kernel/mt76/compile#2.04#0.76#2.54
    ERROR: package/kernel/mt76 failed to build.
make[2]: *** [package/Makefile:129: package/kernel/mt76/compile] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/mnt/openwrt'
mkdir -p /mnt/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-mediatek/stamp
SHELL= flock /mnt/openwrt/tmp/.root-copy.flock -c 'cp -fpR /mnt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/luci-mod-admin-full/.pkgdir/luci-mod-admin-full/. /mnt/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-mediatek/'
touch /mnt/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-mediatek/stamp/.luci-mod-admin-full_installed
echo "luci-mod-admin-full" >> /mnt/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/luci-mod-admin-full.default.install
make[3]: Leaving directory '/mnt/openwrt/feeds/luci/modules/luci-mod-admin-full'
time: package/feeds/luci/luci-mod-admin-full/compile#1.52#0.50#1.98
make[2]: Leaving directory '/mnt/openwrt'
make[1]: *** [package/Makefile:123: /mnt/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/openwrt'
make: *** [/mnt/openwrt/include/toplevel.mk:233: world] Error 2
`
Can you help to understand what I do wrong?

agevlakh, 2025-02-02:
eeprom.c file:

`// SPDX-License-Identifier: ISC
/*
 * Copyright (C) 2016 Felix Fietkau <nbd@nbd.name>
 */
#include <linux/of.h>
#include <linux/of_net.h>
#include <linux/mtd/mtd.h>
#include <linux/mtd/partitions.h>
#include <linux/etherdevice.h>
#include "mt76.h"

static int mt76_get_eeprom_file(struct mt76_dev *dev, int len)
{
        char path[64]="";
        struct file *fp;
        loff_t pos=0;
        int ret;
        struct inode *inode = NULL;
        loff_t size;

        ret = snprintf(path,sizeof(path),"/lib/firmware/mediatek/%s_rf.bin",dev->dev->driver->name);
        if(ret<0)
                return -EINVAL;
        dev_info(dev->dev,"Load eeprom: %s\n",path);
        fp = filp_open(path, O_RDONLY, 0);
        if (IS_ERR(fp)) {
                dev_info(dev->dev,"Open eeprom file failed: %s\n",path);
                return -ENOENT;
        }

        inode = file_inode(fp);
        if ((!S_ISREG(inode->i_mode) && !S_ISBLK(inode->i_mode))) {
                printk(KERN_ALERT "invalid file type: %s\n", path);
                return -ENOENT;
        }
        size = i_size_read(inode->i_mapping->host);
        if (size < 0)
        {
                printk(KERN_ALERT "failed getting size of %s size:%lld \n",path,size);
                return -ENOENT;
        }
        ret = kernel_read(fp, dev->eeprom.data, len, &pos);
        if(ret < size){
                dev_info(dev->dev,"Load eeprom ERR, count %d byte (len:%d)\n",ret,len);
                return -ENOENT;
        }
        filp_close(fp, 0);
        dev_info(dev->dev,"Load eeprom OK, count %d byte\n",ret);

        return 0;
}

int mt76_get_of_eeprom(struct mt76_dev *dev, void *eep, int offset, int len)
{
#if defined(CONFIG_OF) && defined(CONFIG_MTD)
        struct device_node *np = dev->dev->of_node;
        struct mtd_info *mtd;
        const __be32 *list;
        const char *part;
        phandle phandle;
        int size;
        size_t retlen;
        int ret;

        if (!np)
                return -ENOENT;

        list = of_get_property(np, "mediatek,mtd-eeprom", &size);
[... 288 lines cut ...]
        mt76_apply_multi_array_limit(dest->mcs[0], ARRAY_SIZE(dest->mcs[0]),
                                     ARRAY_SIZE(dest->mcs), val, len,
                                     target_power, txs_delta, &max_power);

        val = mt76_get_of_array(np, "rates-ru", &len, ru_rates + 1);
        mt76_apply_multi_array_limit(dest->ru[0], ARRAY_SIZE(dest->ru[0]),
                                     ARRAY_SIZE(dest->ru), val, len,
                                     target_power, txs_delta, &max_power);

        return max_power;
}
EXPORT_SYMBOL_GPL(mt76_get_rate_power_limits);

int
mt76_eeprom_init(struct mt76_dev *dev, int len)
{
        dev->eeprom.size = len;
        dev->eeprom.data = devm_kzalloc(dev->dev, len, GFP_KERNEL);
        if (!dev->eeprom.data)
                return -ENOMEM;

        return (!mt76_get_eeprom_file(dev, len)) ||(!mt76_get_of_eeprom(dev, dev->eeprom.data, 0, len));
}
EXPORT_SYMBOL_GPL(mt76_eeprom_init);
`

agevlakh, 2025-02-02:
eeprom.c file:

// SPDX-License-Identifier: ISC
/*
 * Copyright (C) 2016 Felix Fietkau <nbd@nbd.name>
 */
#include <linux/of.h>
#include <linux/of_net.h>
#include <linux/mtd/mtd.h>
#include <linux/mtd/partitions.h>
#include <linux/etherdevice.h>
#include "mt76.h"

static int mt76_get_eeprom_file(struct mt76_dev *dev, int len)
{
        char path[64]="";
        struct file *fp;
        loff_t pos=0;
        int ret;
        struct inode *inode = NULL;
        loff_t size;

        ret = snprintf(path,sizeof(path),"/lib/firmware/mediatek/%s_rf.bin",dev->dev->driver->name);
        if(ret<0)
                return -EINVAL;
        dev_info(dev->dev,"Load eeprom: %s\n",path);
        fp = filp_open(path, O_RDONLY, 0);
        if (IS_ERR(fp)) {
                dev_info(dev->dev,"Open eeprom file failed: %s\n",path);
                return -ENOENT;
        }

        inode = file_inode(fp);
        if ((!S_ISREG(inode->i_mode) && !S_ISBLK(inode->i_mode))) {
                printk(KERN_ALERT "invalid file type: %s\n", path);
                return -ENOENT;
        }
        size = i_size_read(inode->i_mapping->host);
        if (size < 0)
        {
                printk(KERN_ALERT "failed getting size of %s size:%lld \n",path,size);
                return -ENOENT;
        }
        ret = kernel_read(fp, dev->eeprom.data, len, &pos);
        if(ret < size){
                dev_info(dev->dev,"Load eeprom ERR, count %d byte (len:%d)\n",ret,len);
                return -ENOENT;
        }
        filp_close(fp, 0);
        dev_info(dev->dev,"Load eeprom OK, count %d byte\n",ret);

        return 0;
}

int mt76_get_of_eeprom(struct mt76_dev *dev, void *eep, int offset, int len)
{
#if defined(CONFIG_OF) && defined(CONFIG_MTD)
        struct device_node *np = dev->dev->of_node;
        struct mtd_info *mtd;
        const __be32 *list;
        const char *part;
        phandle phandle;
        int size;
        size_t retlen;
        int ret;

        if (!np)
                return -ENOENT;

        list = of_get_property(np, "mediatek,mtd-eeprom", &size);
[... 287 lines cut ...]
        val = mt76_get_of_array(np, "rates-mcs", &len, mcs_rates + 1);
        mt76_apply_multi_array_limit(dest->mcs[0], ARRAY_SIZE(dest->mcs[0]),
                                     ARRAY_SIZE(dest->mcs), val, len,
                                     target_power, txs_delta, &max_power);

        val = mt76_get_of_array(np, "rates-ru", &len, ru_rates + 1);
        mt76_apply_multi_array_limit(dest->ru[0], ARRAY_SIZE(dest->ru[0]),
                                     ARRAY_SIZE(dest->ru), val, len,
                                     target_power, txs_delta, &max_power);

        return max_power;
}
EXPORT_SYMBOL_GPL(mt76_get_rate_power_limits);

int
mt76_eeprom_init(struct mt76_dev *dev, int len)
{
        dev->eeprom.size = len;
        dev->eeprom.data = devm_kzalloc(dev->dev, len, GFP_KERNEL);
        if (!dev->eeprom.data)
                return -ENOMEM;

        return (!mt76_get_eeprom_file(dev, len)) ||(!mt76_get_of_eeprom(dev, dev->eeprom.data, 0, len));
}
EXPORT_SYMBOL_GPL(mt76_eeprom_init);

frank-w, 2025-02-02:
i said i do not use openwrt...mainly because of these patching hassle...i cannot see anything in this horribly formatted post

agevlakh, 2025-02-03:
Make it on openwrt 22.03.7 and 21.02.7. With higher version can't compiled mt76. Thanks for help...

closed 2025-02-03 by agevlakh